shivasiddharth/GassistPi #552: Youtube Streaming doesn't work

PvtCostUK, 2018-09-30:
### Describe the bug:
```
When i say "Ok Google" and "stream i got you", It shows me error.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes
(I think that it's another error)

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```
ON_RECOGNIZING_SPEECH_FINISHED:
    {"text": "stream i got you"}
stopping vlc
ERROR:__main__:'NoneType' object has no attribute 'group'
Traceback (most recent call last):
    File "/home/pi/GassistPi/src/actions.py", line 1008, in YouTube_No_Autoplay
        say("Getting youtube link")
    File "/home/pi/GassistPi/src/actions.py", line 209, in say
        words= translator.translate(words, dest=language)
    File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 172, in translate
        data = self._translate(text, dest , src)
    File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 75, in _translate
        token = self.token_acquirer.do(text)
    File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
        self._update()
    File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
        code = unicode(self,RE_TKK.search(r.text).group(1)).replace('var ', '')
AttributeError: "NoneType' object has no attribute 'group'

During handling of the above exception, another exception occurred:

Traceback (most recent call last) :
    File "/home/pi/GassistPi/src/main.py", line 662, in <module>
        Myassistant().main()
    File "/home/pi/GassistPi/src/main.py", line 519, in main
        YouTube_No_Autoplay(str(usrcmd). lower())
    File "/home/pi/GassistPi/src/actions.py", line 209, in say
        words= translator.translate(words, dest=language)
    File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 172, in translate
        data = self._translate(text, dest , src)
    File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 75, in _translate
        token = self.token_acquirer.do(text)
    File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
        self._update()
    File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
        code = unicode(self,RE_TKK.search(r.text).group(1)).replace('var ', '')
AttributeError: "NoneType' object has no attribute 'group'
```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

It doesn't exist in /tmp.

shivasiddharth, 2018-09-30:
duplicate of #534

closed 2018-09-30 by shivasiddharth

mihirbhavsar, 2018-12-03:
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "tuning into Radio 1"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "tuning into Radio 1"}
http://prclive1.listenon.in:9960/;
Tuning into 100% NL
WARNING:gtts.tts:Unable to get language list: 'NoneType' object is not subscriptable
ERROR:__main__:substring not found
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 692, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 554, in main
    radio(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 284, in radio
    say("Tuning into " + conv)
  File "/home/pi/GassistPi/src/actions.py", line 231, in say
    tts.save(ttsfilename)
  File "/home/pi/env/lib/python3.5/site-packages/gtts/tts.py", line 247, in save
    self.write_to_fp(f)
  File "/home/pi/env/lib/python3.5/site-packages/gtts/tts.py", line 187, in write_to_fp
    part_tk = self.token.calculate_token(part)
  File "/home/pi/env/lib/python3.5/site-packages/gtts_token/gtts_token.py", line 28, in calculate_token
    seed = self._get_token_key()
  File "/home/pi/env/lib/python3.5/site-packages/gtts_token/gtts_token.py", line 65, in _get_token_key
    lineStr = lineStr[lineStr.index("TKK='")+5:len(lineStr)]
ValueError: substring not found

stilll occurred what shoud i do?