shivasiddharth/GassistPi #553: parcel tracking goes wrong

aandroide, 2018-09-30:
### Describe the bug:
```
{"text": "Where is my Parcel"}
ERROR:__main__:'Mobileclient' object has no attribute 'trackings'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 767, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 635, in main
    track()
  File "/home/pi/GassistPi/src/actions.py", line 346, in track
    text=api.trackings.get(tracking=dict(slug=slug, tracking_number=number))
AttributeError: 'Mobileclient' object has no attribute 'trackings'

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-09-30:
Have you added the aftership API in actions.py file ?

aandroide, 2018-09-30:
Yes and i have uncomment 2 line

aandroide, 2018-10-01:
thanks for the fix now works.
I just wanted to point out to you that in the readme file you wrote that the bees to trace the packages are inserted in config.yaml. while it should be inserted in the actions.py file currently.

aandroide, 2018-10-04:
ok, i have created the pull request