shivasiddharth/GassistPi #559: I can't Play music on spotify

Gabigeek, 2018-10-05:
### Describe the bug:
```
The Google Assistant works great, but when I ask him to play some music on spotify, he always replies me that "this content cannot be played on this device but it can be played on other devices"
I've already put my client ID, client secret and username in the config.yaml file and in the actions.py file and it does nothing.

The command I use to launch the assistant :  googlesamples-assistant-hotword --project-id <project-id> --device-model-id <device-model-id>
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
Traceback (most recent call last):
  File "/home/pi/google/GassistPi/src/main.py", line 18, in <module>
    from kodijson import Kodi, PLAYER_VIDEO
ImportError: No module named 'kodijson'

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.
No log files were created

shivasiddharth, 2018-10-05:
Looks like you have tried to manually run the assistant outside of the environment. I have already given the command in the README.

Gabigeek, 2018-10-05:
you mean that I should use this command to run the assistant ?

> /home/_USER_NAME_/env/bin/python -u /home/_USER_NAME_/GassistPi/src/main.py --device_model_id 'replace this with the model id'

shivasiddharth, 2018-10-05:
Yes.

Gabigeek, 2018-10-05:
The problem is when I run this command I get this error :

`/home/pi/google/env/bin/python: can't find '__main__' module in '/home/pi/google/GassistPi/src/'`

shivasiddharth, 2018-10-05:
Have you renamed the script or has the script not been saved after making edits ? Also can you double the path of the Project folder. Are you using the latest version of raspbian or any other image ?

Gabigeek, 2018-10-05:
I'm currently running the latest version of Raspbian Jessie because Stretch was causing some driver trouble with my wifi card, the script is still named "main.py" and, what do you mean by doubling the path of the project folder ? (sorry I'm french and I'm not very good at english 😅 )

shivasiddharth, 2018-10-05:
Unfortunately this project does not work well with Jessie (snowboy etc). People have run into issues. You really need to be on stretch. I have written and tested all these codes for stretch so troubleshooting for Jessie is difficult from my side (this is just one person doing and managing all the stuff, not a team).

Gabigeek, 2018-10-05:
Oh I see, well np I'll install a fresh Raspbian Stretch,then I'll start again the tutorial and see if there's changes I'll keep you updated 😄

Gabigeek, 2018-10-06:
So, I now have a fresh Raspbian Stretch installation, I've installed GassistPi, all is configured, automatic start in background is set, API keys etc. are set in the confs files. Now when I ask to play a playlist on spotify , I got this Error : 
```

ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "mets ma playlist Dark sur spotify"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "mets ma playlist Dark sur spotify"}
stopping vlc
ERROR:__main__:'NoneType' object has no attribute 'group'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 662, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 646, in main
    spotify_playlist_select(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1337, in spotify_playlist_select
    say("Getting music links")
  File "/home/pi/GassistPi/src/actions.py", line 209, in say
    words= translator.translate(words, dest=language)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 172, in translate
    data = self._translate(text, dest, src)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 75, in _translate
    token = self.token_acquirer.do(text)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
    self._update()
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
    code = unicode(self.RE_TKK.search(r.text).group(1)).replace('var ', '')
AttributeError: 'NoneType' object has no attribute 'group'

```

> "mets ma playlist Dark sur Spotify"

 means : Play the "Dark" Playlist on Spotify (dark is the name of the playlist)

shivasiddharth, 2018-10-06:
The syntax is a bit off. it should be "jouer Dark de Spotify" no need to mention the word playlist. I have done all the configuration for English commands, so you may have to change the following lines according to your language
https://github.com/shivasiddharth/GassistPi/blob/642941a9ff87ed268b726ef13dbbf4f5b9b85300/src/actions.py#L1331
https://github.com/shivasiddharth/GassistPi/blob/642941a9ff87ed268b726ef13dbbf4f5b9b85300/src/actions.py#L1334
https://github.com/shivasiddharth/GassistPi/blob/642941a9ff87ed268b726ef13dbbf4f5b9b85300/src/actions.py#L1335

for the above error refer to #534

Gabigeek, 2018-10-07:
Hi,
I just edited actions.py file to change the play word, and I've created a new gtoken.py file like you said here : https://github.com/shivasiddharth/GassistPi/issues/534#issuecomment-423495674 and now I got 2 (new) problems :
first, My device is no longer registered, I get this message (but Gassistant still works) :

> device_model_id: elite-dreamer-218519
> device_id: 5EBA1EC4507027AA325957B1DC421D69
> 
> 
>     This device is not registered. This means you will not be able to use
>     Device Actions or see your device in Assistant Settings. In order to
>     register this device follow instructions at:
> 
>     https://developers.google.com/assistant/sdk/guides/library/python/embed/register-device

and when I ask to play the playlist on spotify, i get this error : 

> ON_CONVERSATION_TURN_STARTED
> ON_END_OF_UTTERANCE
> ON_END_OF_UTTERANCE
> ON_RECOGNIZING_SPEECH_FINISHED:
>   {"text": "joue Dark sur spotify"}
> ON_RECOGNIZING_SPEECH_FINISHED:
>   {"text": "joue Dark sur spotify"}
> stopping vlc
> Obtenir des liens de musique
> High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
>         version 1.23.8; written and copyright by Michael Hipp and others
>         free software (LGPL) without any warranty but with best wishes
> 
> Directory: /tmp/
> 
> Terminal control enabled, press 'h' for listing of keys and functions.
> 
> Playing MPEG stream 1 of 1: say.mp3 ...
> 
> MPEG 2.0 L III cbr32 24000 mono
> 
> [0:00] Decoding of say.mp3 finished.
> 
> 
> Terminal control enabled, press 'h' for listing of keys and functions.
> 
> Playing MPEG stream 1 of 1: say.mp3 ...
> 
> MPEG 2.0 L III cbr32 24000 mono
> 
> [0:02] Decoding of say.mp3 finished.
> dark
> ERROR:__main__:name 'spotify_token' is not defined
> Traceback (most recent call last):
>   File "/home/pi/GassistPi/src/main.py", line 662, in <module>
>     Myassistant().main()
>   File "/home/pi/GassistPi/src/main.py", line 646, in main
>     spotify_playlist_select(str(usrcmd).lower())
>   File "/home/pi/GassistPi/src/actions.py", line 1339, in spotify_playlist_select
>     playlists,num=scan_spotify_playlists()
>   File "/home/pi/GassistPi/src/actions.py", line 1287, in scan_spotify_playlists
>     if spotify_token:
> NameError: name 'spotify_token' is not defined

shivasiddharth, 2018-10-07:
Dont worry about the device registration. Its an upstream issue. Looks like spotify credentials are missing in the config.yaml file.

shivasiddharth, 2018-10-07:
In the actions.py file, after the following line
https://github.com/shivasiddharth/GassistPi/blob/642941a9ff87ed268b726ef13dbbf4f5b9b85300/src/actions.py#L74
add:
`print(str(spotify_token))`
now share the terminal contents

InventivetalentDev, 2018-10-07:
Getting the same error as above:  

```
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play my music from Spotify"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play my music from Spotify"}
stopping vlc
ERROR:__main__:'NoneType' object has no attribute 'group'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 662, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 646, in main
    spotify_playlist_select(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1337, in spotify_playlist_select
    say("Getting music links")
  File "/home/pi/GassistPi/src/actions.py", line 209, in say
    words= translator.translate(words, dest=language)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 172, in translate
    data = self._translate(text, dest, src)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 75, in _translate
    token = self.token_acquirer.do(text)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
    self._update()
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
    code = unicode(self.RE_TKK.search(r.text).group(1)).replace('var ', '')
AttributeError: 'NoneType' object has no attribute 'group'
```

looking at the code from the trace, it seems like it's having issues with the translator that's supposed to say "Getting music links"

shivasiddharth, 2018-10-07:
@InventivetalentDev Check #534 i have provided the solution there

InventivetalentDev, 2018-10-07:
@shivasiddharth just found it, sorry!

Gabigeek, 2018-10-08:
Hi,

I've added  `print(str(spotify_token))` like this :

```
# Spotify Declarations
# Register with spotify for a developer account to get client-id and client-secret
if configuration['Spotify']['client_id']!= '934050ffb2be475d8xxxxxxxxxxxxxxx' and configuration['Spotify']['client_secret']!='cd82bf2bb359427baxxxxxxxxxxxxxxx':
    client_id = configuration['Spotify']['client_id']
    client_secret = configuration['Spotify']['client_secret']
    username=configuration['Spotify']['username']
    credentials = oauth2.SpotifyClientCredentials(client_id=client_id, client_secret=client_secret)
    spotify_token = credentials.get_access_token()
    print(str(spotify_token))
```
and I got this when i asked for the spotify command (same as before) :

```
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "jouer Dark sur spotify"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "jouer Dark sur spotify"}
stopping vlc
Obtenir des liens de musique
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:02] Decoding of say.mp3 finished.
r dark
ERROR:__main__:name 'spotify_token' is not defined
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 662, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 646, in main
    spotify_playlist_select(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1340, in spotify_playlist_select
    playlists,num=scan_spotify_playlists()
  File "/home/pi/GassistPi/src/actions.py", line 1288, in scan_spotify_playlists
    if spotify_token:
NameError: name 'spotify_token' is not defined

```

shivasiddharth, 2018-10-08:
There is no need to enter your id here. `if configuration['Spotify']['client_id']!= '934050ffb2be475d8xxxxxxxxxxxxxxx' and configuration['Spotify']['client_secret']!='cd82bf2bb359427baxxxxxxxxxxxxxxx': `
You need to  enter your credentials only in the config.yaml file. Replace that part with the default line 
https://github.com/shivasiddharth/GassistPi/blob/b674f776bbe3f0285d66aa97b513952dd6ba07ce/src/actions.py#L69
and change your "joue" in actions.py to "jouer"

Gabigeek, 2018-10-08:
Ok, Thank you so munch it now works I can hear the music !!
3 last things:

first, do i need to pay attention at these new errors ? (from Line 42) :

```

1. ON_RECOGNIZING_SPEECH_FINISHED:
2.   {"text": "joue Dark sur spotify"}
3. ON_RECOGNIZING_SPEECH_FINISHED:
4.   {"text": "joue Dark sur spotify"}
5. stopping vlc
6. Obtenir des liens de musique
7. High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
8.         version 1.23.8; written and copyright by Michael Hipp and others
9.         free software (LGPL) without any warranty but with best wishes
10. 
11. Directory: /tmp/
12. 
13. Terminal control enabled, press 'h' for listing of keys and functions.
14. 
15. Playing MPEG stream 1 of 1: say.mp3 ...
16. 
17. MPEG 2.0 L III cbr32 24000 mono
18. 
19. [0:02] Decoding of say.mp3 finished.
20. dark
21. 23
22. shutdown.
23. summer 2k18 🔥
24. summer 2k18 🔥
25. reborn ⏲☇
26. reborn ⏲☇
27. cocktail été ����
28. cocktail été ����
29. cocktail été ����
30. cocktail été ����
31. #l❤
32. #l❤
33. #l❤
34. #l❤
35. #l❤
36. power 🎧💿
37. power 🎧💿
38. power 🎧💿
39. power 🎧💿
40. dark 🤘
41. Nirvana Smells Like Teen Spirit
42. WARNING:googleapiclient.discovery_cache:file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
43. Traceback (most recent call last):
44.   File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 36, in autodetect
45.     from google.appengine.api import memcache
46. ImportError: No module named 'google.appengine'
47. 
48. During handling of the above exception, another exception occurred:
49. 
50. Traceback (most recent call last):
51.   File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
52.     from oauth2client.contrib.locked_file import LockedFile
53. ImportError: No module named 'oauth2client.contrib.locked_file'
54. 
55. During handling of the above exception, another exception occurred:
56. 
57. Traceback (most recent call last):
58.   File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
59.     from oauth2client.locked_file import LockedFile
60. ImportError: No module named 'oauth2client.locked_file'
61. 
62. During handling of the above exception, another exception occurred:
63. 
64. Traceback (most recent call last):
65.   File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
66.     from . import file_cache
67.   File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
68.     'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
69. ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
70. [0254e710] pulse audio output error: PulseAudio server connection failure: Connection refused
71. ON_CONVERSATION_TURN_FINISHED:
72.   {"with_follow_on_turn": false}
73. ON_CONVERSATION_TURN_FINISHED:
74.   {"with_follow_on_turn": false}

```
2) Is there some vocal commands like play pause next etc. to control spotify ?

3) Awesome Job 😄

Gabigeek, 2018-10-08:
Ok thanks a lot,I suppose there's a way to translate the media control words, where can I found the keywords to translate ?
(btw, is there a way to pair system commands ton vocal command like poweroff so I can turn of the rapberry by voice ?)

shivasiddharth, 2018-10-08:
All the keywords are present in the keywords.yaml file. The command to shutdown pi is "trigger shutdown" you can change trigger to something you want in keywords.yaml file.

Currently extension of the project to all the languages is in progress.

shivasiddharth, 2018-10-08:
Open the /home/pi/ directory, right click and choose to see hidden files and then delete a file called .mute

Gabigeek, 2018-10-08:
I've deleted the .mute file, ran the usb-mic-onboard-jack.sh script to make sure mic and speakers are configured but nothing changed :/

```
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.3
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi                                                                                                                                                             .log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:failed to authenticate
BQDdyW1SByPYW-lcIgnjcZ5X9s2PUY0BqJv_FTY6pEa2y_AcG6J0P6eC3ECRTsijGWhlkuHRcs3BTigj                                                                                                                                                             -Fw
[0305ec78] pulse audio output error: PulseAudio server connection failure: Conne                                                                                                                                                             ction refused
[03052020] pulse audio output error: PulseAudio server connection failure: Conne                                                                                                                                                             ction refused
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                                                                                             35_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                                                                                             or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                                                                                             ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                                                                                             35_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                                                                                             or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                                                                                             ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                                                                                             35_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                                                                                             or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                                                                                             ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                                                                                             35_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                                                                                             or: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directo                                                                                                                                                             ry
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm28                                                                                                                                                             35_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                                                                       [... cut ...]
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unloc                                                                                                                                                             k
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unloc                                                                                                                                                             k
device_model_id: xxxxxxxxxxxxxxxx
device_id: xxxxxxxxxxx

ON_MUTED_CHANGED:
  {"is_muted": true}
ON_MUTED_CHANGED:
  {"is_muted": true}
ON_START_FINISHED
ON_START_FINISHED

```

Gabigeek, 2018-10-08:
Ok I have good news,I still don't know why the mic is off and how to turn it back on (using software method) but I founded a more "manual" solution : I took a breadbord, some cables and a push button, then I used the GPIO23 pin to make a mute/demute play/pause button and it worked !

Niklas9401, 2018-10-08:
@shivasiddharth When i command to play a playlist on spotify the assistant is crashing and isn't available anymore...Then i need to restart the service or the pi....The credentials are in the config file too

shivasiddharth, 2018-10-08:
Manually run the assistant, give the command and post the output of the terminal

Niklas9401, 2018-10-08:
Here is the log

`ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play one and one von spotify"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play one and one von spotify"}
stopping vlc
ERROR:__main__:'NoneType' object has no attribute 'group'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 665, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 649, in main
    spotify_playlist_select(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1337, in spotify_playlist_select
    say("Getting music links")
  File "/home/pi/GassistPi/src/actions.py", line 209, in say
    words= translator.translate(words, dest=language)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 172, in translate
    data = self._translate(text, dest, src)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 75, in _translate
    token = self.token_acquirer.do(text)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
    self._update()
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
    code = unicode(self.RE_TKK.search(r.text).group(1)).replace('var ', '')
AttributeError: 'NoneType' object has no attribute 'group'`

shivasiddharth, 2018-10-08:
Please go through the above discussion, it has already been discussed.

closed 2018-10-08 by shivasiddharth

Gabigeek, 2018-10-13:
Hi,
Ok I will test the branch, I'm reinstalling this new branch at this moment

shivasiddharth, 2018-10-13:
i have fixed all those errors and merged it with master. try cloning the master

Gabigeek, 2018-10-13:
I've juste completed the configuration and I still have the https://github.com/shivasiddharth/GassistPi/issues/559#issuecomment-427532659 issue, so I'm about to follow the solution you told me before.

shivasiddharth, 2018-10-13:
you can change the language in config.yaml file. In the config.yaml file, you will have an option called language and under that choice.

Gabigeek, 2018-10-13:
I've just replaced 'en' by 'fr' in  the config.yaml at this line and I still got the same error as https://github.com/shivasiddharth/GassistPi/issues/559#issuecomment-427532659 :
https://github.com/shivasiddharth/GassistPi/blob/afa2c62582fc18b36c65d6529b337fe237596561/src/config.yaml#L117

shivasiddharth, 2018-10-13:
That error you need to manually fix like how i have mentioned in #534

Gabigeek, 2018-10-13:
So I've just made changes like you said here https://github.com/shivasiddharth/GassistPi/issues/534#issuecomment-423495674 , but now when I try to play a playlist no errors seems to appear, but the "process" to play the playlist seems to stop for whatever reason; here's my terminal :

```
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "joue Dark sur spotify"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "joue Dark sur spotify"}
stopping vlc
Obtenir des liens de musique
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:02] Decoding of say.mp3 finished.
}
23
shutdown.
summer 2k18 🔥
summer 2k18 🔥
reborn ⏲☇
reborn ⏲☇
cocktail été ����
cocktail été ����
cocktail été ����
cocktail été ����
#l❤
#l❤
#l❤
#l❤
#l❤
power 🎧💿
power 🎧💿
power 🎧💿
power 🎧💿
dark 🤘
mes titres shazam
mes titres shazam
mes titres shazam
mes titres shazam
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

```

shivasiddharth, 2018-10-13:
The syntax is "from spotify" not "on spotify" i have set  play to "jouer" in keywords_fr.yaml if you want you can change it to "joue" 

https://github.com/shivasiddharth/GassistPi/blob/afa2c62582fc18b36c65d6529b337fe237596561/src/keywords_fr.yaml#L92

Gabigeek, 2018-10-13:
Ok thanks, btw I can see some translation errors, so if you want I can send here a "adjusted" version of the french keyword file :) non-conjugated verbs (I think we say an "infinitve" verb in english") are different from conjugated verbs (My english is pretty bad so idk if you understood what I am saying so here's an exemple with the "play" verb :

English : the "Play" verb in a sentence : " Play xxx from spotify" -> spelling is the same
French : the "jouer" verb in a sentence : " Joue (and not jouer) xxx sur Spotify" -> the infinitive "jouer" became "joue"

Another exemple with the "Increase" verb :

English : The "Increase" verb in a sentence : "Increase the volume by 5" -> Spelling is the same
French : The "augmenter" verb in a sentence :  Augmente le volume de* 5" -> The infinitive "Jouer" became "joue"

(*the "de" word has multiple meaning according to the context of the sentence)

Gabigeek, 2018-10-15:
-Open curtains is "ouvre (open) les rideaux (curtains)"
-Close curtains is "ferme (close) les rideaux (curtains)"
If I've Understood correctly, halt is a synonym of stop, so :
-Halt curtains is "arrête (halt) les rideaux (curtains)"

If you have other things to translate, you know were I am :)

shivasiddharth, 2018-10-15:
Thanks a lot.. 👍