shivasiddharth/GassistPi #560: Question about YouTube streaming

mr-sinical, 2018-10-06:
first, I have to say awesome work matey and second sorry its probably a dumb question .. but is there a way to output the youtube sound stream as a youtube video stream .. for if there is a screen connected to the pi

shivasiddharth, 2018-10-06:
Please use the gitter for non-issue stuff. Yes, if you want to, can be done. change audiostream to videostream in https://github.com/shivasiddharth/GassistPi/blob/642941a9ff87ed268b726ef13dbbf4f5b9b85300/src/actions.py#L1014

closed 2018-10-06 by shivasiddharth

mr-sinical, 2018-10-06:
thanks for the reply and sorry matey I've never been that good a get my posts in the right place..  to be honest, I've never used gitter, didn't even know of it matey .. But ill head over there and put my 2 cent in on how awesome this project is especially when you combine it with mosquitto and node red

shivasiddharth, 2018-10-06:
I am always looking for suggestions to implement new custom actions. If you have something in mind let me know.