shivasiddharth/GassistPi #562: Domoticz connection

Peerkersezuuker, 2018-10-07:
### Describe the bug:
If I ask to turn a switch, the assistent say's ok and the nothing happens.
When I look in the log is see nothing happening with the def domoticz_control(i,query,index,devicename): in the actions.py ( I put some print() commands in this def but they are not showing)

In the main.py it states : (LINE 318)
if event.type == EventType.ON_DEVICE_ACTION:
for command, params in event.actions:
print('Do command', command, 'with params', str(params))

But no further actions are defined after this print command.

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No
## If answer to question 2 is Yes, then paste the contents of the terminal below.

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

Piece of the syslog when stating the command :

Oct  6 11:15:18 defiant python[28846]: ON_RECOGNIZING_SPEECH_FINISHED:
Oct  6 11:15:18 defiant python[28846]:   {"text": "speaker switch on"}
Oct  6 11:15:18 defiant python[28846]: ON_DEVICE_ACTION:
Oct  6 11:15:18 defiant python[28846]: {
Oct  6 11:15:18 defiant python[28846]:   "inputs": [
Oct  6 11:15:18 defiant python[28846]:     {
Oct  6 11:15:18 defiant python[28846]:       "context": {
Oct  6 11:15:18 defiant python[28846]:         "locale_language": "en"
Oct  6 11:15:18 defiant python[28846]:       },
Oct  6 11:15:18 defiant python[28846]:       "intent": "action.devices.EXECUTE",
Oct  6 11:15:18 defiant python[28846]:       "payload": {
Oct  6 11:15:18 defiant python[28846]:         "commands": [
Oct  6 11:15:18 defiant python[28846]:           {
Oct  6 11:15:18 defiant kernel: [75612.497578] pcm512x 1-004d: No SCLK, using BCLK: -2
Oct  6 11:15:18 defiant python[28846]:             "devices": [
Oct  6 11:15:18 defiant python[28846]:               {
Oct  6 11:15:18 defiant python[28846]:                 "id": "E92044DDDCDA86901336786644CADF68"
Oct  6 11:15:18 defiant python[28846]:               }
Oct  6 11:15:18 defiant python[28846]:             ],
Oct  6 11:15:18 defiant python[28846]:             "execution": [
Oct  6 11:15:18 defiant python[28846]:               {
Oct  6 11:15:18 defiant python[28846]:                 "command": "action.devices.commands.OnOff",
Oct  6 11:15:19 defiant python[28846]:                 "params": {
Oct  6 11:15:19 defiant python[28846]:                   "on": true
Oct  6 11:15:19 defiant python[28846]:                 }
Oct  6 11:15:19 defiant python[28846]:               }
Oct  6 11:15:19 defiant python[28846]:             ]
Oct  6 11:15:19 defiant python[28846]:           }
Oct  6 11:15:19 defiant python[28846]:         ]
Oct  6 11:15:19 defiant python[28846]:       }
Oct  6 11:15:19 defiant python[28846]:     }
Oct  6 11:15:19 defiant python[28846]:   ],
Oct  6 11:15:19 defiant python[28846]:   "requestId": "12402147356067811675"
Oct  6 11:15:19 defiant python[28846]: }
Oct  6 11:15:19 defiant python[28846]: ON_DEVICE_ACTION:
Oct  6 11:15:19 defiant python[28846]: {
Oct  6 11:15:19 defiant python[28846]:   "inputs": [
Oct  6 11:15:19 defiant python[28846]:     {
Oct  6 11:15:19 defiant python[28846]:       "context": {
Oct  6 11:15:19 defiant python[28846]:         "locale_language": "en"
Oct  6 11:15:19 defiant python[28846]:       },
Oct  6 11:15:19 defiant python[28846]:       "intent": "action.devices.EXECUTE",
Oct  6 11:15:19 defiant python[28846]:       "payload": {
Oct  6 11:15:19 defiant python[28846]:         "commands": [
Oct  6 11:15:19 defiant python[28846]:           {
Oct  6 11:15:19 defiant python[28846]:             "devices": [
Oct  6 11:15:19 defiant python[28846]:               {
Oct  6 11:15:19 defiant python[28846]:                 "id": "E92044DDDCDA86901336786644CADF68"
Oct  6 11:15:19 defiant python[28846]:               }
Oct  6 11:15:19 defiant python[28846]:             ],
Oct  6 11:15:19 defiant python[28846]:             "execution": [
Oct  6 11:15:19 defiant python[28846]:               {
Oct  6 11:15:19 defiant python[28846]:                 "command": "action.devices.commands.OnOff",
Oct  6 11:15:19 defiant python[28846]:                 "params": {
Oct  6 11:15:19 defiant python[28846]:                   "on": true
Oct  6 11:15:19 defiant python[28846]:                 }
Oct  6 11:15:19 defiant python[28846]:               }
Oct  6 11:15:19 defiant python[28846]:             ]
Oct  6 11:15:19 defiant python[28846]:           }
Oct  6 11:15:19 defiant python[28846]:         ]
Oct  6 11:15:19 defiant python[28846]:       }
Oct  6 11:15:19 defiant python[28846]:     }
Oct  6 11:15:19 defiant python[28846]:   ],
Oct  6 11:15:19 defiant python[28846]:   "requestId": "12402147356067811675"
Oct  6 11:15:19 defiant python[28846]: }
Oct  6 11:15:19 defiant python[28846]: Do command action.devices.commands.OnOff with params {'on': True}
Oct  6 11:15:19 defiant python[28846]: ON_RESPONDING_STARTED:
Oct  6 11:15:19 defiant python[28846]:   {"is_error_response": false}
Oct  6 11:15:19 defiant python[28846]: ON_RESPONDING_STARTED:
Oct  6 11:15:19 defiant python[28846]:   {"is_error_response": false}

shivasiddharth, 2018-10-07:
1. By default, i have disabled the domoticz control. Can you confirm if you have enabled it in the config.yaml file
https://github.com/shivasiddharth/GassistPi/blob/642941a9ff87ed268b726ef13dbbf4f5b9b85300/src/config.yaml#L24
Also the IP address of the domoticz server needs to be entered there.

2. The domoticz server should be running when the google assistant starts. The assistant gets the devices list at the begining.

Peerkersezuuker, 2018-10-07:
Hi,
Yes I enabled domoticz int the config.yaml, and I see the assistent retrieving the devices.
Attached the domoticz_device_list.json it creates at start up (renamed it *.txt fot the upload)
[domoticz_device_list.json.txt](https://github.com/shivasiddharth/GassistPi/files/2453885/domoticz_device_list.json.txt)

shivasiddharth, 2018-10-07:
Lets debug this setup by step. First i need to verify if the device is going into the domoticz custom action. After this line : 
https://github.com/shivasiddharth/GassistPi/blob/642941a9ff87ed268b726ef13dbbf4f5b9b85300/src/main.py#L470
Add the following print command:
`print(domoticz_devices['result'][i]['HardwareName'])`

Can you manually start the assistant and post the terminal output after making the above mentioned changes.

Peerkersezuuker, 2018-10-07:
Hi I made the adjustment, but it is not showing I the log.
I put extra print cmd (dbug -xx-) In so I could see where main is in the log. 
First I put a print debug in the for loop, but then I see a lot of debug -xx- messgaes, so the for loop is working.

               if Domoticz_Device_Control==True and len(domoticz_devices['result'])>0:
                    print("debug -01-")
                    for i in range(0,len(domoticz_devices['result'])):
                        if str(domoticz_devices['result'][i]['HardwareName']).lower() in str(usrcmd).lower():
                            print("debug -02-")
                            assistant.stop_conversation()
                            print("debug -03-")
                            print(domoticz_devices['result'][i]['HardwareName'])
                            domoticz_control(i,str(usrcmd).lower(),domoticz_devices['result'][i]['idx'],domoticz_devices['result'][i]['HardwareName'])
                            break

The log :

 7 15:30:20 defiant python[1084]: ON_RECOGNIZING_SPEECH_FINISHED:
Oct  7 15:30:20 defiant python[1084]:   {"text": "speaker switch on"}
Oct  7 15:30:20 defiant python[1084]: debug -01-
Oct  7 15:30:20 defiant kernel: [ 1110.536951] pcm512x 1-004d: No SCLK, using BCLK: -2
Oct  7 15:30:20 defiant python[1084]: ON_DEVICE_ACTION:
Oct  7 15:30:20 defiant python[1084]: {
Oct  7 15:30:20 defiant python[1084]:   "inputs": [
Oct  7 15:30:20 defiant python[1084]:     {
Oct  7 15:30:20 defiant python[1084]:       "context": {
Oct  7 15:30:20 defiant python[1084]:         "locale_language": "en"
Oct  7 15:30:20 defiant python[1084]:       },
Oct  7 15:30:20 defiant python[1084]:       "intent": "action.devices.EXECUTE",
Oct  7 15:30:20 defiant python[1084]:       "payload": {
Oct  7 15:30:20 defiant python[1084]:         "commands": [
Oct  7 15:30:20 defiant python[1084]:           {
Oct  7 15:30:20 defiant python[1084]:             "devices": [
Oct  7 15:30:20 defiant python[1084]:               {
Oct  7 15:30:20 defiant python[1084]:                 "id": "E92044DDDCDA86901336786644CADF68"
Oct  7 15:30:20 defiant python[1084]:               }
Oct  7 15:30:20 defiant python[1084]:             ],
Oct  7 15:30:20 defiant python[1084]:             "execution": [
Oct  7 15:30:20 defiant python[1084]:               {
Oct  7 15:30:20 defiant python[1084]:                 "command": "action.devices.commands.OnOff",
Oct  7 15:30:20 defiant python[1084]:                 "params": {
Oct  7 15:30:20 defiant python[1084]:                   "on": true
Oct  7 15:30:20 defiant python[1084]:                 }
Oct  7 15:30:20 defiant python[1084]:               }
Oct  7 15:30:20 defiant python[1084]:             ]
Oct  7 15:30:20 defiant python[1084]:           }
Oct  7 15:30:20 defiant python[1084]:         ]
Oct  7 15:30:20 defiant python[1084]:       }
Oct  7 15:30:20 defiant python[1084]:     }
Oct  7 15:30:20 defiant python[1084]:   ],
Oct  7 15:30:20 defiant python[1084]:   "requestId": "6241472844236604031"
Oct  7 15:30:20 defiant python[1084]: }
Oct  7 15:30:20 defiant python[1084]: ON_DEVICE_ACTION:
Oct  7 15:30:20 defiant python[1084]: {
Oct  7 15:30:20 defiant python[1084]:   "inputs": [
Oct  7 15:30:20 defiant python[1084]:     {
Oct  7 15:30:20 defiant python[1084]:       "context": {
Oct  7 15:30:20 defiant python[1084]:         "locale_language": "en"
Oct  7 15:30:20 defiant python[1084]:       },
Oct  7 15:30:20 defiant python[1084]:       "intent": "action.devices.EXECUTE",
Oct  7 15:30:20 defiant python[1084]:       "payload": {
Oct  7 15:30:20 defiant python[1084]:         "commands": [
Oct  7 15:30:20 defiant python[1084]:           {
Oct  7 15:30:20 defiant python[1084]:             "devices": [
Oct  7 15:30:20 defiant python[1084]:               {
Oct  7 15:30:20 defiant python[1084]:                 "id": "E92044DDDCDA86901336786644CADF68"
Oct  7 15:30:20 defiant python[1084]:               }
Oct  7 15:30:20 defiant python[1084]:             ],
Oct  7 15:30:20 defiant python[1084]:             "execution": [
Oct  7 15:30:20 defiant python[1084]:               {
Oct  7 15:30:20 defiant python[1084]:                 "command": "action.devices.commands.OnOff",
Oct  7 15:30:20 defiant python[1084]:                 "params": {
Oct  7 15:30:20 defiant python[1084]:                   "on": true
Oct  7 15:30:20 defiant python[1084]:                 }
Oct  7 15:30:20 defiant python[1084]:               }
Oct  7 15:30:20 defiant python[1084]:             ]
Oct  7 15:30:20 defiant python[1084]:           }
Oct  7 15:30:20 defiant python[1084]:         ]
Oct  7 15:30:20 defiant python[1084]:       }
Oct  7 15:30:20 defiant python[1084]:     }
Oct  7 15:30:20 defiant python[1084]:   ],
Oct  7 15:30:20 defiant python[1084]:   "requestId": "6241472844236604031"
Oct  7 15:30:20 defiant python[1084]: }
Oct  7 15:30:20 defiant python[1084]: Do command action.devices.commands.OnOff with params {'on': True}
Oct  7 15:30:20 defiant python[1084]: debug -01-
Oct  7 15:30:21 defiant python[1084]: ON_RESPONDING_STARTED:
Oct  7 15:30:21 defiant python[1084]:   {"is_error_response": false}
Oct  7 15:30:21 defiant python[1084]: ON_RESPONDING_STARTED:
Oct  7 15:30:21 defiant python[1084]:   {"is_error_response": false}
Oct  7 15:30:21 defiant python[1084]: debug -01-
Oct  7 15:30:21 defiant python[1084]: ON_RESPONDING_FINISHED
Oct  7 15:30:21 defiant python[1084]: ON_RESPONDING_FINISHED
Oct  7 15:30:21 defiant python[1084]: debug -01-
Oct  7 15:30:21 defiant python[1084]: ON_CONVERSATION_TURN_FINISHED:
Oct  7 15:30:21 defiant python[1084]:   {"with_follow_on_turn": false}
Oct  7 15:30:21 defiant python[1084]: ON_CONVERSATION_TURN_FINISHED:
Oct  7 15:30:21 defiant python[1084]:   {"with_follow_on_turn": false}
Oct  7 15:30:21 defiant python[1084]: debug -01-

Peerkersezuuker, 2018-10-07:
Id put some more print commands in the section to see the hardware name and the usercmd: 

                if Domoticz_Device_Control==True and len(domoticz_devices['result'])>0:
                    print("debug -01-")
                    for i in range(0,len(domoticz_devices['result'])):
                        print("domoticz hardware name : ",str(domoticz_devices['result'][i]['HardwareName']).lower())
                        print("Usercmd : ",str(usrcmd).lower())
                        if str(domoticz_devices['result'][i]['HardwareName']).lower() in str(usrcmd).lower():
                            print("debug -02-")
                            assistant.stop_conversation()
                            print("debug -03-")
                            print(domoticz_devices['result'][i]['HardwareName'])
                            domoticz_control(i,str(usrcmd).lower(),domoticz_devices['result'][i]['idx'],domoticz_devices['result'][i]['HardwareName'])
                            break

I attached the log in txt file because it has to many lines to put in here.
[log.txt](https://github.com/shivasiddharth/GassistPi/files/2453952/log.txt)

shivasiddharth, 2018-10-07:
Ok, its not going into the domoticz loop. In the following lines, change "HardwareName" to "Name"

https://github.com/shivasiddharth/GassistPi/blob/642941a9ff87ed268b726ef13dbbf4f5b9b85300/src/main.py#L469

https://github.com/shivasiddharth/GassistPi/blob/642941a9ff87ed268b726ef13dbbf4f5b9b85300/src/main.py#L471

It should start working now.

Peerkersezuuker, 2018-10-07:
Ok, now it is going into the loop but, I got a failure in the main.py

Oct  7 16:15:52 defiant python[1837]: debug -01-
Oct  7 16:15:52 defiant python[1837]: ON_RECOGNIZING_SPEECH_FINISHED:
Oct  7 16:15:52 defiant python[1837]:   {"text": "speaker switch on"}
Oct  7 16:15:52 defiant python[1837]: ON_RECOGNIZING_SPEECH_FINISHED:
Oct  7 16:15:52 defiant python[1837]:   {"text": "speaker switch on"}
Oct  7 16:15:52 defiant python[1837]: debug -01-
Oct  7 16:15:52 defiant python[1837]: debug -02-
Oct  7 16:15:52 defiant python[1837]: debug -03-
Oct  7 16:15:53 defiant python[1837]: Speaker Switch
Oct  7 16:15:53 defiant python[1837]: https://192.168.0.31:9443/json.htm?type=command&param=switchlight&idx=230&switchcmd=On
Oct  7 16:15:53 defiant python[1837]: /home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py:857: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
Oct  7 16:15:53 defiant python[1837]:   InsecureRequestWarning)
Oct  7 16:15:54 defiant python[1837]: ERROR:__main__:'NoneType' object has no attribute 'group'
Oct  7 16:15:54 defiant python[1837]: Traceback (most recent call last):
Oct  7 16:15:54 defiant python[1837]:   File "/home/pi/GassistPi/src/main.py", line 669, in <module>
Oct  7 16:15:54 defiant python[1837]:     Myassistant().main()
Oct  7 16:15:54 defiant python[1837]:   File "/home/pi/GassistPi/src/main.py", line 478, in main
Oct  7 16:15:54 defiant python[1837]:     domoticz_control(i,str(usrcmd).lower(),domoticz_devices['result'][i]['idx'],domoticz_devices['result'][i]['HardwareName'])
Oct  7 16:15:54 defiant python[1837]:   File "/home/pi/GassistPi/src/actions.py", line 1367, in domoticz_control
Oct  7 16:15:54 defiant python[1837]:     say('Turning on ' + devicename + ' .')
Oct  7 16:15:54 defiant python[1837]:   File "/home/pi/GassistPi/src/actions.py", line 215, in say
Oct  7 16:15:54 defiant python[1837]:     words= translator.translate(words, dest=language)
Oct  7 16:15:54 defiant python[1837]:   File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 172, in translate
Oct  7 16:15:54 defiant python[1837]:     data = self._translate(text, dest, src)
Oct  7 16:15:54 defiant python[1837]:   File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 75, in _translate
Oct  7 16:15:54 defiant python[1837]:     token = self.token_acquirer.do(text)
Oct  7 16:15:54 defiant python[1837]:   File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
Oct  7 16:15:54 defiant python[1837]:     self._update()
Oct  7 16:15:54 defiant python[1837]:   File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
Oct  7 16:15:54 defiant python[1837]:     code = unicode(self.RE_TKK.search(r.text).group(1)).replace('var ', '')
Oct  7 16:15:54 defiant python[1837]: AttributeError: 'NoneType' object has no attribute 'group'

Peerkersezuuker, 2018-10-07:
Yep and that did the trick.
You are the greatest, keep up the good work here.

With Regards Peter

closed 2018-10-07 by Peerkersezuuker