shivasiddharth/GassistPi #563: Domoticz issue continue

Peerkersezuuker, 2018-10-08:
After solving the first issue with domoticz [https://github.com/shivasiddharth/GassistPi/issues/562](url) I was not able to test it further.
This morning I did some test, and the domoticz response is ok, but now when I e.g. ask what the weather is I get the response "turning on Vero"  And asking what is the time gives no response.
In the log is see the response the assistant should give, but is not giving :

8 10:11:22 defiant python[1581]: ON_RECOGNIZING_SPEECH_FINISHED:
Oct  8 10:11:22 defiant python[1581]:   {"text": "what's the weather"}
Oct  8 10:11:22 defiant python[1581]: debug -01-
Oct  8 10:11:22 defiant python[1581]: ON_RENDER_RESPONSE:
Oct  8 10:11:22 defiant python[1581]:   {"text": "Cumulonimbus clouds are some of my favourites!", "type": 0}
Oct  8 10:11:22 defiant python[1581]: ON_RENDER_RESPONSE:
Oct  8 10:11:22 defiant python[1581]:   {"text": "Cumulonimbus clouds are some of my favourites!", "type": 0}
Oct  8 10:11:22 defiant python[1581]: debug -01-
Oct  8 10:11:22 defiant kernel: [ 7155.020777] pcm512x 1-004d: No SCLK, using BCLK: -2
Oct  8 10:11:22 defiant python[1581]: debug -02-
Oct  8 10:11:22 defiant python[1581]: debug -03-
Oct  8 10:11:22 defiant python[1581]: 0
Oct  8 10:11:22 defiant python[1581]: ON_RENDER_RESPONSE:
Oct  8 10:11:22 defiant python[1581]: {
Oct  8 10:11:22 defiant python[1581]:   "text": "Currently in Mierlo it's 8 degrees Celsius and mostly cloudy. Today, it'll be partly cloudy, with a forecast high of 18 and a low of 4.\n---\n( More on weather.com )",
Oct  8 10:11:22 defiant python[1581]:   "type": 0
Oct  8 10:11:22 defiant python[1581]: }
Oct  8 10:11:22 defiant python[1581]: ON_RENDER_RESPONSE:
Oct  8 10:11:22 defiant python[1581]: {
Oct  8 10:11:22 defiant python[1581]:   "text": "Currently in Mierlo it's 8 degrees Celsius and mostly cloudy. Today, it'll be partly cloudy, with a forecast high of 18 and a low of 4.\n---\n( More on weather.com )",
Oct  8 10:11:22 defiant python[1581]:   "type": 0
Oct  8 10:11:22 defiant python[1581]: }
Oct  8 10:11:22 defiant python[1581]: debug -01-
Oct  8 10:11:22 defiant python[1581]: debug -02-
Oct  8 10:11:22 defiant python[1581]: debug -03-
Oct  8 10:11:22 defiant python[1581]: 0
Oct  8 10:11:22 defiant python[1581]: https://192.168.0.31:9443/json.htm?type=command&param=switchlight&idx=383&switchcmd=On
Oct  8 10:11:23 defiant python[1581]: /home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py:857: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
Oct  8 10:11:23 defiant python[1581]:   InsecureRequestWarning)

Oct  8 10:11:45 defiant python[1581]: ON_RECOGNIZING_SPEECH_FINISHED:
Oct  8 10:11:45 defiant python[1581]:   {"text": "what's the time"}
Oct  8 10:11:45 defiant python[1581]: ON_RECOGNIZING_SPEECH_FINISHED:
Oct  8 10:11:45 defiant python[1581]:   {"text": "what's the time"}
Oct  8 10:11:45 defiant python[1581]: debug -01-
Oct  8 10:11:46 defiant python[1581]: ON_RENDER_RESPONSE:
Oct  8 10:11:46 defiant python[1581]:   {"text": "It's 10:11.", "type": 0}
Oct  8 10:11:46 defiant python[1581]: ON_RENDER_RESPONSE:
Oct  8 10:11:46 defiant python[1581]:   {"text": "It's 10:11.", "type": 0}
Oct  8 10:11:46 defiant python[1581]: debug -01-
Oct  8 10:11:46 defiant kernel: [ 7178.778979] pcm512x 1-004d: No SCLK, using BCLK: -2
Oct  8 10:11:46 defiant python[1581]: debug -02-

Peerkersezuuker, 2018-10-08:
I am using a hifiberry dac pro.
When i disable Domoticz in the config the assistant reacts just fine and responds to all my questions correct.

shivasiddharth, 2018-10-08:
I have updated the git. Can you give it a try?

closed 2018-10-08 by shivasiddharth

Peerkersezuuker, 2018-10-08:
This works perfect.
I wil keep on testing, when I find something I will let you know.

Thanks for the effort

shivasiddharth, 2018-10-08:
Right now this controls just bulbs and switches, i would like to add thermostat controls as well. Do you have a thermostat lying around to test. It would be great if you can test that out and get back.

commit 36885eb9 (2018-10-08): Fix domoticz delay

Fixes #563  Fixes #564