shivasiddharth/GassistPi #565: Spotify music doesn't work after fixingthe gtoken.py

Niklas9401, 2018-10-08:
### Describe the bug:
```
I fixed the bug in the gtoken.py and now the assistant is saying: Getting music links...But after that, the assistant seems to crash and no error is occuring in the terminal
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No
Yes
### 2. Is the issue related to assistant installation ?
Yes/No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play 1 & 1 von Spotify"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play 1 & 1 von Spotify"}
stopping vlc
Getting music links
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
1 & 1 von spotify
1
2
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No
Yes
## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-10-08:
> ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play 1 & 1 von Spotify"}

As you can see, its not deciphering your command properly. What is the name of the playlist in your spotify account ?

Niklas9401, 2018-10-08:
1&1 is the name of the playlist and i already tried other playlists

shivasiddharth, 2018-10-08:
Syntax is wrong. it should be from `play 1 & 1 from spotify` The assistant will sometimes recognize & as and. I really suggest you to change th playlist name and try. Also, right now it will play only user created playlists and not other's playlists that you have added to your account.

Niklas9401, 2018-10-08:
I also tried other names and the playlists are generated by myself

Niklas9401, 2018-10-08:
I dont know?! Should i say it in German: "Spiele PLAYLIST von Spotify" or in English...The response will be translated in english even when i change it in the actions.py

shivasiddharth, 2018-10-09:
As of today, the custom commands have to be in english only. Support for other languages is work in progress.

Niklas9401, 2018-10-09:
I reinstalled Raspbian and installed the dev branch but now the assistant seems to be executed twice when i start it manually...I changed the language to german and tried the command with the playlist name "Wiese" but nothing changed.

```pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id niklasraspi-home-raspberry-cxe01d
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.3
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:failed to authenticate
[02b57098] pulse audio output error: PulseAudio server connection failure: Connection refused
[02b4d5d8] pulse audio output error: PulseAudio server connection failure: Connection refused
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
[... cut ...]
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
wiese
1
2
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}```

shivasiddharth, 2018-10-09:
Either the username entered in the config.yaml is incorrect or the playlist is not user created. Thats what i can see from the post above.

uncomment the following line:
https://github.com/shivasiddharth/GassistPi/blob/0938b39cb2e30d72522575f68f96d7603fb589a4/src/actions.py#L1326

Try to give the command again. If that does not print the tracks in the playlist then, its what i said above.

Niklas9401, 2018-10-09:
Now it is giving me the track names. Nevertheless there aren't all tracks of the playlist and it doesn't play anyting

shivasiddharth, 2018-10-09:
Can you post the thing that is coming on the terminal

Niklas9401, 2018-10-09:
```pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id niklasraspi-home-raspberry-cxe01d
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.3
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:failed to authenticate
[02fd7ba8] pulse audio output error: PulseAudio server connection failure: Connection refused
[03013dc8] pulse audio output error: PulseAudio server connection failure: Connection refused
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
[... cut ...]  'Kontra K Diamanten', 'Trailerpark Sterben kannst du überall', '257ers Holz', 'Ingo ohne Flamingo Saufen morgens, mittags, abends', 'Metallica Enter Sandman', 'SDP Tanz aus der Reihe!', 'K.I.Z Ein Affe und ein Pferd', 'SXTN Fotzen im Club', 'Scooter Maria (I Like It Loud) - Remastered', 'Green Day 21 Guns', 'Trailerpark New Kids on the Blech', 'Beginner Ahnma']]]
2
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
```

Niklas9401, 2018-10-09:
The playlist has 290 titles and the script gets only 86 of them

shivasiddharth, 2018-10-09:
> [[0, '2', ['SDP Die Nacht von Freitag auf Montag', 'Ingo ohne Flamingo Hartz 4 und der Tag gehört dir', 'Scooter Bora! Bora! Bora!', 'K.I.Z Boom Boom Boom', 'Alligatoah Trostpreis (feat. Timi [...]

You have requested the playlist called "Wiese". But the title of the playlist that you have in spotify is "2". That is why nothing is playing.

shivasiddharth, 2018-10-09:
Regarding the number of tracks that it is fetching, it could be a limitation of the spotipy api. That needs to be looked into further and that would not be classified as an issue of this project.

Niklas9401, 2018-10-09:
No this was the first try with the playlist 2 now i tried it with wiese...sorry copied wrong request

Niklas9401, 2018-10-09:
Oh no wait....It doesn't matter which playlist i choose its saying ```[[0, '2', ['SDP Die Nacht von Freitag auf Montag'....``` everytime

Niklas9401, 2018-10-09:
I changed the name of the playlist and now its always choosing this one

shivasiddharth, 2018-10-09:
I can just say what the output says to me. There might be other playlists that you have added, but looks like only the playlist "2" has been created by you. Are the other playlists created by you or by some others that you have added?

Niklas9401, 2018-10-09:
Now i have the Wiese and the Baum playlist(random names xD)...Both of them are created on my account...But when i say Play Wiese from Spotify it's choosing the baum playlist....After that it is throwing some errors and plays the music from the playlist...

This happens when i say Play Wiese from Spotify:
```ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "Spiele Wiese von Spotify"}
stopping vlc
Musiklinks erhalten
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
test
1
  total tracks 86
[[0, 'Baum', ['SDP Die Nacht von Freitag auf Montag', 'Ingo ohne Flamingo Hartz 4 und der Tag gehört dir', 'Scooter Bora! Bora! Bora!', 'K.I.Z Boom Boom Boom', 'Alligatoah Trostpreis (feat. Timi Hendrix)', 'Axwell /\\ Ingrosso Sun Is Shining', "Jetta I'd Love To Change The World - Matstubs Remix", "Genetikk Lieb's oder lass es", 'Alligatoah Willst du', 'K.I.Z Käfigbett', 'Eminem Survival', 'Bonez MC Palmen aus Plastik', 'Linkin Park Numb', 'K.I.Z Ariane', 'Bonez MC Ohne mein Team', 'Kontra K Erfolg ist kein Glück', 'Scooter Oi', 'Trailerpark Endlich normale Leute', '257ers Holland', "AronChupa I'm an Albatraoz", 'Ran-D Zombie', 'Trailerpark Selbstbefriedigung', 'Eminem Berzerk', 'K.I.Z Urlaub fürs Gehirn', 'Scooter One (Always Hardcore) - Remastered', 'Kraftklub Dein Lied', 'K.I.Z Wir', 'Kraftklub Schüsse in die Luft', 'Raggabund Ganjatherapie', 'SDP Deine Freundin', '257ers Lisa', 'Honk Hallo Helmut (Andere Farbe)', 'SDP #DAMDAMDAM!', 'Die Orsons Schwung in die Kiste', 'Scooter The Logical Song - Radio Edit', 'Honk Hallo I Bims 1 Bätmän', 'Trailerpark Sexualethisch desorientiert', 'Imagine Dragons Friction', 'Portugal. The Man Feel It Still', 'DJ Antoine House Party - Radio Edit', 'Die Orsons Ventilator', 'ItaloBrothers Stamp On The Ground', 'Alligatoah Fick ihn doch', "Snoop Dogg Drop It Like It's Hot", 'SDP Zeit verschwenden', 'Kay One Louis Louis', 'Eminem Rap God', 'SXTN Ich bin schwarz', 'Linkin Park In The End', 'Deep Purple Smoke On The Water - Remastered 2012', 'Benny Benassi Cinema (Skrillex Remix) - LUCA LUSH Flip', 'Chumbawamba Tubthumping', 'Alligatoah Amnesie', 'Alligatoah Narben', 'Trailerpark Falsche Band', 'SXTN So high', 'Ikke Hüftgold Modeste Song', 'Trailerpark Koks auf Hawaii', 'Trailerpark Bleib in der Schule', "Linkin Park What I've Done", 'SDP Lied für die Fans von den anderen Bands', 'Timmy Trumpet Oracle', '257ers Warum', 'Linkin Park BURN IT DOWN', 'Scooter How Much Is The Fish?', 'Trailerpark Fledermausland', 'Green Day Boulevard of Broken Dreams', 'SDP Wenn ich groß bin', 'K.I.Z Walpurgisnacht', 'K.I.Z Hurra die Welt geht unter', 'Alligatoah Was der Bauer nicht kennt (feat. Timi Hendrix & Basti DNP)', 'Alligatoah Du bist schön', 'Trailerpark TP4L', 'SDP Die Wahrheit in schön', 'Kontra K Diamanten', 'Trailerpark Sterben kannst du überall', '257ers Holz', 'Ingo ohne Flamingo Saufen morgens, mittags, abends', 'Metallica Enter Sandman', 'SDP Tanz aus der Reihe!', 'K.I.Z Ein Affe und ein Pferd', 'SXTN Fotzen im Club', 'Scooter Maria (I Like It Loud) - Remastered', 'Green Day 21 Guns', 'Trailerpark New Kids on the Blech', 'Beginner Ahnma']]]
baum
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}```

And this happens when i say Play Baum von Spotify:
```ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "Spiele Baum von spotify"}
stopping vlc
Musiklinks erhalten
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
baum
1
  total tracks 86
[[0, 'Baum', ['SDP Die Nacht von Freitag auf Montag', 'Ingo ohne Flamingo Hartz 4 und der Tag gehört dir', 'Scooter Bora! Bora! Bora!', 'K.I.Z Boom Boom Boom', 'Alligatoah Trostpreis (feat. Timi Hendrix)', 'Axwell /\\ Ingrosso Sun Is Shining', "Jetta I'd Love To Change The World - Matstubs Remix", "Genetikk Lieb's oder lass es", 'Alligatoah Willst du', 'K.I.Z Käfigbett', 'Eminem Survival', 'Bonez MC Palmen aus Plastik', 'Linkin Park Numb', 'K.I.Z Ariane', 'Bonez MC Ohne mein Team', 'Kontra K Erfolg ist kein Glück', 'Scooter Oi', 'Trailerpark Endlich normale Leute', '257ers Holland', "AronChupa I'm an Albatraoz", 'Ran-D Zombie', 'Trailerpark Selbstbefriedigung', 'Eminem Berzerk', 'K.I.Z Urlaub fürs Gehirn', 'Scooter One (Always Hardcore) - Remastered', 'Kraftklub Dein Lied', 'K.I.Z Wir', 'Kraftklub Schüsse in die Luft', 'Raggabund Ganjatherapie', 'SDP Deine Freundin', '257ers Lisa', 'Honk Hallo Helmut (Andere Farbe)', 'SDP #DAMDAMDAM!', 'Die Orsons Schwung in die Kiste', 'Scooter The Logical Song - Radio Edit', 'Honk Hallo I Bims 1 Bätmän', 'Trailerpark Sexualethisch desorientiert', 'Imagine Dragons Friction', 'Portugal. The Man Feel It Still', 'DJ Antoine House Party - Radio Edit', 'Die Orsons Ventilator', 'ItaloBrothers Stamp On The Ground', 'Alligatoah Fick ihn doch', "Snoop Dogg [... cut ...]
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
[014c86b0] pulse audio output error: PulseAudio server connection failure: Connection refused
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
``` And playlist baum works and the assistant is playing the music

shivasiddharth, 2018-10-09:
https://github.com/shivasiddharth/GassistPi/blob/0938b39cb2e30d72522575f68f96d7603fb589a4/src/actions.py#L1326

This does not print the chosen playlist. It prints all the playlists that you have created. If it prints baum only then it means that it seeing only that playlist. Since the fundamental issue has been resolved, I am closing this thread.

> Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth

These are just warnings, dont worry about these.

closed 2018-10-09 by shivasiddharth