shivasiddharth/GassistPi #566: Multiple gassitpi in the same piece

bimbim2401, 2018-10-10:
When I launch multiple raspberry with gassitpi, they all respond. Do you think it can be possible to manage the question only by one gassitpi?

shivasiddharth, 2018-10-10:
I have added mute buton. You need to mute the others that you are not using. Please use gitter for general queries.

closed 2018-10-10 by shivasiddharth

bimbim2401, 2018-10-10:
I do not ask for a mute button 

All the gassistpi are available but only one respond

shivasiddharth, 2018-10-10:
When ever two or more assistants are triggered simultaneously only one will respond. That is how google assistant works. If that is what you meant.

bimbim2401, 2018-10-10:
Thanks but I know how google assistant work but gassitpi do not do the same. All my gassistpi respond and not exactly in the same time this is my problem. I want the same comportement like google home

shivasiddharth, 2018-10-10:
They will not respond at the same time..

bimbim2401, 2018-10-10:
I want only one to respond and the problem is not all à the same time or not a the same time.