shivasiddharth/GassistPi #568: Custom Wakeword

bancroftd, 2018-10-11:
I see custom wakeword was added back into the project. I can't seem to get this to work. I install stretch, did all updates, installed GassistPi as per readme. Both "OkGoogle" and "Custom Wakeword" are enabled by default. still not working. I tried, snoyboy, alexa, sam, none work. 

Any Ideas? 

BTW thanks Shivasiddarth, this project is amazing!! I use it to control my house and media center, I have 4 raspberry PI's and about 30 SONOFF switches. When I had surgery, it was great to not need to move at all to turn things on and off without moving! Having Kodi now using voice is great too now :)

serhatkimya, 2018-10-14:
I tried to custom wakeword but its not usefull like hey google. It doesnt understand perfectly. I searched why , because of .umdl is better than the .pmdl. is there anyone experiences using about custom wakeword.

shivasiddharth, 2018-12-16:
Closing this for inactivity.

closed 2018-12-16 by shivasiddharth