shivasiddharth/GassistPi #578: something went wrong/there was a glitch

trex78, 2018-10-15:
Hi, recently installed GassistPi on my raspberry pi 2. Since first boot the only response i'm getting is 'something went wrong/there was a glitch'. I checked the known issues and googled around, most cases come down with the following causes:
1) A netwerk issue, my pi is connected to lan and currently it's the only device active on the network. I also have the same issue when i connect to Wi-Fi and rebooted router
2) 500 wakewords limit, i got the error right after installation so i don't think that's the cause
3) A problem with google itself, can't check that one

I installed the google assistant to my son's android tablet and there i'm getting the error that google assistant isn't available in my country so i'm thinking that could cause the issue?
If so, is there a way to fix it? Many thanks.

shivasiddharth, 2018-10-15:
Its first time i am coming across something like this. Dont know if anyone from unsupported regions ahve successfully installed this project. To get Google Assistant on tablet or phone, you need to set the system language to English US. After setting the language to English US, restart your device and you should have the google assistant. may be then you can give the project a shot after that.

shivasiddharth, 2018-12-02:
Do we have a status on this issue ?

shivasiddharth, 2018-12-16:
Closing this for inactivity.

closed 2018-12-16 by shivasiddharth