shivasiddharth/GassistPi #579: Feature added, Domoticz control on GassistPi

Hoeby, 2018-10-15:
I made a backup image from it and just tried the adding the commandos
I added "open", "close" and "stop".
They all work. Only with "stop" you need to give the speaking commando fast, otherwise Google thinks you want to stop the "OK Google" or "Hey Google" command you have given.
It is correct that "open" sends an "off" command to domoticz and "close" send an "on" command.
I tested it whit a "Venetian Blinds EU" in domoticz. This works inverted. When normal blinds are used, you can change the json string, or make a blocky whit dummy switch in domoticz and inverted it there.
The changed domoticz script in actions.py which i modified
```
def domoticz_control(query,index,devicename):
    global hexcolour,bright,devorder
    try:
        for j in range(0,len(domoticz_devices['result'])):
            if domoticz_devices['result'][j]['idx']==index:
                devorder=j
                break
        if ' on ' in query or ' on' in query or 'on ' in query:
            devreq=requests.head("https://" + configuration['Domoticz']['Server_IP'][0] + ":" + configuration['Domoticz']['Server_port'][0] + "/json.htm?type=command&param=switchlight&idx=" + index + "&switchcmd=On",verify=False)
            say('Turning on ' + devicename + ' .')
        if ' open ' in query or ' open' in query or 'open ' in query:
            devreq=requests.head("https://" + configuration['Domoticz']['Server_IP'][0] + ":" + configuration['Domoticz']['Server_port'][0] + "/json.htm?type=command&param=switchlight&idx=" + index + "&switchcmd=Off",verify=False)
            say('Opening ' + devicename + ' .')
        if 'off' in query:
            devreq=requests.head("https://" + configuration['Domoticz']['Server_IP'][0] + ":" + configuration['Domoticz']['Server_port'][0] + "/json.htm?type=command&param=switchlight&idx=" + index + "&switchcmd=Off",verify=False)
            say('Turning off ' + devicename + ' .')
        if ' close ' in query or ' close' in query or 'close ' in query:
            devreq=requests.head("https://" + configuration['Domoticz']['Server_IP'][0] + ":" + configuration['Domoticz']['Server_port'][0] + "/json.htm?type=command&param=switchlight&idx=" + index + "&switchcmd=On",verify=False)
            say('Closing ' + devicename + ' .')
        if ' stop ' in query or ' stop' in query or 'stop ' in query:
            devreq=requests.head("https://" + configuration['Domoticz']['Server_IP'][0] + ":" + configuration['Domoticz']['Server_port'][0] + "/json.htm?type=command&param=switchlight&idx=" + index + "&switchcmd=Stop",verify=False)
            say('Stopping ' + devicename + ' .')
	if 'toggle' in query:
            devreq=requests.head("https://" + configuration['Domoticz']['Server_IP'][0] + ":" + configuration['Domoticz']['Server_port'][0] + "/json.htm?type=command&param=switchlight&idx=" + index + "&switchcmd=Toggle",verify=False)
            say('Toggling ' + devicename + ' .')
        if 'colour' in query:
            if 'RGB' in domoticz_devices['result'][devorder]['SubType']:
                rcolour,gcolour,bcolour,hexcolour,colour=getcolours(query)
                hexcolour=hexcolour.replace("#","",1)
                hexcolour=hexcolour.strip()
                print(hexcolour)
                if bright=='':
                    bright=str(domoticz_devices['result'][devorder]['Level'])
                devreq=requests.head("https://" + configuration['Domoticz']['Server_IP'][0] + ":" + configuration['Domoticz']['Server_port'][0] + "/json.htm?type=command&param=setcolbrightnessvalue&idx=" + index + "&hex=" + hexcolour + "&brightness=" + bright + "&iswhite=false",verify=False)
                say('Setting ' + devicename + ' to ' + colour + ' .')
            else:
                say('The requested light is not a colour bulb')
        if 'brightness' in query:
            if domoticz_devices['result'][devorder]['HaveDimmer']:
                if 'hundred' in query or 'hundred'.lower() in query or 'maximum' in query:
                    bright=str(100)
                elif 'zero' in query or 'minimum' in query:
                    bright=str(0)
                else:
                    bright=re.findall('\d+', query)
                    bright=bright[0]
                devreq=requests.head("https://" + configuration['Domoticz']['Server_IP'][0] + ":" + configuration['Domoticz']['Server_port'][0] + "/json.htm?type=command&param=switchlight&idx=" + index + "&switchcmd=Set%20Level&level=" + bright ,verify=False)
                say('Setting ' + devicename + ' brightness to ' + str(bright) + ' percent.')
            else:
                say('The requested light does not have a dimer')

    except (requests.exceptions.ConnectionError,TypeError) as errors:
        if str(errors)=="'NoneType' object is not iterable":
            print("Type Error")
        else:
            say("Device or Domoticz server is not online")
#------------------------End of Domoticz Control Functions----------------------
```

shivasiddharth, 2018-10-15:
Ok, thanks. Stop command is for stopping music so it needs to be changed to Halt. I will take care of that.

Hoeby, 2018-10-20:
because of the new version, the programming looks different.
I updated, but couldn't copy/paste from a backup. Because of the custom function.
The commands OPEN, CLOSE, HALT (stop) where not added in the new version. So modified them so it works

Modified **keywords_en.yaml**, and added these 3 words unter DICT
Sorry, only modified that one, my languages are not that great to modify the others to.
```
'Open': 'open'
'Close': 'close'
'Halt': 'halt'
```
Modified actions.py, and added these codes
```
if (' ' + custom_action_keyword['Dict']['Open'] + ' ') in query or (' ' + custom_action_keyword['Dict']['Open']) in query or (custom_action_keyword['Dict']['Open'] + ' ') in query:
            devreq=requests.head("https://" + configuration['Domoticz']['Server_IP'][0] + ":" + configuration['Domoticz']['Server_port'][0] + "/json.htm?type=command&param=switchlight&idx=" + index + "&switchcmd=Off",verify=False)
            say('Opening ' + devicename + ' .')

```
```
if (' ' + custom_action_keyword['Dict']['Close'] + ' ') in query or (' ' + custom_action_keyword['Dict']['Close']) in query or (custom_action_keyword['Dict']['Close'] + ' ') in query:
            devreq=requests.head("https://" + configuration['Domoticz']['Server_IP'][0] + ":" + configuration['Domoticz']['Server_port'][0] + "/json.htm?type=command&param=switchlight&idx=" + index + "&switchcmd=On",verify=False)
            say('Closing ' + devicename + ' .')
```
```
if (' ' + custom_action_keyword['Dict']['Halt'] + ' ') in query or (' ' + custom_action_keyword['Dict']['Halt']) in query or (custom_action_keyword['Dict']['Halt'] + ' ') in query:
            devreq=requests.head("https://" + configuration['Domoticz']['Server_IP'][0] + ":" + configuration['Domoticz']['Server_port'][0] + "/json.htm?type=command&param=switchlight&idx=" + index + "&switchcmd=Stop",verify=False)
            say('Stopping ' + devicename + ' .')
```

They are tested and work nicely.
I am working whit Domoticz Stable V4.9700

Hoeby, 2018-10-20:
A small change.
I noticed that everything GassistPi says was englisch, also when setting up an other language.
First i thoughed it was a google setting. But thoughed twice about it, it is because of the "say" command. Therefor i changed it. This is only for the domoticz part i have done. 
It is an ugly fix. Don't know how to change it in the say command.
A nix fix would be, but this doesn't work:  `say(['Dict']['Turning'] + devicenamen + ' . ')`

I made it like this, unfortunally only for englisch or dutch. @shivasiddharth, do you know a nice way to fix it?

```
if (' ' + custom_action_keyword['Dict']['On'] + ' ') in query or (' ' + custom_action_keyword['Dict']['On']) in query or (custom_action_keyword['Dict']['On'] + ' ') in query:
            devreq=requests.head("https://" + configuration['Domoticz']['Server_IP'][0] + ":" + configuration['Domoticz']['Server_port'][0] + "/json.htm?type=command&param=switchlight&idx=" + index + "&switchcmd=On",verify=False)
            if configuration['Language']['Choice']=='en':
                say('Turning on ' + devicename + ' .')
            elif configuration['Language']['Choice']=='nl':
                say('Schakel ' + devicename + ' aan.')
            else:
                say('Turning on' + devicename + ' .') 
```

shivasiddharth, 2018-10-20:
If you change the language option in config.yaml, it will change the language for the say function. say function goes through google translate. So turning will be autopmatically changed to the language that you want.