shivasiddharth/GassistPi #583: GassistPi gpio control on windows

bimbim2401, 2018-10-16:
hello,

i saw your tutorial for installing gassistpi on windows with virtual box but do you thing there is a workaround for activating the neopixel or something that can do the same job ?

Thanks

bimbim2401, 2018-10-16:
ok i find a way to control an arduino with windows using serial usb 
https://www.makeuseof.com/tag/program-control-arduino-python/

so now i have another question, do you think that i can adapt your project without using virtual box like in this link 
https://troubleshooter.xyz/wiki/how-to-install-google-assistant-on-windows-10/ 
i want to keep the spotify access for example

shivasiddharth, 2018-11-11:
Custom features will not work in the second link that you gave. If you want the custom features, then you need to use the virtualbox.

shivasiddharth, 2018-12-16:
Closing this for inactivity.

closed 2018-12-16 by shivasiddharth