shivasiddharth/GassistPi #592: GassistPi not opening with respeaker 4mic

sameergovekar, 2018-10-20:
### Describe the bug
I am trying to install the GassistPi along with respeaker hat with 4 mic options

The hardware seems to work when i do a arecord with simple record commands 
but when i try to start the assistant either automatic or manual it gives below issue
pi@raspberrypi:~/GassistPi $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id RPIGOOGLE1
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 31, in <module>
    import snowboydecoder
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 5, in <module>
    import snowboydetect
  File "/home/pi/GassistPi/src/snowboydetect.py", line 18, in <module>
    _snowboydetect = swig_import_helper()
  File "/home/pi/GassistPi/src/snowboydetect.py", line 17, in swig_import_helper
    return importlib.import_module('_snowboydetect')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: dynamic module does not define init function (init_snowboydetect)

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

## If answer to question 2 is Yes, then paste the contents of the terminal below.
pi@raspberrypi:~/GassistPi $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id RPIGOOGLE1
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 31, in <module>
    import snowboydecoder
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 5, in <module>
    import snowboydetect
  File "/home/pi/GassistPi/src/snowboydetect.py", line 18, in <module>
    _snowboydetect = swig_import_helper()
  File "/home/pi/GassistPi/src/snowboydetect.py", line 17, in swig_import_helper
    return importlib.import_module('_snowboydetect')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: dynamic module does not define init function (init_snowboydetect)

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-10-20:
Which version of the OS are you running and did you setup the Respeaker HAT using my instructions or seeedstudio's ?

sameergovekar, 2018-10-21:
hi
i am using Raspbian , i set up with your instructions as well as the seeedstudio  , after a lot of struggle it worked , but some how now the Google developers account my device registration is failing , i tried to do everything from scratch again. thanks for the response , if any issue comes , i shall revert.

sameergovekar, 2018-10-21:
hi,

i tried the setup with the seeed studio alexa portion which seems to be working , but with your setup , it still shows either segmention falut or if i try to run it manually it does not start gives the same issue 

Traceback (most recent call last):
File "/home/pi/GassistPi/src/main.py", line 31, in 
import snowboydecoder
File "/home/pi/GassistPi/src/snowboydecoder.py", line 5, in 
import snowboydetect
File "/home/pi/GassistPi/src/snowboydetect.py", line 18, in 
_snowboydetect = swig_import_helper()
File "/home/pi/GassistPi/src/snowboydetect.py", line 17, in swig_import_helper
return importlib.import_module('_snowboydetect')
File "/usr/lib/python2.7/importlib/init.py", line 37, in import_module
import(name)
ImportError: dynamic module does not define init function (init_snowboydetect)

sameergovekar, 2018-10-21:
Did everything again it worked

closed 2018-10-21 by sameergovekar

ParkChur, 2019-07-19:
Do you solve this problem?? 
I face same problem..!