shivasiddharth/GassistPi #593: Segmentation Fault

HWL1968, 2018-10-21:
Hello all,

I tried to install Gassist Pi onto a RPi2 (using the RPi3 scripts). After a while (and installing all kind of dependencies) I get:
: (lots of messages about updating ..)
credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Testing the installed google assistant. Make a note of the generated Device-Id
device_model_id: (my model ID)
device_id: (my device ID)

And then (unfortunately):
./GassistPi/scripts/gassist-installer-pi3.sh: line 59:  5554 Segmentation fault      googlesamples-assistant-hotword --project_id $projid --device_model_id $modelid

I did try several things to solve:
- brand new SD card
- new installation of Raspbian
- several times a restart and afterwards tried to start the script again

No luck, doesn't get any further then this.

How to solve this?
My system: RPi2, 8GB SD, enough free space (df shows:
pi@raspberrypi:~ $ df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        7548460 3795008   3390804  53% /
devtmpfs          470116       0    470116   0% /dev
tmpfs             474724       0    474724   0% /dev/shm
tmpfs             474724   12288    462436   3% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             474724       0    474724   0% /sys/fs/cgroup
/dev/mmcblk0p1     44144   22502     21642  51% /boot
tmpfs              94944       0     94944   0% /run/user/1000
)

Please help.

Best regards.

sameergovekar, 2018-10-21:
Use rpi3 it's better ,

shivasiddharth, 2018-10-21:
There are a number of reasons for segmentation fault. Already discussed in #366 #369 #392 #531 #542

closed 2018-10-21 by shivasiddharth