shivasiddharth/GassistPi #595: Pir motion help

serhatkimya, 2018-10-22:
Anyone help me about my project?
I try to integrated pir sensor and when i activated assistank like hotword 'security mode on' and then if pir detect motion, assistan says to me motion detected. My problem pir motion code in have to be in loop because when security mode on, and so in terminal no output display without sensor activity, thats why assistant can't get my order because of in loop and i can not break loop.

serhatkimya, 2018-10-22:
Sorry, not assistant say to me motion detected, google translate say to me. I only want to use assistant like magicmirror in this main.py code. I tell to assistant and then pir code will start. I need interrupt code. Thanks everyone

shivasiddharth, 2018-10-23:
Please paste the code here and close the issue.

serhatkimya, 2018-11-02:
if 'security'.lower() in str(usrcmd).lower():
                    assistant.stop_conversation()
                    scommand=str(usrcmd).lower()
                    if 'on'.lower() in scommand:
                        say('Security mode activated')
                        def motion(pirpin):
                            if GPIO.input(pirpin):
                                say('someone is walking')
                        GPIO.add_event_detect(pirpin, GPIO.RISING,callback=moti$
                        while GPIO.event_detected(pirpin):
                            time.sleep(0.1)
                    if 'off'.lower() in scommand:
                        GPIO.remove_event_detect(pirpin)
                        say('Security canceled')

closed 2018-11-02 by serhatkimya