shivasiddharth/GassistPi #596: google translate problem again

BarryAlvord, 2018-10-22:
My terminal output:
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED
INFO:snowboy:Keyword 1 detected at time: 2018-10-22 12:52:18
Assistant is listening....
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "who are you"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "who are you"}
I am not the one you expect to be
ERROR:__main__:name 'Keyerror' is not defined
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 463, in main
    say(selectedans[0])
  File "/home/pi/GassistPi/src/actions.py", line 221, in say
    tts.save(ttsfilename)
  File "/home/pi/env/lib/python3.5/site-packages/gtts/tts.py", line 247, in save
    self.write_to_fp(f)
  File "/home/pi/env/lib/python3.5/site-packages/gtts/tts.py", line 187, in write_to_fp
    part_tk = self.token.calculate_token(part)
  File "/home/pi/env/lib/python3.5/site-packages/gtts_token/gtts_token.py", line 28, in calculate_token
    seed = self._get_token_key()
  File "/home/pi/env/lib/python3.5/site-packages/gtts_token/gtts_token.py", line 61, in _get_token_key
    result = re.search("\d{6}\.[0-9]+", tkk_expr).group(0)
NameError: name 'tkk_expr' is not defined

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 666, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 465, in main
    except Keyerror:
NameError: name 'Keyerror' is not defined

Please help me
P/S i hve tried the old way that u gave me 1 month ago but it dont work right.

shivasiddharth, 2018-11-11:
Since this issue was opened, no one else reported any problems with the text to speech. Are you still stuck ?

majorroli, 2018-11-19:
very similar issue here in Germany (i did not change the language in the config.yaml though)

device_model_id: tacteam-222322-tacteam-ouwe93
device_id: E0EF8F1BD91AE0B6DF3F8DE4C77CA9F0

ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "how are you"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "how are you"}
ON_RESPONDING_STARTED:
  {"is_error_response": false}
ON_RESPONDING_STARTED:
  {"is_error_response": false}
ON_RESPONDING_FINISHED
ON_RESPONDING_FINISHED
ON_RENDER_RESPONSE:
{
  "text": "I'm doing great, thanks for asking. What can I help you with?",
  "type": 0
}
ON_RENDER_RESPONSE:
{
  "text": "I'm doing great, thanks for asking. What can I help you with?",
  "type": 0
}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": true}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": true}
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "hey Google who are you"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "hey Google who are you"}
ERROR:__main__:name 'Keyerror' is not defined
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 470, in main
    say(selectedans[0])
  File "/home/pi/GassistPi/src/actions.py", line 219, in say
    newword=trans(words,language)
  File "/home/pi/GassistPi/src/actions.py", line 210, in trans
    transword= translator.translate(words, dest=lang)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 172, in translate
    data = self._translate(text, dest, src)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 75, in _translate
    token = self.token_acquirer.do(text)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
    self._update()
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
    code = unicode(self.RE_TKK.search(r.text).group(1)).replace('var ', '')
AttributeError: 'NoneType' object has no attribute 'group'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 673, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 472, in main
    except Keyerror:
NameError: name 'Keyerror' is not defined

shivasiddharth, 2018-11-19:
@majorroli  Your error is not the same as the ones previously posted. 
Already sevral of us have gone through the steps to fix the token errors for both gtts and google translate in #534 . It has also been updated with latest instructions.

closed 2018-11-19 by shivasiddharth