shivasiddharth/GassistPi #597: Unify volume between "standard" GA voice and TTS

BriKKou, 2018-10-25:
Hi,
First of all, thank you this very cool project. I just installed it (totaly newby) and achieve to connect my Sonoff-Tasmota (had some trouble to make it work, but finally works).
**My probleme at this time is related to the volume.**
When the PI start, startup.wav volum is very high (I edited it with Audacity to make it less louder) even if I seted up GA volume to 40%.
Same problème arise when I ask to my sonoff to turn on/off, the TSS speak as louder as the wav files  (function say() with say.mp3?).
So my guess is that - for the moment - their is no relation between the GA sound and the PI overall sound.
Is that a bug ? Is that in the pipe ?
Thanks for your advice and help.
Brice

**My conf : PI3 + AYI v1 + Sketch fresh install + Using GA in french**

BriKKou, 2018-10-26:
May be linked to? https://github.com/google/aiyprojects-raspbian/issues/325

shivasiddharth, 2018-10-26:
Right now its not possible. Audio handlers for the hotword have not been exposed. GA volume is independent of Pi's.

BriKKou, 2018-10-26:
OK, thanks for your answer.
Would it be possible to add an option in the config file to manually set the volume of those files.
I mean, do not toych the PI main volume, only set the volume of TTS generated sound and wav ?
Thanks

shivasiddharth, 2018-10-26:
We may have to change the player for that. Can be done but.  What actually can be done is, you need to give a command to set google assistant volume to maximum. From then on, using plugins for the alsa, we can normalize all the volumes. People have done it using LADSPA. It will take some time for me to check its working for each and every audio config that i am offering here. I will add an enhancement label. But still i am not sure as to what will happen if you lower any of the volumes (vlc or GA).

BriKKou, 2018-10-26:
Thanks for your reply.
I use the following, and it's okay for me.
`amixer sset 'Master' 65%`