shivasiddharth/GassistPi #599: Request : Trigger custom scripts by Voice control

BriKKou, 2018-10-26:
Hi,
I have some bluetooth light bulbs and I use a python script (found on the Internet) to change theire colors and intensities.
It will be awesome to say "Ok Noodle, set the living room light to maximum".

As is does not appears to be possible at the moment, here is my tought on how to achieve this with extra flexibility for other kind of scripts (I guess) :

(Sorry, but I'm totally ignorante on how works YAML...)

**In config.yaml : Create a new option's "section"** 
```
User_scripts:
  script_ID:
    - '1'
    - '2'
  friendly-names:
    - 'Living room light'
    - 'bedroom light'
  script-path:
    - '/home/${USER}/scripts/lightbulb.py'
    - '/home/${USER}/scripts/lightbulb.py'
  voice_command:
    - '1'
        - 'to maximum' 
        - 'to minimum' 
    - '2'
        - 'to maximum' 
        - 'to minimum' 
  scripts_command:
    - '1'
        - 'FF:BB:CC:11:22:33 set brightness to 100%' 
        - 'FF:BB:CC:11:22:33 set brightness to 10%' 
    - '2'
        - '11:22:33:FF:BB:CC set brightness to 100%' 
        - '11:22:33:FF:BB:CC set brightness to 10%' 
```

**In actions.py : Create a new generic function** 
This generic function will handle the command and trigger the action to the desired script. 

As I don't know anything about python nor YAML, I don't know how to achieve this, but I hope you'll consider my request and if so I would love to help you (even if I don't know how to).

Thanks.

Brice

shivasiddharth, 2018-10-26:
Can you paste the contents of the lightbulb.py script here.

shivasiddharth, 2018-10-26:
So how do you normally send commands using this script?

BriKKou, 2018-10-26:
Like this for example:
`sudo -u root python2 /home/pi/Bulb/lightbulb.py AA:BB:5E:A9:17:DC set brightness to 100%
`
In Jeedom, I used to use it like this
[image]

What do you think of the idea ?

BriKKou, 2018-10-27:
Hey!
After few hours of testing, I finally achieve what I wanted to do :)
That's the second time I'm using Python and the first time dealing with yaml, so be indulgent.

At the end of **config.yaml**
```
#Custom script launcher
User_scripts:
  #Assign the device ids in the same order as the devices listed above
  #Use {PATH} in your commands to use the script-path to your file
  #Use '' (two single) to escape single quotes in your commands
  Id:
    - '1'
    - '2'
    - '3'
  friendly-names:
    - 'simba'
    - 'pocahontas'
    - 'raspi'
  script-path:
    - '/home/pi/scripts/beewi.py'
    - '/home/pi/scripts/beewi.py'
    - ''
  command:
    1:
        - voice: 'maximum'
          command: 'sudo -u root python2 {PATH} F4:B8:5E:A9:17:DC bright'
        - voice: '50%'
          command: 'sudo -u root python2 {PATH} F4:B8:5E:A9:17:DC medium'
        - voice: 'minimum'
          command: 'sudo -u root python2 {PATH} F4:B8:5E:A9:17:DC dim'
    2:
        - voice: 'maximum'
          command: 'sudo -u root python2 {PATH} F4:B8:5E:64:04:6A bright'
        - voice: 'minimum'
          command: 'sudo -u root python2 {PATH} F4:B8:5E:64:04:6A dim'
    3:
        - voice: 'maximum'
          command: 'amixer sset ''Master'' 100%'
        - voice: 'minimum'
          command: 'amixer sset ''Master'' 65%'
```

After #Sonoff-Tasmota Declarations declation **in main.py**
```
#Custom script launcher
#Make sure that the device name assigned here does not overlap any of your smart device names in the google home app
uscripts_id=configuration['User_scripts']['Id']
uscripts_names=configuration['User_scripts']['friendly-names']
uscripts_path=configuration['User_scripts']['script-path']
uscripts_cmd=configuration['User_scripts']['command']
```

After tasmota_control() declation in  **in main.py**
```
#Function to launch user scripts
def scripts_launcher(phrase,devname,id,path,cmd):
    try:
        if devname in phrase:
            for key, objID in cmd.items():
                if key == id:
                    for cond in objID:
                        if cond.get('voice') in phrase:
                            print (cond.get('command'))
                            os.system(cond.get('command').replace('{PATH}', path))
                            say("Done on "+devname)
                         

    except requests.exceptions.ConnectionError:
        say("Oops there is a problem")
```
After tasmota_control() call in  **in main.py**
```
                for num, name in enumerate(uscripts_names):
                    if name.lower() in str(usrcmd).lower():
                        assistant.stop_conversation()
                        scripts_launcher(str(usrcmd).lower(), name.lower(),int(uscripts_id[num]),uscripts_path[num],uscripts_cmd)
                        break
```

At the moment I can control my two bluetooth light bulbs and the main volume of the raspberry.

As I do not know python or yaml at all, I think my code is very perfectible (security issue!?).
If you decide to include this new feature I think this will open many opportunities and make this projet even greater!

Brice

jeffdeville, 2018-10-27:
Device Actions is the thing you want to use here.
https://developers.google.com/assistant/sdk/device-actions-overview

Once you've set that up (which admittedly is a bit of a PITA on the google side, right now), you can do something like:

if event.type == EventType.ON_DEVICE_ACTION:
            for command, params in event.actions:
                print('Do command', command, 'with params', str(params))
                if command == 'com.example.commands.AwardPoints':
                    award(params["house"], params["number"])

The advantage is that you can declare multiple phrases and handle input types / ranges for each of your devices in a unified way.

BriKKou, 2018-10-28:
Hi Jeff,
Thanks for this tip.
I rapidly took a look, and as you adviced, this is really unclear on how to use (even more to me as I'm totally not familiare with this).
For the moment, what I did feats my "tinny" needs, but I'll surely investigate more on this later.

shivasiddharth, 2018-12-02:
Closing this for inactivity.

closed 2018-12-02 by shivasiddharth