shivasiddharth/GassistPi #601: USB microphone isn't working

gopalasubramanium, 2018-10-27:
### Describe the bug:
```
Speaker is fine but Mic isn't working.

I used this option 3.3. USB MIC AND HDMI users, but I'm getting errors:

pi@smartmirror:~ $ arecord -D hw:1,0 -d 5 -f cd test.wav -c 1
arecord: main:788: audio open error: Device or resource busy

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```
pi@smartmirror:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 6/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@smartmirror:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [(LCS) USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
pi@smartmirror:~ $ arecord -D hw:1,0 -d 5 -f cd test.wav -c 1
arecord: main:788: audio open error: Device or resource busy
pi@smartmirror:~ $

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```
Not reached this stage.
```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
Not reached this stage.
```

### Paste the contents of the terminal below
```
Not reached this stage.
```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-10-27:
Please do not open multiple issues for the same purpose. Closing #600

shivasiddharth, 2018-10-27:
I am not sure if something is running in the background. I take here issues pertaining to the project only. Try this:
`arecord -d 10 -D hw:1,0 -r 16000 -f S16_LE -c1 /home/pi/mic-test.wav`

gopalasubramanium, 2018-10-27:
> Please do not open multiple issues for the same purpose. Closing #600

Sincere apologies, that was by mistake.

gopalasubramanium, 2018-10-27:
> I am not sure if something is running in the background. I take here issues pertaining to the project only. Try this:
> `arecord -d 10 -D hw:1,0 -r 16000 -f S16_LE -c1 /home/pi/mic-test.wav`

pi@smartmirror:~ $ arecord -d 10 -D hw:1,0 -r 16000 -f S16_LE -c1 /home/pi/mic-test.wav
arecord: main:788: audio open error: Device or resource busy

shivasiddharth, 2018-10-27:
Are you running something already as a service or in the background ? Try opening it as a plug

`arecord -d 10 -D plughw:1,0 -r 16000 -f S16_LE -c1 /home/pi/mic-test.wav`

gopalasubramanium, 2018-10-27:
> Are you running something already as a service or in the background ? Try opening it as a plug
> 
> `arecord -d 10 -D plughw:1,0 -r 16000 -f S16_LE -c1 /home/pi/mic-test.wav`

I have MagicMirror running on the same Pi. I stopped that and tried again but getting the same error :(

closed 2018-10-27 by gopalasubramanium

Tsury, 2019-01-27:
@gopalasubramanium How did you resolve this?