shivasiddharth/GassistPi #602: Gassistpi crash after youtube playback

sameergovekar, 2018-10-27:
hi

I am getting below error when i try to play any video using the voice command ok google play xyz on youtube , it plays the video but the program stops response , when i tried to see the log below error was shown

 File "/home/pi/GassistPi/src/actions.py", line 1018, in YouTube_No_Autoplay
    say("Getting youtube link")
  File "/home/pi/GassistPi/src/actions.py", line 219, in say
    newword=trans(words,language)
  File "/home/pi/GassistPi/src/actions.py", line 210, in trans
    transword= translator.translate(words, dest=lang)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 172, in translate
    data = self._translate(text, dest, src)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 75, in _translate
    token = self.token_acquirer.do(text)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
    self._update()
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
    code = unicode(self.RE_TKK.search(r.text).group(1)).replace('var ', '')
AttributeError: 'NoneType' object has no attribute 'group'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 666, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 523, in main
    YouTube_No_Autoplay(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1031, in YouTube_No_Autoplay
    say('It is no longer available in youtube')
  File "/home/pi/GassistPi/src/actions.py", line 219, in say
    newword=trans(words,language)
  File "/home/pi/GassistPi/src/actions.py", line 210, in trans
    transword= translator.translate(words, dest=lang)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 172, in translate
    data = self._translate(text, dest, src)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 75, in _translate
    token = self.token_acquirer.do(text)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
    self._update()
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
    code = unicode(self.RE_TKK.search(r.text).group(1)).replace('var ', '')
AttributeError: 'NoneType' object has no attribute 'group'

shivasiddharth, 2018-10-27:
Duplicate of #534. Sticking to issue template would have automatically taken you there.

closed 2018-10-27 by shivasiddharth

sameergovekar, 2018-10-27:
thanks changed the gtoken.py , it is working now