shivasiddharth/GassistPi #605: receiving alertes and data from esp sensors or other and say it or perform an action

nclbill, 2018-10-29:
hi what about receiving alerts and data from esp sensors or others 
when sensor is activated send data to pi to perform an action
its possible for pi to check by sending request to server on esp but the idea is to make the pi a broker 
sorry for my bad english
tkx

NeoMod, 2018-10-29:
Technically the "broker part" is usually performed by a broker, like for example an MQTT Broker like Mosquitto, which works very well on raspberry and esp-based sensors.

In this case scenario, assuming that your sensor is publishing to the broker a status when activated (for example: a PIR detects movement), you might need a controller which subscribe to the same topic on the MQTT broker to "actually use" the data.

In a real case scenario, like this one, we could have and esp-based PIR-sensor that reports to Mosquitto and then a Controller running on raspberry who subscribes Mosquitto, reads the data and renders a visual confirmation (like a virtual notification toggle) of the movement reported by the PIR.

As for GassistPI, I'm not sure the Assistant can be "pinged" by data and then "advice" you about it (for example emitting a sound when movement is detected).

Anyway I would suggest using simpler methods to do so (like using mosquitto+domoticz integration, which is very easy to get started. and then maybe migrating to something more complex to have more logic conditions).

[ in this case, you can easily define a custom action to have Assistant read you parameters from a sensor, like a temp/hum sensory, by building a custom action that uses the domoticz api).

NeoMod, 2018-10-30:
> https://github.com/projetsdiy/esp8266-dht22-mqtt-home-assistant

I have no experience with Home Assistant as a controller, but I'm guessing it should expose some API to make a simple request and obtain sensor reading via JSON.

If that is the case you can write a custom action for GassistPI where a certain command "for example Get Room Temperature" would result in getting the data via API-to-HomeAssistant, parsing the JSON and passing the values to Google via the "say" command.

nclbill, 2018-10-31:
hi 
after making mosquitto working and connect to it from nodemcu(esp)  i can receive the values of the sensor on pi terminal : 
mosquitto_sub -h localhost -t "sensor/temperature"
where in the sketch of esp
#define temperature_topic "sensor/temperature"  //Topic température
.
.
.
void loop() {
.
.
client.publish(temperature_topic, String(t).c_str(), true); 
.
.

so first step is ok
------------------------------------------------------------------------ 
second step
to integrate that to the assistant "they work with home assistant" and they just add the following code to the configuration.yaml

mqtt: 
  broker: localhost 
  port: 1883        #par défaut 
  client_id: home-assistant-1 
  keepalive: 60
  username: USERNAME #optionnel 
  password: PASSWORD #optionnel 
  protocol: 3.1      #par défaut

# now add sensor and take the value from sensor/temperature.
sensor:
  platform: mqtt
  state_topic: "sensor/temperature"
  name: "Température"
  qos: 0
  unit_of_measurement: "°C"
  value_template: '{{ payload }}'

##################to switch on a led on the nodemcu
switch:&nbsp; 
  platform: mqtt
  name: "Cuisine"
  command_topic: "homeassistant/switch1" #Topic sur lequel on publie l'état de l'interrupteur
  payload_on: "ON" # A vous de choisir le message envoyé lorsque l'interrupteur est allumé&nbsp;
  payload_off: "OFF" # et éteint
  optimistic: true # Mettez à true pour maintenir l'état
  qos: 0
  retain: true
  value_template: '{{ value.x }}'
---------------------------------------------------------------------------------------------
is it possible to make some thing similar on GassistPI

NeoMod, 2018-10-31:
Ok, so first of all congratulations for your working sketch! 

Now you have a sensor that reports its values to an MQTT Broker.

**However, you are still missing one piece: a Controller.**

_GassistPI it’s not a controller_: this means it is not predisposed to “talk” with an MQTT Broker and “ask” him about the “values” from a sensor.
This is accomplished by the **Controller**, a piece of software that **“understands” the machine language** and hence is able to **“talk”** with the **MQTT Broker** asking for updates about sensors.  
Then **the Controller “beautify” that data in order to “translate” them** from machine language **into “human language”**: this is what happens, for example, in Domoticz or Home Assistant when you see a visualization of a sensor data output in the form of a graph for temperature or a percentage indicator for humidity.

Once you have the Controller of your choice set up and working _(and again, I would suggest starting with Domoticz because I thing all of this is fairly new to you)_ you can move to the last step: creating a custom action for GassistPI to read Sensor Data via the Controller.

The _**Custom Action**_ will follow this logic:

- [ ] When **_“TRIGGER WORD”_** detected 

- [ ] send **_GET Request_** to **Controller** to obtain Data for a specified sensor 

- [ ] the Controller will then **reply** with latest **data obtained trough MQTT** 

- [ ] the **response** will be, mostly, in the form of a _JSON_ so you need to **filter out specified values** from the JSON and store those values in a variable

- [ ] Then you can **pass** this variable to **_GassistPi_** in the form of a _SAY action_ and only then GassistPI will “READ” the VALUES for that sensor.

nclbill, 2018-11-01:
hi tks for all your help
I am making reserch and reading tutos on your advices tks for all 
---------------------------------------------------------------------------------
However,I misunderstand the custom action logic
When “TRIGGER WORD” detected : I must say trigger word to beging the process ?

im looking for logic action like this
# in esp
if data is true  send data 
# in PI
Wake gassistPI and say
----------------------------------------------------------------------
examples
Warning there is smoke on .....
or 
Warning there is not enough humidity

NeoMod, 2018-11-01:
You are welcome. I know it is a lot to take in, the first time it was the same for me: thousands of tutorials and a lot of try-and-error to understand how things work. 

As for the _“TRIGGER WORD”_, sorry I was not enough clear. 
The **trigger word** is a **word** or two, of your choice, **that you want to say in order to activate the process of reading sensor data**.

For example, you could use “_sensor_” and “_data_” as trigger words for your action, meaning that **when you say _“Hey Google, Sensor Data”_** the **GassistPI** will recognize the custom action and **will** **fetch the sensor data** (in the way discussed previously).

Now the crucial point is this: **you cannot wake the Assistant per Data Input.** It’s not supposed to work like that (yet).
**You have to “ask” the Assistant for something** in order to start any kind of action. That is why you need to write a “custom action” to recognize your specific request, and give the Assistant all the data he needs.

To clarify **let’s imagine** this:
-	**You are inside the house** and you can’t look outside
-	**Jack** is your friend, and he i### s near the door: he can’t look outside either, but he has a **cellphone**.
-	**Martin** is a common friend, and he **is outside** the house. He has a **cellphone** too, and he can’t enter the house. 

Now, **If you want to know what the weather** outside is **you can ask Jack** _(the Assistant)_ that in turn will use his cellphone to **call Martin** _(the Sensor)_ **and ask him about the weather**. When **Jack** has **received** the **information** from Martin _(that is also complaining because it’s freezing cold outside :-p )_ he will then **answer your question** _(the User)_ and tell you what temperature is outside, but he will omit (_filtering_) all of Martin complains for the situation.

Pardon me for the little theatrical example, but I hope this way it may be easier to figure out the logic of the entire process.

On the other hand, if you want to develop a _Warning System_ (like you mentioned in the last part) then you have to rely on other mechanics.
Luckily, **that is what a Controller is for**: a Controller can implement various logic and one of these is the “alert feature” that you want.

An _“Alert Feature_” works like this:

-	A parameter is periodically monitored, for example temperature
-	A set point is compared against the parameter, for example temperature <= 26°C
-	An IF – THEN cycle is executed to decide what action is needed:
o	If temperature <=26°C then “do nothing”
o	Else if temperature >26°C then “Sound The Alarm”

This is possible literarily  in a thousands of different ways: for example within _Domoticz_ you have a _Logic Builder_ that works perfectly within this circumstances and can send you alerts on your phone (for example on your e-mail or on Telegram).

From there on your imagination is the only limit: you can learn how to do complex things and can set-up complex alert systems. **However, this goes beyond the scope of GassistPI.**

shivasiddharth, 2018-11-02:
@NeoMod  Has given some excellent points above.  Let me add a little more to it.

1. You can use the google translate and text to speech converter with any script. It has no binding on the Google Assistant. Create your own script to get the temperature and use the IF criteria to generate your own conditions ans get an audio readout.
2. The hotword assistant has totally eclipsed the gRPC API and trust me gRPC has got a lot more potential for customization.  If you want a custom action to be taken by the google assistant, you can use the gRPC API. For example, if you want the assistant to get outside temperature when your thermostat temperature drops, use the pushtotalk or textinput samples. Both of these can take a pre-defined input unlike the library (hotword sample).

nclbill, 2018-11-02:
hi sir 
I got the idea and working on tks for you and for NeoMod
must i close the request or let it open for others ?

nclbill, 2019-09-24:
Hi Mrs can gassistpi work offline without google assistant just executing local commands like controlling home with esps.
(i already did the automatisations using esps, to control light, tv, ac, and publishing in mqtt locally from different sensors i used say query like you advised me and perform actions when conditions is true , now my assistant is ready .)
is it possible to run it locally without internet, just custom actions, perhaps later activate google assistant with a custom action and stop it with an other custom action leaving the custom actions always on. tks

closed 2019-09-24 by close-issue-app[bot]