shivasiddharth/GassistPi #606: autoplay links youtube gets the same link again and again

badsmoke, 2018-10-30:
### Describe the bug:
```
as soon as i look for a playlist on youtube on "autoplay", the same youtube link comes out again and again. The speech recognition recognizes me but reasonable.

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

    This device is not registered. This means you will not be able to use
    Device Actions or see your device in Assistant Settings. In order to
    register this device follow instructions at:

    https://developers.google.com/assistant/sdk/guides/library/python/embed/register-device

ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "autoplay und Spiele 90er Jahre Musik von YouTube"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "autoplay und Spiele 90er Jahre Musik von YouTube"}
stopping vlc
Autoplay-Links erhalten
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:02] Decoding of say.mp3 finished.
}
WARNING:googleapiclient.discovery_cache:file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 36, in autodetect
    from google.appengine.api import memcache
ImportError: No module named 'google.appengine'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ImportError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ImportError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
['jhuSF2WcpqE', 'i6RV8O4Q9rc', 'jAE5BPKDutk', 'Ue38v79x8d0', 'BtY5o3DBe1o']
Hinzufügen von Autoplay-Links zur Wiedergabeliste
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:04] Decoding of say.mp3 finished.
https://r6---sn-8xgn5uxa-cxge.googlevideo.com/videoplayback?mm=31%2C29&mn=sn-8xgn5uxa-cxge%2Csn-4g5e6ne6&id=o-AGr_FQkHcqDuREqD10HqxaqQxyM4nzMpgXm3xK8F9NTn&mime=audio%2Fwebm&pl=40&ipbits=0&ip=2001%3A16b8%3A55df%3A6100%3Adc67%3A14e6%3A86ad%3Aae00&requiressl=yes&mt=1540887699&mv=m&txp=5511222&gir=yes&ms=au%2Crdu&dur=720.381&initcwndbps=1292500&ei=BRXYW6_aMY_E-gaG6LjICA&expire=1540909413&sparams=clen%2Cdur%2Cei%2Cgir%2Cid%2Cinitcwndbps%2Cip%2Cipbits%2Citag%2Ckeepalive%2Clmt%2Cmime%2Cmm%2Cmn%2Cms%2Cmv%2Cpl%2Crequiressl%2Csource%2Cexpire&clen=11729578&source=youtube&signature=6BC842AD028092B6A6775EDB11B257F7F0D232D5.B15DFC37FFA276F34A5DE544104E3F315C535B5A&c=WEB&key=yt6&lmt=1540142567240243&keepalive=yes&itag=251&fvip=6&ratebypass=yes
PuTTY X11 proxy: unable to connect to forwarded X server: Network error: Connection refused
xcb_connection_has_error() returned true
[023a51e8] pulse audio output error: PulseAudio server connection failure: Connection refused
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

i would always like to play several songs via youtube, "autoplay" doesn't fit well into the command. Why not just change the if/else query

'''

 if (custom_action_keyword['Keywords']['YouTube_music_stream'][0]).lower() in str(usrcmd).lower() and 'kodi' not in str(usr$
                    assistant.stop_conversation()
                    vlcplayer.stop_vlc()
                    if 'autoplay'.lower() in str(usrcmd).lower():
                        YouTube_Autoplay(str(usrcmd).lower())
                    else:
                        YouTube_No_Autoplay(str(usrcmd).lower())
'''
to:
'''

 if (custom_action_keyword['Keywords']['YouTube_music_stream'][0]).lower() in str(usrcmd).lower() and 'kodi' not in str(usr$
                    assistant.stop_conversation()
                    vlcplayer.stop_vlc()
                    if 'autoplay'.lower() in str(usrcmd).lower():
                        YouTube_Autoplay(str(usrcmd).lower())
                    else:
                        YouTube_Autoplay(str(usrcmd).lower())
'''
so he always opens several links, but every time the same link comes out:

[0:04] Decoding of say.mp3 finished.
https://r6---sn-8xgn5uxa-cxge.googlevideo.com/videoplayback?lmt=1540142567240243&pcm2=no&dur=720.381&key=yt6&initcwndbps=1348750&ipbits=0&c=WEB&keepalive=yes&id=o-ACLPTw3MXhTMALgU5af0jeePRAC9uFEEBSuIMn1gyFhZ&ip=2001%3A16b8%3A55df%3A6100%3Adc67%3A14e6%3A86ad%3Aae00&gir=yes&signature=A846F085BD5D0F6326C0BAB4052BFDC755B8174C.258177206D07248CC14353FA4700285BEFE48C07&sparams=clen%2Cdur%2Cei%2Cgir%2Cid%2Cinitcwndbps%2Cip%2Cipbits%2Citag%2Ckeepalive%2Clmt%2Cmime%2Cmm%2Cmn%2Cms%2Cmv%2Cpcm2%2Cpl%2Crequiressl%2Csource%2Cexpire&fvip=6&txp=5511222&mime=audio%2Fwebm&itag=251&mt=1540887569&pl=40&mv=m&ei=dxTYW8mdJcvm1gL5toCoDQ&ms=au%2Crdu&mm=31%2C29&mn=sn-8xgn5uxa-cxge%2Csn-4g5e6ne6&clen=11729578&requiressl=yes&source=youtube&expire=1540909271&ratebypass=yes

where is the problem ?

NeoMod, 2018-10-31:
> ### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.
> ```
> This device is not registered. This means you will not [...]

First of all, judging by the log file it seems the device is not correctly registered.

You probably copied the wrong "Model ID" from Actions Console, please double check that.

Also, check that you have copied the full "Project ID" from Cloud Console and that you are entering the full name "name.json" for the credentials file.

shivasiddharth, 2018-11-01:
I have mentioned in the README as to how this autoplay works. Thats the reason why you are getting the same links.

badsmoke, 2018-11-05:
Thank you for your replies. But don't understand what you mean by mentioning how autoplay works. Is there a way that I always have autoplay active, but don't always get the same link?

shivasiddharth, 2018-11-05:
If you are looking for a playlist, then the syntax is different. Using this syntax the first song will always be "90er Jahre Musik" the autoplay kicks in only from the second track.

badsmoke, 2018-11-09:
@shivasiddharth 

i think i've put it wrong, the link i've received every time has nothing to do with the song i'm looking for. Even if I'm looking for completely different songs, the same link comes up again and again (you can open the link, it seems to be an audiobook).

shivasiddharth, 2018-12-02:
The song is played upon doing a search in Youtube using the words deciphered in the command and then first result is chosen and played. What appears as the first result is not decided by my scripts, but by the youtube search. Hope i cleared some things.

shivasiddharth, 2019-01-06:
Closing this for inactivity.

closed 2019-01-06 by shivasiddharth