shivasiddharth/GassistPi #607: Use of __USER__ in GassistPi/systemd/gassistpi-ok-google.service

cnoorman, 2018-10-30:
When installing the `gassistpi-ok-google.service` the variable `__USER__` is not translated (into pi). Consequently this service doesn't start.
Am I doing something wrong or should `__USER__` be replaced everywhere by `${USER}`  ??

cnoorman, 2018-10-30:
Same story for `Assistants-Pi/systemd/alexa.service` in the Assistants-Pi installation script.

shivasiddharth, 2018-11-01:
Issue seems to be isolated one. Its works for me and no one has reported issues with the service so far. Will leave the issue open to see if someone else reports the same thing.

cnoorman, 2018-11-01:
It's fine with me. I worked around it, but I cannot find any use of leading and trailing underscores in a bash script (only in Python). And you see a second example of the problem in the reference above.

cnoorman, 2018-11-04:
I changed `__USER__` to `%u`, but that translates into `root`.

shivasiddharth, 2018-11-04:
I tried the scripts in ubuntu and __USER__ got translated to the user name. I have changed __USER__ to ${USER} in master. can you check if it solved the issue?

cnoorman, 2018-11-05:
Unfortunately this does not solve the issue. $(USER) is not translated either. I understand what you want to accomplish, but pending the finale solution I suggest that you replace this just with `pi` in all `.service` files. At least that will work for 90% of the users.

commit 6de14f70 (2018-11-05): Fix autostart issue

Fixes #607
closed 2018-11-05 by shivasiddharth