shivasiddharth/GassistPi #612: say fuction error...

gudfkr123, 2018-11-04:
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "television"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "television"}
ERROR:__main__:'NoneType' object has no attribute 'group'
Traceback (most recent call last):
  File "main.py", line 670, in <module>
    Myassistant().main()
  File "main.py", line 445, in main
    TELEVISION(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 909, in TELEVISION
    say("OK! Turning on Televison!")
  File "/home/pi/GassistPi/src/actions.py", line 220, in say
    newword=trans(words,language)
  File "/home/pi/GassistPi/src/actions.py", line 211, in trans
    transword= translator.translate(words, dest=lang)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 172, in translate
    data = self._translate(text, dest, src)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/client.py", line 75, in _translate
    token = self.token_acquirer.do(text)
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 180, in do
    self._update()
  File "/home/pi/env/lib/python3.5/site-packages/googletrans/gtoken.py", line 59, in _update
    code = unicode(self.RE_TKK.search(r.text).group(1)).replace('var',' ')
AttributeError: 'NoneType' object has no attribute 'group'

what can i do.. ?

gudfkr123, 2018-11-04:
i just add 'def TELEVISION(phrase): say("ok turning on television")' on action.py. 
and add    
for event in events:
                self.process_event(event)
                usrcmd=event.args
                if 'television'.lower() in str(usrcmd).lower():
                    assistant.stop_conversation()
                    TELEVISION(str(usrcmd).lower())
on main.py

shivasiddharth, 2018-11-04:
duplicate of #534

closed 2018-11-04 by shivasiddharth