shivasiddharth/GassistPi #615: Issue on systemed service execution.

Vino-git, 2018-11-05:
### Describe the bug:
```
The google assistant bootable execute throw the below error. Please assist to fix the issue. Manual execution is working very well.

● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2018-11-05 12:05:30 +07; 11h ago
  Process: 17340 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id xxxxxxxxx (code=exited, status=210/CHROOT
 Main PID: 17340 (code=exited, status=210/CHROOT)

Nov 05 12:05:29 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 12:05:29 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 12:05:30 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 12:05:30 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 12:05:30 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
Nov 05 12:05:30 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
Nov 05 12:05:30 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 12:05:30 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
Followed the same steps as described in the auto-execute 
```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-11-05:
post the output of 
`journalctl -u gassistpi-ok-google.service`

Vino-git, 2018-11-06:
Here the output.

--------------------------------

-- Logs begin at Fri 2016-11-04 00:16:43 +07, end at Tue 2018-11-06 10:40:01 +07. --
Nov 05 11:48:36 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:15] Invalid user/group name or numeric ID, ignoring: ${USER}
Nov 05 11:48:49 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 11:48:50 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 11:48:50 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 11:48:50 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 11:48:50 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 11:48:50 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 11:48:50 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 11:48:51 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 11:48:51 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 11:48:51 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
Nov 05 11:48:51 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:51:07 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:15] Invalid user/group name or numeric ID, ignoring: ${USER}
Nov 05 11:51:20 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:15] Invalid user/group name or numeric ID, ignoring: ${USER}
Nov 05 11:53:20 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 11:53:21 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 11:53:21 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 11:53:21 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 11:53:21 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
[... cut ...]
Nov 05 12:05:30 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 12:05:30 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 12:05:30 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
Nov 05 12:05:30 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
Nov 05 12:05:30 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 12:05:30 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.

Vino-git, 2018-11-06:
When I run the manual execution. I get the below error.

---------------------------
pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'raspberryg-13599-raspberry-qmwjrh'
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.3
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:failed to authenticate
[01584290] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01648268] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
[... cut ...]
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started

shivasiddharth, 2018-11-06:
your post is truncated. please post the entire terminal output.

Vino-git, 2018-11-06:
-- Logs begin at Fri 2016-11-04 00:16:43 +07, end at Tue 2018-11-06 22:35:38 +07. --
Nov 05 11:48:36 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:15] Invalid user/group name or numeric ID, ignoring: ${USER}
Nov 05 11:48:49 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 11:48:50 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 11:48:50 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 11:48:50 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 11:48:50 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 11:48:50 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 11:48:50 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:48:50 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 11:48:51 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 11:48:51 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 11:48:51 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
Nov 05 11:48:51 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:48:51 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:51:07 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:15] Invalid user/group name or numeric ID, ignoring: ${USER}
Nov 05 11:51:20 raspberrypi systemd[1]: [/lib/systemd/system/gassistpi-ok-google.service:15] Invalid user/group name or numeric ID, ignoring: ${USER}
Nov 05 11:53:20 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 11:53:21 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 11:53:21 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 11:53:21 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 11:53:21 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Main process exited, code=exited, status=210/CHROOT
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 05 11:53:21 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
[... cut ...]
Nov 05 12:05:30 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 05 12:05:30 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 05 12:05:30 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
Nov 05 12:05:30 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
Nov 05 12:05:30 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 05 12:05:30 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.

shivasiddharth, 2018-11-06:
I want to see the full output of the manual start.

Vino-git, 2018-11-06:
Here is my manual execution output. Note: I am able interact with google assistant but youtube streaming or GassistPi service are not working.

Thanks very much for your quick support. This is another github repo which is in full control.

pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'raspberryg-13599-raspberry-qmwjrh'
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.3
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:failed to authenticate
[0129e218] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[0135c308] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_alsa.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
[... cut ...]
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
device_model_id: raspbxxxxxxxxxxxxwjrh
device_id: 3DA9xxxxxxxxxxxxxxxxxxxxxxxxDB

ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED

Vino-git, 2018-11-07:
Here is the file from /GassistPi/systemd/ directory,

-------------------------------------------------------------------------------------------
[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/${USER}/env
Environment=PATH=/home/${USER}/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'raspbxxxxxxxxxxxxxxwjrh'

WorkingDirectory=/home/${USER}/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=${USER}

[Install]
WantedBy=multi-user.target
-------------------------------------------------------------------------------------------------------

And here is the file from my systemd folder,

--------------------------------------------------------------------------------------
cat gassistpi-ok-google.service 
[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'raspxxxxxxxxxxxxxxxxxxxxxjrh'

WorkingDirectory=/home/${USER}/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
-----------------------------------------------------------------------

shivasiddharth, 2018-11-07:
run:
sudo systemctl disable gassistpi-ok-google.service

In the GassistPI/systemd/ service file change all ${USER} to pi and run the service installer again.

Vino-git, 2018-11-07:
That's cool. Thanks for your response. The issue fixed. I will close this ticket.

closed 2018-11-07 by Vino-git