shivasiddharth/GassistPi #618: No matching playlist when applying Deezer command

shaltut, 2018-11-07:
Now after fixing gtoken and gtts_token the command goas on but stop and giving a sound that say ' no matching playlist ' and the pi stop answering

shivasiddharth, 2018-11-07:
Please stick to the issue template.

shivasiddharth, 2018-11-07:
Manually Start the assistant and post the output after giving the command.

shaltut, 2018-11-07:
the output after giving the command

`ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Leicester number one on Deezer"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Leicester number one on Deezer"}
stopping vlc
/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py:847: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
  InsecureRequestWarning)
{'checksum': 'f9eb0bbdd7f6ac6a38f44dccc6cf21a840cd750bba9870f18aada2478b24840a', 'total': 1, 'data': [{'creator': {'name': 'Ashraf.Ayman', 'id': 2414732224, 'type': 'user', 'tracklist': 'https://api.deezer.com/user/2414732224/flow'}, 'picture_medium': 'https://e-cdns-images.dzcdn.net/images/cover/5662acdbbd564d893839551e2d4e994a/250x250-000000-80-0-0.jpg', 'link': 'https://www.deezer.com/playlist/5076241704', 'nb_tracks': 1, 'time_mod': 1541602282, 'picture_xl': 'https://e-cdns-images.dzcdn.net/images/cover/5662acdbbd564d893839551e2d4e994a/1000x1000-000000-80-0-0.jpg', 'type': 'playlist', 'tracklist': 'https://api.deezer.com/playlist/5076241704/tracks', 'id': 5076241704, 'time_add': 1541602282, 'picture_big': 'https://e-cdns-images.dzcdn.net/images/cover/5662acdbbd564d893839551e2d4e994a/500x500-000000-80-0-0.jpg', 'duration': 139, 'picture_small': 'https://e-cdns-images.dzcdn.net/images/cover/5662acdbbd564d893839551e2d4e994a/56x56-000000-80-0-0.jpg', 'creation_date': '2018-11-07 14:51:22', 'title': 'Number one', 'public': True, 'is_loved_track': False, 'fans': 0, 'rating': 0, 'picture': 'https://api.deezer.com/playlist/5076241704/image', 'collaborative': False, 'checksum': '09f49ddfb52edb5e6d9dcd662b47482f'}]}
No matching playlists found
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:02] Decoding of say.mp3 finished.
ERROR:__main__:local variable 'tracklisturl' referenced before assignment
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 666, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 658, in main
    deezer_playlist_select(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1508, in deezer_playlist_select
    deezer_tracks_response = requests.get(tracklisturl,verify=False)
UnboundLocalError: local variable 'tracklisturl' referenced before assignment`

shivasiddharth, 2018-11-08:
Syntax is wrong. check the readme for the correct syntax.

closed 2018-11-08 by shivasiddharth

rutherb, 2018-11-08:
Hi - I have the same issue and I used the correct syntax as per the Readme. Log details from manual start of GassistPi are:-
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play playlist number one from Deezer"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play playlist number one from Deezer"}
stopping vlc
/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py:847: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
  InsecureRequestWarning)
No matching playlists found
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:02] Decoding of say.mp3 finished.
ERROR:__main__:local variable 'tracklisturl' referenced before assignment
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 672, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 664, in main
    deezer_playlist_select(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1507, in deezer_playlist_select
    deezer_tracks_response = requests.get(tracklisturl,verify=False)
UnboundLocalError: local variable 'tracklisturl' referenced before assignment

shivasiddharth, 2018-11-08:
As you can see, the command being deciphered is "Playlist number one" the command should be "Playlist 1" it should contain the number as a numeral and not as a word. Try just saying "Play playlist 1 from Deezer"

shivasiddharth, 2018-11-08:
@rutherb Did you go though my above post (https://github.com/shivasiddharth/GassistPi/issues/618#issuecomment-436969647) before opening up a new issue ?

rutherb, 2018-11-08:
Sorry I updated #618 but then thought that I could not re-open the issue as I was not the originator so I raised a new issue.

I have tried again with syntax "play playlist 1 from Deezer" and it gets to the point of "getting tracks from playlist 1" but then fails. Log details are:

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play playlist 1 from Deezer"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play playlist 1 from Deezer"}
stopping vlc
/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py:847: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
  InsecureRequestWarning)
Getting links for playlist number 1
Getting links for playlist number 1
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:02] Decoding of say.mp3 finished.
Getting the tracks from 1
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:02] Decoding of say.mp3 finished.
No More I Love You`s by Annie Lennox from The Very Best Of
WARNING:googleapiclient.discovery_cache:file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 36, in autodetect
    from google.appengine.api import memcache
ImportError: No module named 'google.appengine'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ImportError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ImportError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
[... 88 lines cut ...]
  File "/home/pi/GassistPi/src/actions.py", line 1516, in deezer_playlist_select
    vlcplayer.gaana_player(currenttrackid)
  File "/home/pi/GassistPi/src/mediaplayer.py", line 206, in gaana_player
    audiostream, videostream = youtube_stream_link(fullurl)
  File "/home/pi/GassistPi/src/youtube_search_engine.py", line 125, in youtube_stream_link
    video = pafy.new(video_url)
  File "/home/pi/env/lib/python3.5/site-packages/pafy/pafy.py", line 124, in new
    return Pafy(url, basic, gdata, size, callback, ydl_opts)
  File "/home/pi/env/lib/python3.5/site-packages/pafy/backend_youtube_dl.py", line 29, in __init__
    super(YtdlPafy, self).__init__(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/pafy/backend_shared.py", line 96, in __init__
    self._fetch_basic()
  File "/home/pi/env/lib/python3.5/site-packages/pafy/backend_youtube_dl.py", line 41, in _fetch_basic
    raise IOError(str(e).replace('YouTube said', 'Youtube says'))
OSError: ERROR: Signature extraction failed: Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py", line 1221, in _decrypt_signature
    video_id, player_url, s
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py", line 1130, in _extract_signature_function
    res = self._parse_sig_js(code)
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py", line 1193, in _parse_sig_js
    jscode, 'Initial JS player signature function name', group='sig')
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/common.py", line 983, in _search_regex
    raise RegexNotFoundError('Unable to extract %s' % _name)
youtube_dl.utils.RegexNotFoundError: Unable to extract Initial JS player signature function name; please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.
 (caused by RegexNotFoundError('Unable to extract \x1b[0;34mInitial JS player signature function name\x1b[0m; please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.',)); please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.

shivasiddharth, 2018-11-08:
Open terminal and enter the following:
```
source env/bin/activate
pip install -U youtube-dl
sudo pip install -U youtube-dl
```

rutherb, 2018-11-08:
Yes that worked - thanks very much shivasiddharth ! Brilliant support and help :-) n