shivasiddharth/GassistPi #619: After the installation the assistant stop working

HexRebuilt, 2018-11-07:
### Describe the bug:
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### 3. Does the assistant service start normally?
no

### 4. Is the assistant service restarting automatically?
yes

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```
the service starts but i cannot trigger any action by using "hey google" or "ok google". when the installation completes, the assistant works, but after stopping it in order to reboot there is no way to have any kind of response.

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
pi@BMOpi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'bmo-pi-b7a58-raspberry-pi-3-qjv9nr'
```

### Paste the contents of the terminal below
```
`
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 40, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from pushbullet import Pushbullet
  File "/home/pi/env/lib/python3.5/site-packages/pushbullet/__init__.py", line 4, in <module>
    from .listener import Listener
  File "/home/pi/env/lib/python3.5/site-packages/pushbullet/listener.py", line 9, in <module>
    import websocket
  File "/home/pi/env/lib/python3.5/site-packages/websocket/__init__.py", line 22, in <module>
    from ._abnf import *
  File "/home/pi/env/lib/python3.5/site-packages/websocket/_abnf.py", line 34, in <module>
    import numpy
  File "/home/pi/env/lib/python3.5/site-packages/numpy/__init__.py", line 160, in <module>
    from . import random
  File "/home/pi/env/lib/python3.5/site-packages/numpy/random/__init__.py", line 94, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 669, in exec_module
  File "<frozen importlib._bootstrap_external>", line 773, in get_code
  File "<frozen importlib._bootstrap_external>", line 484, in _compile_bytecode
ValueError: bad marshal data (unknown type code)`
```

shivasiddharth, 2018-11-07:
Which OS and version are you using ?

HexRebuilt, 2018-11-08:
hi, i'm using the october release of raspbian

shivasiddharth, 2018-11-08:
The installed python packages could be corrupt, can you try re-running the installer.

shivasiddharth, 2018-11-09:
```
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 40, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from pushbullet import Pushbullet
  File "/home/pi/env/lib/python3.5/site-packages/pushbullet/__init__.py", line 4, in <module>
    from .listener import Listener
  File "/home/pi/env/lib/python3.5/site-packages/pushbullet/listener.py", line 9, in <module>
    import websocket
  File "/home/pi/env/lib/python3.5/site-packages/websocket/__init__.py", line 22, in <module>
    from ._abnf import *
  File "/home/pi/env/lib/python3.5/site-packages/websocket/_abnf.py", line 34, in <module>
    import numpy
  File "/home/pi/env/lib/python3.5/site-packages/numpy/__init__.py", line 160, in <module>
    from . import random
  File "/home/pi/env/lib/python3.5/site-packages/numpy/random/__init__.py", line 94, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 669, in exec_module
  File "<frozen importlib._bootstrap_external>", line 773, in get_code
  File "<frozen importlib._bootstrap_external>", line 484, in _compile_bytecode
ValueError: bad marshal data (unknown type code)`
```

Seems to be a part of the error. Can you manually start the assistant and post the full terminal content.

HexRebuilt, 2018-11-09:
`pi@BMOpi:~ $ sudo service gassistpi-ok-google stop
pi@BMOpi:~ $ sudo systemctl is-active gassistpi-ok-google.service
inactive
pi@BMOpi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'bmo-pi-b7a58-raspberry-pi-3-qjv9nr'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 40, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from pushbullet import Pushbullet
  File "/home/pi/env/lib/python3.5/site-packages/pushbullet/__init__.py", line 4, in <module>
    from .listener import Listener
  File "/home/pi/env/lib/python3.5/site-packages/pushbullet/listener.py", line 9, in <module>
    import websocket
  File "/home/pi/env/lib/python3.5/site-packages/websocket/__init__.py", line 22, in <module>
    from ._abnf import *
  File "/home/pi/env/lib/python3.5/site-packages/websocket/_abnf.py", line 34, in <module>
    import numpy
  File "/home/pi/env/lib/python3.5/site-packages/numpy/__init__.py", line 160, in <module>
    from . import random
  File "/home/pi/env/lib/python3.5/site-packages/numpy/random/__init__.py", line 94, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 669, in exec_module
  File "<frozen importlib._bootstrap_external>", line 773, in get_code
  File "<frozen importlib._bootstrap_external>", line 484, in _compile_bytecode
ValueError: bad marshal data (unknown type code)`

sure thing. this is the try with stopping the service before starting the assistant. same result without doing that. 
In addition i don't know if it's related but on the googleAPI dashboard i can see a discrete number of request with quite the same number of errors.
[image]

HexRebuilt, 2018-11-14:
so do you ahve any idea? because in those days i've tried the AIY version of the assistant and that works without a prolem. still i would like to use your repository because it has much more function. Do you have some specific release of rasbian that i should install?

HexRebuilt, 2018-11-15:
`Package      Version Latest Type
------------ ------- ------ -----
click        6.7     7.0    wheel
cryptography 2.3.1   2.4.1  wheel
google-auth  1.5.1   1.6.1  wheel
grpcio       1.16.0  1.16.1 wheel
httplib2     0.11.3  0.12.0 wheel
PyYAML       3.12    3.13   wheel
requests     2.20.0  2.20.1 wheel
RPi.GPIO     0.6.3   0.6.4  wheel
setuptools   40.5.0  40.6.2 wheel
tenacity     4.12.0  5.0.2  wheel`

HexRebuilt, 2018-11-15:
oh i see. there is the third time that i re-run the installer. there is a way to install those package manually and then install the Gassistant?

shivasiddharth, 2018-11-16:
run the following command and share the output
```
source env/bin/activate
pip list
```

HexRebuilt, 2018-11-16:
`Package                  Version
------------------------ ----------
aftership                0.2
appdirs                  1.4.3
asn1crypto               0.24.0
beautifulsoup4           4.6.3
bs4                      0.0.1
cachetools               3.0.0
casttube                 0.2.0
certifi                  2018.10.15
cffi                     1.11.5
chardet                  3.0.4
click                    6.7
cryptography             2.3.1
decorator                4.3.0
enum34                   1.1.6
feedparser               5.2.1
future                   0.17.1
futures                  3.1.1
gmusicapi                11.0.3
google-api-python-client 1.7.4
google-assistant-grpc    0.2.0
google-assistant-library 1.0.0
google-assistant-sdk     0.5.0
google-auth              1.5.1
google-auth-httplib2     0.0.3
google-auth-oauthlib     0.2.0
googleapis-common-protos 1.5.5
googletrans              2.3.0
gpiozero                 1.4.1
gpsoauth                 0.4.1
grpcio                   1.16.0
gTTS                     2.0.1
gTTS-token               1.1.2
httplib2                 0.11.3
idna                     2.7
ifaddr                   0.1.4
ipaddress                1.0.22
kodi-json                1.0.0
lxml                     4.2.5
MechanicalSoup           0.11.0
mock                     2.0.0
mutagen                  1.41.1
numpy                    1.15.4
oauth2client             4.1.3
oauthlib                 2.1.0
pafy                     0.5.4
pathlib2                 2.3.2
pbr                      5.1.1
pip                      18.1
pkg-resources            0.0.0
proboscis                1.2.6.0
protobuf                 3.6.1
psutil                   5.4.8
pushbullet.py            0.11.0
pyasn1                   0.4.4
pyasn1-modules           0.2.2
PyAudio                  0.2.11
PyChromecast             2.3.0
pycparser                2.19
pycryptodomex            3.7.0
pyOpenSSL                18.0.0
python-dateutil          2.7.5
python-magic             0.4.15
python-vlc               3.0.4106
PyYAML                   3.12
requests                 2.20.0
requests-oauthlib        1.0.0
RPi.GPIO                 0.6.3
rsa                      4.0
setuptools               40.5.0
six                      1.11.0
sounddevice              0.3.12
spidev                   3.2
spotipy                  2.4.4
tenacity                 4.12.0
uritemplate              3.0.0
urllib3                  1.24.1
validictory              1.1.2
websocket-client         0.54.0
wheel                    0.32.2
youtube-dl               2018.11.7
zeroconf                 0.21.3`

shivasiddharth, 2018-11-19:
Try this:
```
source env/bin/activate
pip uninstall numpy
pip install numpy==1.15.3
```

HexRebuilt, 2018-11-21:
hi i've done that and also retry the comand
`source env/bin/activate
pip list -o --format columns`

and the result seems quite the same to me unfortunatly. do you have any other suggestions?

`Package      Version   Latest     Type
------------ --------- ---------- -----
click        6.7       7.0        wheel
cryptography 2.3.1     2.4.2      wheel
gmusicapi    11.0.3    11.0.4     wheel
google-auth  1.5.1     1.6.1      wheel
grpcio       1.16.0    1.16.1     wheel
httplib2     0.11.3    0.12.0     wheel
numpy        1.15.3    1.15.4     wheel
PyYAML       3.12      3.13       wheel
requests     2.20.0    2.20.1     wheel
RPi.GPIO     0.6.3     0.6.5      wheel
setuptools   40.5.0    40.6.2     wheel
tenacity     4.12.0    5.0.2      wheel
wheel        0.32.2    0.32.3     wheel
youtube-dl   2018.11.7 2018.11.18 wheel`

shivasiddharth, 2018-11-21:
I am out of ideas at this point...  I will leave the issue open to see if someone comes up with a solution.. as far as I know this is due to a corrupted package. Try the November OS if possible.

HexRebuilt, 2018-12-02:
Hi the november release fixed the problem and now it works for the most part. thanks again

shivasiddharth, 2018-12-02:
Looks like the image was corrupt or something. Closing this issue then.

closed 2018-12-02 by shivasiddharth