shivasiddharth/GassistPi #622: Pi freezes/crashes as soon as audio detected

Colin1964, 2018-11-07:
### Describe the bug:
```
Demo initiates, I hear the start up melody and assistant says “Hi” Screen output then shows the info messages about gmusicapi and PulseAudio server errors. As soon as any audio is detected the program crashes 
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Yes/No - Not tested

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
Running demo with: $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'gassistpi-XXXXX' --device-model-id 'gassistpi-XXXXX-gassistpi-XXXXX'
```

### Paste the contents of the terminal below
```
pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'gassistpi-XXXXX' --device-model-id 'gassistpi-XXXXX-gassistpi-XXXXX'
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.3
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:failed to authenticate
[01427348] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01432cc0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
USB-MIC-JACK
{No more output – Pi freezes/crashes as soon as audio detected)

README says this works on all Pi Boards so trying to get this to work on Raspberry Pi 1 although I see
some comments say Google Assistant does not work on RP1? Hardware, OS and Audio setup as shown below:

pi@raspberrypi:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 000e
Serial		: 000000002e84823c
I am running clean install of Raspbian “Stretch” from Raspberry Pi.Org
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL=http://www.raspbian.org/RaspbianBugs

Using Blue Snowball USB Mic and powered speakers connected to 3.5mm jack
USB MIC AND AUDIO JACK configuration used and speaker-test working correctly.
Pi is running in headless configuration and I am connected via SSH with PuTTy

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.
No log file in /tmp

penbob3, 2018-11-08:
I too have this issue. Newest Raspiant Stretch and Model B+. Using a USB Razer mic, and I get the exact same terminal output. This happens regardless of starting the service or via the demo command. Sometimes it gets past the jingle and says 'Hi', other times it cuts off during it.

shivasiddharth, 2018-11-08:
The problem is not all mics are compatible with the google assistant. In the asoundrc and asound.conf file, try changing the channels to 2

penbob3, 2018-11-08:
Just tried this, the problem still persists. It crashes before it even says "Hi".

shivasiddharth, 2018-11-08:
Manually start the assistant and post the output here.

penbob3, 2018-11-08:
`
pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/p         ushbutton.py --project-id rpiassistant-XXXXX  --device-model-id rpiassistant-XXXXXXXXX
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.3
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:failed to authenticate
USB-MIC-JACK
`
Identical output to his

Colin1964, 2018-11-08:
This could be related to a potential Mic compatibility issue as I have also tried running the Googlesample script pushtotalk.py. In this case the script runs without crashing, audio in appears fine - ie I can issue a command and the command is understood and a response is provided. The only problem here is I get several buffer underflow errors from SoundDeviceStream and very choppy audio out. This also seems to be a common problem (see [issue](https://github.com/googlesamples/assistant-sdk-python/issues/303) raised over on Googlesamples.

Colin1964, 2018-11-08:
Tested with 3 different USB mics:

[Snowball](https://www.amazon.co.uk/gp/product/B006DIA77E/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1) iCE USB Microphone
[Sienoc](https://www.amazon.co.uk/gp/product/B00XU1GHO4/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1) mini microphone USB
Samson [SAMETEORITE](https://www.amazon.co.uk/gp/product/B00I9RK97K/ref=oh_aui_detailpage_o04_s00?ie=UTF8&psc=1) Meteorite USB Condensor Microphone

For each mic I tried changing the asoundrc and asound.conf file channels to 2. 
The resulting output is exactly the same as shown at top of this thread in every case. 

The suggestion over on googlesamples is that the PI1 CPU gets maxxed out trying to handle the input and output audio streams. TBH the technical detail is beyond me but its interesting that the pushtotalk.py script works (albeit with buffer underflow errors) yet the pushputton.py sample crashes the Pi as soon as the audio input stream is active.

penbob3, 2018-11-10:
I tried installing and running the Googlesamples scripts, but I end up with no audio output or input. I think it's just the config files or whatever Pulseaudio is messing with ALSA cause my mic and speaker output work in AlexaPi. The 'audiotest' script doesnt record or play anything, and the 'pushtotalk' script gets to "Press  enter to whatever" but upon pressing it just does nothing and displays that same line again after a couple seconds.

Colin1964, 2018-11-12:
OK making progress. I have managed to stop `pushbutton.py` from crashing the Pi and it is now working - sort of! I used the Python debugger `pdb` with:

```
import pdb
pdb.set_trace()
```

inserted this at the beginning of the code and stepped through the program to try and find where the crash was being caused. It appears that it might have something to with the hard coded pushbutton and indicator routines. I am running a headless Pi connected via VNC and have no additional hardware e.g push buttons or indicator lights connected. After noting where the program crashed I commented out the following lines of code:

```
#from indicator import stoppushbutton
#from indicator import assistantindicator
```

```
 #self.t3 = Thread(target=self.stopbutton)
 #self.t3.start()
```

```
    #def stopbutton(self):
    #    while mediastopbutton:
    #        time.sleep(0.25)
    #        if not GPIO.input(stoppushbutton):
    #            print('Stopped')
    #            stop()
```

`    #assistantindicator('listening')`

`   #assistantindicator('speaking')`

`   #assistantindicator('off')`

`   #assistantindicator('off')` - There are two instances of this line in the code

With above changes made to `pushbutton.py` I get the following output. 
```
pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'gassistpi-xxxxx' --device-model-id 'gassistpi-xxxxxx-gassistpi-xxxxx'
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.3
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:failed to authenticate
[0148e900] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01487978] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
^Cwhat time is it
where are the malvern hills
tell me about tasmania
who is mickey mouse
```

After the initial start up melody and "Hi" there is no further output. Pressing the Enter key has no effect however when I press CTRL+C I hear what I think is the "listening" chime. If I then speak a command (without pressing Enter or saying a hotword) the text of the command is presented in the terminal window and the assistant provides a verbal response. It is limited success though because (possibly related to stopbutton routine being disbled) the "listening" chime repeats every few seconds so it is effectively always on. An additional downside is that I am unable to exit the program. As mentioned all keyboard entry including CTRL+C is just relayed to to the terminal window as text. The program can only be exited by closing the terminal. So it works but only sort of! Anybody know what might be causing the Pi to crash in the stoppushbutton and assistantindicator routines?

Colin1964, 2018-11-12:
#544 Also reports `pushbutton.py` going into "listening loop" after pressing CTRL+C. In my case config.yaml has wakeword set to enabled ie No GPIO pushbutton trigger.

shivasiddharth, 2018-11-13:
Stop button is controlled by a thread. In python you cannot exit or end a thread easily. If Wakeword is enabled, then pushbutton will not work. I have already mentioned that in #544 . The audio issues started after the Aug 9 commit in the SDK. https://github.com/googlesamples/assistant-sdk-python/commit/d05f675f5f28895efac6245a2f129e83adfd45da May be you can try to undo the change what they did. you can find your audiohelper file in the googlesamples folder within the environment.

shivasiddharth, 2018-11-13:
And adding to that, ENTER key is not the mode to trigger the pushbutton script of this project. Its only for the googlesamples file. If you disable wakeword, then pushbutton trigger will become active.

bachoo786, 2019-06-12:
@Colin1964 did you ever sort this issue out? I am having same problem with pushbutton.py etc. Running a pi zero w and 2 mic pi hat. 

Assistant starts working but and I can ask a few questions but then it would all of a sudden stop working until I reboot the pi.

navas, 2019-07-14:
I too encountered the same issue, I am using pi0w with,

The main thing is the assistant does like sharing the audio. (using "dmix" in asound.conf or .asoundrc") It needs to be dedicated. Use this following .asoundrc to override, change the hw ID and it should work. But the bad thing is you can't use any other tools like aplay or anything

pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}
pcm.mic {
  type plug
  slave {
    pcm "hw:0,0"
  }
}
pcm.speaker {
  type plug
  slave {
    pcm "hw:1,0"
  }
}

alias13e, 2021-03-22:
I'm having a similar issue.  I got the assistant installed, started up, and working fine, but it couldn't access youtube to play music, which is a big part of its intended function.  I followed the steps to enable youtube and run on startup, they appeared to go fine, but when I rebooted I get the chime, a "hello", and then it crashes.  I ran the line to manually start and it appears to do speech to text, but nothing else.  Manual start results pasted below.  Thanks in advance!

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dacplus], device 0: HiFiBerry DAC+ HiFi pcm512x-hifi-0 [HiFiBerry DAC+ HiFi pcm512x-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 1: ArrayUAC10 [ReSpeaker 4 Mic Array (UAC1.0)], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py  --project_id 'robin-speaker' --device_model_id 'robin-speaker-robin-speaker-clush1'
/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:32: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  dependencies=[google_dot_api_dot_annotations__pb2.DESCRIPTOR,google_dot_longrunning_dot_operations__pb2.DESCRIPTOR,google_dot_protobuf_dot_any__pb2.DESCRIPTOR,google_dot_protobuf_dot_duration__pb2.DESCRIPTOR,google_dot_protobuf_dot_empty__pb2.DESCRIPTOR,google_dot_protobuf_dot_timestamp__pb2.DESCRIPTOR,google_dot_rpc_dot_status__pb2.DESCRIPTOR,])
/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:45: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  type=None),
/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:49: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  type=None),
/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:53: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  type=None),
/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:57: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  type=None),
/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:61: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  type=None),
/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:65: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  type=None),
/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:69: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  type=None),
/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:73: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  type=None),
/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:78: DeprecationWarning: Call to deprecated create function EnumDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_end=1382,
/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:91: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  type=None),
/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:95: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  type=None),
/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:100: DeprecationWarning: Call to deprecated create function EnumDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find desc [... cut ...]
/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: robin-speaker-robin-speaker-clush1
device_id: BEE8FF5229409EF176D0167DC95853C4

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_MEDIA_STATE_IDLE

ON_START_FINISHED

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what time is it"}

ON_RESPONDING_STARTED:
  {"is_error_response": false}