shivasiddharth/GassistPi #625: Spotify error

oceank82, 2018-11-08:
### Describe the bug:
```
Every time I try to use spotify "play home from spotify" I get error:

nov 08 18:29:21 raspberrypi python[4913]:   File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line
nov 08 18:29:21 raspberrypi python[4913]:     from oauth2client.locked_file import LockedFile
nov 08 18:29:21 raspberrypi python[4913]: ImportError: No module named 'oauth2client.locked_file'
nov 08 18:29:21 raspberrypi python[4913]: During handling of the above exception, another exception occurred:
nov 08 18:29:21 raspberrypi python[4913]: Traceback (most recent call last):
nov 08 18:29:21 raspberrypi python[4913]:   File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41
nov 08 18:29:21 raspberrypi python[4913]:     from . import file_cache
nov 08 18:29:21 raspberrypi python[4913]:   File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line
nov 08 18:29:21 raspberrypi python[4913]:     'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
nov 08 18:29:21 raspberrypi python[4913]: ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
~

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
Play home from Spotify
```

### Paste the contents of the terminal below
```
nov 08 18:29:21 raspberrypi python[4913]:   File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line
nov 08 18:29:21 raspberrypi python[4913]:     from oauth2client.locked_file import LockedFile
nov 08 18:29:21 raspberrypi python[4913]: ImportError: No module named 'oauth2client.locked_file'
nov 08 18:29:21 raspberrypi python[4913]: During handling of the above exception, another exception occurred:
nov 08 18:29:21 raspberrypi python[4913]: Traceback (most recent call last):
nov 08 18:29:21 raspberrypi python[4913]:   File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41
nov 08 18:29:21 raspberrypi python[4913]:     from . import file_cache
nov 08 18:29:21 raspberrypi python[4913]:   File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line
nov 08 18:29:21 raspberrypi python[4913]:     'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
nov 08 18:29:21 raspberrypi python[4913]: ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
~

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-11-08:
In-spite of that still it would play the songs.  The assistant would not crash. These are warnings.

shivasiddharth, 2018-11-08:
You have just posted here a segment of the terminal output. If you look at the entire terminal output, the assistant would be working normally.

shivasiddharth, 2018-11-08:
Paste the entire output of the terminal then.

oceank82, 2018-11-08:
ok cool , thanx! Well im new to this so what command do I type for getting the whole log?

shivasiddharth, 2018-11-08:
Run the assistant manually, give the command, copy the entire thing that you have on the terminal and paste it here.

oceank82, 2018-11-08:
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play home from Spotify"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play home from Spotify"}
stopping vlc
Getting music links
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
home
4
home
Elvis Presley Hound Dog
WARNING:googleapiclient.discovery_cache:file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 36, in autodetect
    from google.appengine.api import memcache
ImportError: No module named 'google.appengine'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ImportError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ImportError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
ERROR: Signature extraction failed: Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py", line 1221, in _decrypt_signature
    video_id, player_url, s
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py", line 1130, in _extract_signature_function
    res = self._parse_sig_js(code)
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py", line 1193, in _parse_sig_js
    jscode, 'Initial JS player signature function name', group='sig')
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/common.py", line 983, in _search_regex
    raise RegexNotFoundError('Unable to extract %s' % _name)
youtube_dl.utils.RegexNotFoundError: Unable to extract Initial JS player signature function name; please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.
 (caused by RegexNotFoundError('Unable to extract \x1b[0;34mInitial JS player signature function name\x1b[0m; please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.',)); please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.
ERROR:__main__:ERROR: Signature extraction failed: Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py", line 1221, in _decrypt_signature
    video_id, player_url, s
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py", line 1130, in _extract_signature_function
    res = self._parse_sig_js(code)
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/youtube.py", line 1193, in _parse_sig_js
    jscode, 'Initial JS player signature function name', group='sig')
  File "/home/pi/env/lib/python3.5/site-packages/youtube_dl/extractor/common.py", line 983, in _search_regex
[... 67 lines cut ...]
  File "/home/pi/GassistPi/src/actions.py", line 1358, in spotify_playlist_select
    vlcplayer.spotify_player(currenttrackid)
  File "/home/pi/GassistPi/src/mediaplayer.py", line 195, in spotify_player
    audiostream, videostream = youtube_stream_link(fullurl)
  File "/home/pi/GassistPi/src/youtube_search_engine.py", line 125, in youtube_stream_link
    video = pafy.new(video_url)
  File "/home/pi/env/lib/python3.5/site-packages/pafy/pafy.py", line 124, in new
    return Pafy(url, basic, gdata, size, callback, ydl_opts)
  File "/home/pi/env/lib/python3.5/site-packages/pafy/backend_youtube_dl.py", line 29, in __init__
    super(YtdlPafy, self).__init__(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/pafy/backend_shared.py", line 96, in __init__
    self._fetch_basic()
  File "/home/pi/env/lib/python3.5/site-packages/pafy/backend_youtube_dl.py", line 41, in _fetch_basic
    raise IOError(str(e).replace('YouTube said', 'Youtube says'))
[... cut ...]
 (caused by RegexNotFoundError('Unable to extract \x1b[0;34mInitial JS player signature function name\x1b[0m; please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.',)); please report this issue on https://yt-dl.org/bug . Make sure you are using the latest version; see  https://yt-dl.org/update  on how to update. Be sure to call youtube-dl with the --verbose flag and include its complete output.

shivasiddharth, 2018-11-08:
```
Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes
```
Your error has already been discussed in #618 https://github.com/shivasiddharth/GassistPi/issues/618#issuecomment-436975900

closed 2018-11-08 by shivasiddharth