shivasiddharth/GassistPi #630: Spotify won't work

ricky0178, 2018-11-11:
### Describe the bug:
```When i ask to play downloader from spotify (music is the playlist name while saying it in Italian) it just crashes the assistant
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
{"text": "Riproduci Downloader su Spotify"}
```

### Paste the contents of the terminal below
```
Traceback (most recent call last):
   File "/home/pi/GassistPi/src/main.py", line 673, in <module>
     Myassistant().main()
   File "/home/pi/GassistPi/src/main.py", line 657, in main
     spotify_playlist_select(str(usrcmd).lower())
   File "/home/pi/GassistPi/src/actions.py", line 1349, in spotify_playlist_select
     playlists,num=scan_spotify_playlists()
   File "/home/pi/GassistPi/src/actions.py", line 1325, in scan_spotify_playlists
     playlistdetails.append(playlist_name)
 UnboundLocalError: local variable 'playlist_name' referenced before assignment
```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

No logs can be found

ricky0178, 2018-11-11:
for somewhat reason it didn't include the
Describe the bug:
soo..
Well, i wanted to play Downloader (my spotify playlist) from spotify, but it didn't work

shivasiddharth, 2018-11-11:
Is the playlist created by you or was it created by someone else that you have added to your account ?

ricky0178, 2018-11-11:
the playlist is public and is indeed mine

shivasiddharth, 2018-11-11:
Check #469 .

ricky0178, 2018-11-11:
Tried to uncomment the command 
`print (spotify_playlists)`
btu it gives me the same output of @aandroide; {'Playlists': [0, 0, 0, 0]}
the downloader playlist is indeed mine though, and it's the only playlist that i've created

shivasiddharth, 2018-11-11:
If you have added other playlists that you have not created this is the kind of result we would get.. even though you are requesting your playlist, if you have playlists that we're not created by you the Spotify feature will not work. That's what was discussed in #469

aandroide, 2018-11-11:
Esattamente sid. Io ho risolto eliminando le altre playlist, e tenendo soltanto la mia playlist creata da me.

ricky0178, 2018-11-11:
faaantatic, another brand new error

`WARNING:googleapiclient.discovery_cache:file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 36, in autodetect
    from google.appengine.api import memcache
ImportError: No module named 'google.appengine'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ImportError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ImportError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth`

ricky0178, 2018-11-11:
Seems to work now, oddly enough

aandroide, 2018-11-11:
A volte basta un semplice riavvio in modo da rendere effettive le modifiche fatte ai file.

shivasiddharth, 2018-11-11:
You can find almost all the info in the issues or README. Check #603

shivasiddharth, 2018-12-02:
Closing this for inactivity.

closed 2018-12-02 by shivasiddharth