shivasiddharth/GassistPi #632: GassistPi stopps working after sending wrong light name

aro28, 2018-11-17:
Hi Shiva!
A few days ago, I began to notice strange behavior, when i try to send voice command with with wrong light's name GassisPi freezes and stops answering me. Also, "Ok, Google" command stops working.
I have Sonoff - Tasmota devices and i've added them in config file. In same time Hue Bridge is runing on same Pi. But when i send voice commands with undefined light name, my gassistpi-ok-google.service freezes with below error..The diyhue guys seems changed the call patterns.

● gassistpi-ok-google.service - Google Assistant-Voice
Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
Active: active (running) since Thu 2018-11-15 20:46:57 +06; 21h ago
Main PID: 423 (python)
CGroup: /system.slice/gassistpi-ok-google.service
└─423 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id saved-model-id

Nov 16 18:06:54 magicmirror python[423]: File "/home/pi/GassistPi/src/actions.py", line 1247, in hue_control
Nov 16 18:06:54 magicmirror python[423]: currentxval=hueconfig['lights'][lightindex]['state']['xy'][0]
Nov 16 18:06:54 magicmirror python[423]: KeyError: 'xy'
Nov 16 18:06:54 magicmirror python[423]: During handling of the above exception, another exception occurred:
Nov 16 18:06:54 magicmirror python[423]: Traceback (most recent call last):
Nov 16 18:06:54 magicmirror python[423]: File "/home/pi/GassistPi/src/main.py", line 673, in <module>
Nov 16 18:06:54 magicmirror python[423]: Myassistant().main()
Nov 16 18:06:54 magicmirror python[423]: File "/home/pi/GassistPi/src/main.py", line 457, in main
Nov 16 18:06:54 magicmirror python[423]: except Keyerror:
Nov 16 18:06:54 magicmirror python[423]: NameError: name 'Keyerror' is not defined

aro28, 2018-11-17:
[image]

shivasiddharth, 2018-11-24:
Can you check if the light that you requested has 'xy' state from the config.json. Load your config.json in this site https://jsoneditoronline.org this will make searching easy. On the right top, choose Open and open from disk and then select your config.json file. 
Was the triggering that you requested for a switch or a light ?

aro28, 2018-11-28:
@shivasiddharth  I don't have "config.json" file, only config.yaml and config.js. Which one should i upload?
Switch was not  triggered. I will observe the behavior of the program and will post here errors.

aro28, 2018-11-28:
[image]
For instance. I say: " Turn Kitchen On" and it's stopped working and got that error.

aro28, 2018-11-28:
[image]
Also, i have Hall, Kitchen and Lobby lights.

aro28, 2018-11-28:
As i remember this issue was started after you changed diyHue installation process. You removed diyHue from GassistPi installation script. Now we should follow the guidelines given in the diyHue's Wiki to setup the Emulated Hue Service.

aro28, 2018-11-28:
I've reinstalled GassistPi and same behavior.

shivasiddharth, 2018-11-29:
YOu should have a config.json file in the /opt/hue-emulator/ directory. SOmething like this. https://github.com/mariusmotea/diyHue/blob/master/BridgeEmulator/config.json

aro28, 2018-11-29:
I just realized that it's not relared light name. It crashes when i use any lights name.

● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-11-29 16:28:31 +06; 2h 39min ago
 Main PID: 428 (python)
   CGroup: /system.slice/gassistpi-ok-google.service
           └─428 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id saved-model-id

Nov 29 19:07:52 magicmirror python[428]:   File "/home/pi/GassistPi/src/actions.py", line 1262, in hue_control
Nov 29 19:07:52 magicmirror python[428]:     currentxval=hueconfig['lights'][lightindex]['state']['xy'][0]
Nov 29 19:07:52 magicmirror python[428]: KeyError: 'xy'
Nov 29 19:07:52 magicmirror python[428]: During handling of the above exception, another exception occurred:
Nov 29 19:07:52 magicmirror python[428]: Traceback (most recent call last):
Nov 29 19:07:52 magicmirror python[428]:   File "/home/pi/GassistPi/src/main.py", line 692, in <module>
Nov 29 19:07:52 magicmirror python[428]:     Myassistant().main()
Nov 29 19:07:52 magicmirror python[428]:   File "/home/pi/GassistPi/src/main.py", line 476, in main
Nov 29 19:07:52 magicmirror python[428]:     except Keyerror:
Nov 29 19:07:52 magicmirror python[428]: NameError: name 'Keyerror' is not defined

aro28, 2018-11-29:
This device is not registered. This means you will not be able to use
    Device Actions or see your device in Assistant Settings. In order to
    register this device follow instructions at:

    https://developers.google.com/assistant/sdk/guides/library/python/embed/register-device

ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED
print(hueconfig['lights'][lightindex])
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "turn on Lobby"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "turn on Lobby"}
ERROR:__main__:name 'Keyerror' is not defined
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 474, in main
    hue_control(str(usrcmd).lower(),str(i),str(hueconfig['lights_address'][str(i)]['ip']))
  File "/home/pi/GassistPi/src/actions.py", line 1262, in hue_control
    currentxval=hueconfig['lights'][lightindex]['state']['xy'][0]
KeyError: 'xy'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 692, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 476, in main
    except Keyerror:
NameError: name 'Keyerror' is not defined
print(hueconfig['lights'][lightindex])

shivasiddharth, 2018-11-29:
I guess you have placed the print command in the wrong place. Why is even the print command getting printed ?

aro28, 2018-11-29:
I've added above line "print(hueconfig['lights'][lightindex])" into actions.py file and run manually.
I got below error:

    This device is not registered. This means you will not be able to use
    Device Actions or see your device in Assistant Settings. In order to
    register this device follow instructions at:

    https://developers.google.com/assistant/sdk/guides/library/python/embed/register-device

ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "turn on Lobby"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "turn on Lobby"}
{'swversion': '1.29.0_r21169', 'swconfigid': '321D79EA', 'modelid': 'LWB014', 'manufacturername': 'Philips', 'uniqueid': '00:17:88:01:03:3d:b9:eb-0b', 'productid': 'Philips-LWB014-1-A19DLv4', 'state': {'reachable': True, 'bri': 254, 'alert': 'none', 'mode': 'homeautomation', 'on': True}, 'productname': 'Hue white lamp', 'swupdate': {'state': 'noupdates', 'lastinstall': None}, 'capabilities': {'control': {'mindimlevel': 5000, 'maxlumen': 840}, 'certified': True, 'streaming': {'proxy': False, 'renderer': False}}, 'type': 'Dimmable light', 'config': {'direction': 'omnidirectional', 'function': 'functional', 'archetype': 'classicbulb'}, 'name': 'Lobby'}
ERROR:__main__:name 'Keyerror' is not defined
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 474, in main
    hue_control(str(usrcmd).lower(),str(i),str(hueconfig['lights_address'][str(i)]['ip']))
  File "/home/pi/GassistPi/src/actions.py", line 1263, in hue_control
    currentxval=hueconfig['lights'][lightindex]['state']['xy'][0]
KeyError: 'xy'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 692, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 476, in main
    except Keyerror:
NameError: name 'Keyerror' is not defined

aro28, 2018-11-29:
> I guess you have placed the print command in the wrong place. Why is even the print command getting printed ?
Now it seems to do right. See below. Thanks!

aro28, 2018-11-29:
Sorry. i forgot to say that i reinstalled diyHue program and not able to link emulated bridge with Hue app.
Working on it now.

aro28, 2018-11-30:
Hi @Shiva
I confirm that issue is happening with Tasmota device only. If i indicate sonoff-Tasmota device in config.yaml file than i'm getting below error message.

pi@raspberrypi:~ $ sudo systemctl status gassistpi.service 
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-11-30 15:14:22 +06; 1min 21s ago
 Main PID: 392 (python)
   CGroup: /system.slice/gassistpi.service
           └─392 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id mymirror-27be8-MyMirror

Nov 30 15:15:32 raspberrypi python[392]: ON_CONVERSATION_TURN_STARTED
Nov 30 15:15:32 raspberrypi python[392]: ON_CONVERSATION_TURN_STARTED
Nov 30 15:15:35 raspberrypi python[392]: ON_END_OF_UTTERANCE
Nov 30 15:15:35 raspberrypi python[392]: ON_END_OF_UTTERANCE
Nov 30 15:15:35 raspberrypi python[392]: ON_RECOGNIZING_SPEECH_FINISHED:
Nov 30 15:15:35 raspberrypi python[392]:   {"text": "turn on living light"}
Nov 30 15:15:35 raspberrypi python[392]: ON_RECOGNIZING_SPEECH_FINISHED:
Nov 30 15:15:35 raspberrypi python[392]:   {"text": "turn on living light"}
Nov 30 15:15:40 raspberrypi python[392]: Tunring on living light
Nov 30 15:15:42 raspberrypi python[392]: WARNING:gtts.tts:Unable to get language list: 'NoneType' object is not subscriptable
pi@raspberrypi:~ $ sudo systemctl status gassistpi.service 
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-11-30 15:14:22 +06; 1min 22s ago
 Main PID: 392 (python)
   CGroup: /system.slice/gassistpi.service
           └─392 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id mymirror-27be8-MyMirror

Nov 30 15:15:44 raspberrypi python[392]:     tts.save(ttsfilename)
Nov 30 15:15:44 raspberrypi python[392]:   File "/home/pi/env/lib/python3.5/site-packages/gtts/tts.py", line 247, in save
Nov 30 15:15:44 raspberrypi python[392]:     self.write_to_fp(f)
Nov 30 15:15:44 raspberrypi python[392]:   File "/home/pi/env/lib/python3.5/site-packages/gtts/tts.py", line 187, in write_to_fp
Nov 30 15:15:44 raspberrypi python[392]:     part_tk = self.token.calculate_token(part)
Nov 30 15:15:44 raspberrypi python[392]:   File "/home/pi/env/lib/python3.5/site-packages/gtts_token/gtts_token.py", line 28, in calculate_token
Nov 30 15:15:44 raspberrypi python[392]:     seed = self._get_token_key()
Nov 30 15:15:44 raspberrypi python[392]:   File "/home/pi/env/lib/python3.5/site-packages/gtts_token/gtts_token.py", line 57, in _get_token_key
Nov 30 15:15:44 raspberrypi python[392]:     tkk_expr = re.search(".*?(TKK=.*?;)W.*?", line).group(1)
Nov 30 15:15:44 raspberrypi python[392]: AttributeError: 'NoneType' object has no attribute 'group'

shivasiddharth, 2018-11-30:
Run the following:
```
source env/bin/activate
pip install -U gTTS
pip install -U gTTS-token
```

aro28, 2018-11-30:
pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ pip install -U gTTS
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: gTTS in ./env/lib/python3.5/site-packages (2.0.1)
Requirement already satisfied, skipping upgrade: bs4 in ./env/lib/python3.5/site-packages (from gTTS) (0.0.1)
Requirement already satisfied, skipping upgrade: click in ./env/lib/python3.5/site-packages (from gTTS) (6.7)
Requirement already satisfied, skipping upgrade: gtts-token in ./env/lib/python3.5/site-packages (from gTTS) (1.1.2)
Requirement already satisfied, skipping upgrade: six in ./env/lib/python3.5/site-packages (from gTTS) (1.11.0)
Requirement already satisfied, skipping upgrade: requests in ./env/lib/python3.5/site-packages (from gTTS) (2.20.1)
Requirement already satisfied, skipping upgrade: beautifulsoup4 in ./env/lib/python3.5/site-packages (from bs4->gTTS) (4.6.3)
Requirement already satisfied, skipping upgrade: urllib3<1.25,>=1.21.1 in ./env/lib/python3.5/site-packages (from requests->gTTS) (1.24.1)
Requirement already satisfied, skipping upgrade: chardet<3.1.0,>=3.0.2 in ./env/lib/python3.5/site-packages (from requests->gTTS) (3.0.4)
Requirement already satisfied, skipping upgrade: idna<2.8,>=2.5 in ./env/lib/python3.5/site-packages (from requests->gTTS) (2.7)
Requirement already satisfied, skipping upgrade: certifi>=2017.4.17 in ./env/lib/python3.5/site-packages (from requests->gTTS) (2018.11.29)
(env) pi@raspberrypi:~ $ pip install -U gTTS-token
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting gTTS-token
  Downloading https://www.piwheels.org/simple/gtts-token/gTTS_token-1.1.3-py3-none-any.whl
Requirement already satisfied, skipping upgrade: requests in ./env/lib/python3.5/site-packages (from gTTS-token) (2.20.1)
Requirement already satisfied, skipping upgrade: idna<2.8,>=2.5 in ./env/lib/python3.5/site-packages (from requests->gTTS-token) (2.7)
Requirement already satisfied, skipping upgrade: urllib3<1.25,>=1.21.1 in ./env/lib/python3.5/site-packages (from requests->gTTS-token) (1.24.1)
Requirement already satisfied, skipping upgrade: chardet<3.1.0,>=3.0.2 in ./env/lib/python3.5/site-packages (from requests->gTTS-token) (3.0.4)
Requirement already satisfied, skipping upgrade: certifi>=2017.4.17 in ./env/lib/python3.5/site-packages (from requests->gTTS-token) (2018.11.29)
Installing collected packages: gTTS-token
  Found existing installation: gTTS-token 1.1.2
    Uninstalling gTTS-token-1.1.2:
      Successfully uninstalled gTTS-token-1.1.2
Successfully installed gTTS-token-1.1.3

shivasiddharth, 2018-11-30:
Now try the sonoff commands, it should not crash.

aro28, 2018-11-30:
pi@raspberrypi:~ $ sudo systemctl status gassistpi.service 
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-11-30 16:19:30 +06; 1min 4s ago
 Main PID: 382 (python)
   CGroup: /system.slice/gassistpi.service
           └─382 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id xxxxxxxxxxxxx

Nov 30 16:20:35 raspberrypi python[382]:     response = meth(req, response)
Nov 30 16:20:35 raspberrypi python[382]:   File "/usr/lib/python3.5/urllib/request.py", line 582, in http_response
Nov 30 16:20:35 raspberrypi python[382]:     'http', request, response, code, msg, hdrs)
Nov 30 16:20:35 raspberrypi python[382]:   File "/usr/lib/python3.5/urllib/request.py", line 510, in error
Nov 30 16:20:35 raspberrypi python[382]:     return self._call_chain(*args)
Nov 30 16:20:35 raspberrypi python[382]:   File "/usr/lib/python3.5/urllib/request.py", line 444, in _call_chain
Nov 30 16:20:35 raspberrypi python[382]:     result = func(*args)
Nov 30 16:20:35 raspberrypi python[382]:   File "/usr/lib/python3.5/urllib/request.py", line 590, in http_error_default
Nov 30 16:20:35 raspberrypi python[382]:     raise HTTPError(req.full_url, code, msg, hdrs, fp)
Nov 30 16:20:35 raspberrypi python[382]: urllib.error.HTTPError: HTTP Error 503: Service Unavailable

aro28, 2018-11-30:
Tasmota light is triggered and than it crashes with above output.

aro28, 2018-11-30:
Sure.. 
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED
INFO:snowboy:Keyword 3 detected at time: 2018-11-30 16:32:10
Assistant is listening....
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "turn off living light"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "turn off living light"}
ERROR:__main__:HTTP Error 503: Service Unavailable
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 692, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 482, in main
    tasmota_control(str(usrcmd).lower(), name.lower(),tasmota_deviceip[num],tasmota_deviceportid[num])
  File "/home/pi/GassistPi/src/main.py", line 145, in tasmota_control
    say("Tunring off "+devname)
  File "/home/pi/GassistPi/src/actions.py", line 225, in say
    newword=trans(words,language)
  File "/home/pi/GassistPi/src/actions.py", line 219, in trans
    transword= translator.translate(words, lang)
  File "/home/pi/env/lib/python3.5/site-packages/goslate.py", line 409, in translate
    return _unwrapper_single_element(self._translate_single_text(text, target_language, source_language))
  File "/home/pi/env/lib/python3.5/site-packages/goslate.py", line 334, in _translate_single_text
    results = list(self._execute(make_task(i) for i in split_text(text)))
  File "/home/pi/env/lib/python3.5/site-packages/goslate.py", line 203, in _execute
    yield each()
  File "/home/pi/env/lib/python3.5/site-packages/goslate.py", line 332, in <lambda>
    return lambda: self._basic_translate(text, target_language, source_lauguage)[0]
  File "/home/pi/env/lib/python3.5/site-packages/goslate.py", line 251, in _basic_translate
    response_content = self._open_url(url)
  File "/home/pi/env/lib/python3.5/site-packages/goslate.py", line 191, in _open_url
    raise e
  File "/home/pi/env/lib/python3.5/site-packages/goslate.py", line 181, in _open_url
    response = self._opener.open(request, timeout=self._TIMEOUT)
  File "/usr/lib/python3.5/urllib/request.py", line 472, in open
    response = meth(req, response)
  File "/usr/lib/python3.5/urllib/request.py", line 582, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.5/urllib/request.py", line 510, in error
    return self._call_chain(*args)
  File "/usr/lib/python3.5/urllib/request.py", line 444, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.5/urllib/request.py", line 590, in http_error_default
    raise HTTPError(req.full_url, code, msg, hdrs, fp)
urllib.error.HTTPError: HTTP Error 503: Service Unavailable

shivasiddharth, 2018-11-30:
Run: 
```
source env/bin/activate
pip install git+https://github.com/shivasiddharth/py-googletrans.git --upgrade
pip uninstall goslate
```
replace your existing actions.py file with the new file from master. These errors should not appear.

aro28, 2018-11-30:
Yes. I did and now it works.. Many thanks Shiva!!!

shivasiddharth, 2018-11-30:
Will fix the hue error over this weekend.