shivasiddharth/GassistPi #636: 'OK Google' doesn't work if 'Custom_Wakeword is disabled'

pmishev, 2018-11-18:
### Describe the bug:
```
When setting Custom_Wakeword: 'Disabled', GA doesn't detect 'Ok Keyword' wake word.

When Custom_Wakeword: 'Enabled', 'OK Google', as well as the custom wake word are detected, however there is another problem then - GA is activated on almost any other random sound it hears (but this may well be unrelated)
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Yes
## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
± ~/env/bin/python3 -u ~/GassistPi/src/main.py
```

### Paste the contents of the terminal below
```
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.3
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:failed to authenticate
Initializing GPIOs 5 and 6 for assistant activity indication
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
device_model_id: magic-frame-magic-frame-oe6ock
device_id: C72D0C05179E77A8CBEF05AF8F7C771A

ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-11-18:
Such a thing is impossible to happen. Custom wakeword is independent of Ok_Google. Disable the custom wakeword, manually start the assistant and post the terminal contents here. Also share your config.yaml file.

pmishev, 2018-11-19:
I apologise, it seems the problem with 'ok google' not working to be random and I can't reproduce it again after a reboot. I suspect there might be something wrong about my audio setup.

I do experience other issues and I'm not even sure where to start looking for the problem:

1. When I activate custom hotword, it keeps activating itself by certain loud sounds. Even when it's replying to a command, it activates by its own voice, so by the time it finishes talking, it's already starting to listen again (I tried different personal and public voice models)

2. When I manually run it the output is duplicated twice or sometimes 4 times like this:
```
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
INFO:snowboy:Keyword 2 detected at time: 2018-11-19 23:00:59

Assistant is listening....
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "nevermind"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "nevermind"}
ON_NO_RESPONSE
ON_NO_RESPONSE
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

ON_MUTED_CHANGED:
  {"is_muted": true}
ON_MUTED_CHANGED:
  {"is_muted": true}
ON_MUTED_CHANGED:
  {"is_muted": true}
ON_MUTED_CHANGED:
  {"is_muted": true}
```

In the volume controls I see 2 inputs active:

[image]

Is that normal at all?
Can you please advise what I can try?

pmishev, 2018-11-19:
Furthermore, I tried running in another terminal the demo app from the snowboy binaries I downloaded elsewhere: `python2 ~/Alexa/third-party/rpi-arm-raspbian-8.0-1.1.1/demo.py ~/Alexa/jarvis.pmdl` and it doesn't give these false detections. Below is the output I get from both terminals for the same interaction:

From snowboy demo:
```
INFO:snowboy:Keyword 1 detected at time: 2018-11-19 23:11:58
```
from your script:
```
INFO:snowboy:Keyword 1 detected at time: 2018-11-19 23:11:58
Assistant is listening....
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
INFO:snowboy:Keyword 2 detected at time: 2018-11-19 23:12:00
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what's the weather in London"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what's the weather in London"}
ON_RENDER_RESPONSE:
  {"text": "Pavements may be wet", "type": 0}
ON_RENDER_RESPONSE:
  {"text": "Pavements may be wet", "type": 0}
ON_RENDER_RESPONSE:
{
  "text": "Currently in London it's 5 degrees Celsius with rain. Tonight, it's predicted to be 5 with showers.\n---\n( More on weather.com )",
  "type": 0
}
ON_RENDER_RESPONSE:
{
  "text": "Currently in London it's 5 degrees Celsius with rain. Tonight, it's predicted to be 5 with showers.\n---\n( More on weather.com )",
  "type": 0
}
ON_RESPONDING_STARTED:
  {"is_error_response": false}
ON_RESPONDING_STARTED:
  {"is_error_response": false}
INFO:snowboy:Keyword 2 detected at time: 2018-11-19 23:12:07
ON_RESPONDING_FINISHED
ON_RESPONDING_FINISHED
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

INFO:snowboy:Keyword 1 detected at time: 2018-11-19 23:12:11
Assistant is listening....
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

ON_MUTED_CHANGED:
  {"is_muted": true}
ON_MUTED_CHANGED:
  {"is_muted": true}
INFO:snowboy:Keyword 2 detected at time: 2018-11-19 23:12:13
Assistant is listening....
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
INFO:snowboy:Keyword 2 detected at time: 2018-11-19 23:12:15
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": ""}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": ""}
ON_CONVERSATION_TURN_TIMEOUT
ON_CONVERSATION_TURN_TIMEOUT
ON_MUTED_CHANGED:
  {"is_muted": true}
ON_MUTED_CHANGED:
  {"is_muted": true}

```

In my config I have:
```
Wakewords:
  Ok_Google: 'Disabled'
  Custom_Wakeword: 'Enabled'
  Custom_wakeword_models:
    - '/home/pi/snowboy_voice_models/jarvis.pmdl'
    - '/home/pi/snowboy_voice_models/pedro.pmdl'
```
You can see how there seem to be 2 assistants running at the same time - they even returned different results.

markus-nclose, 2018-11-21:
I had a similar issues regarding the random activation. I did not have snowboy enabled.
My setup has the ReSpeaker 4.

Sometimes when we just talking around it, it would activate or when doing house work.

shivasiddharth, 2018-11-21:
False positive triggering is very common, its not an issue. Even my original Google home starts listening for some weird reason occasionally.