shivasiddharth/GassistPi #638: Assistant only start after the installation but then it doesn't work

SilkierLyric146, 2018-11-19:
When i have installed the assitant with the command  sudo  ./GassistPi/scripts/gassist-installer-pi3.sh and complete the procedure the assistant worked perfectly but then when i try to launch it manually it report this error : 
 
pi@raspberrypi:~/GassistPi/systemd $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'rpi-zephyr-ai-rpi-zephyr-2kzz5r' 

/home/pi/env/lib/python3.4/site-packages/cryptography/hazmat/bindings/openssl/binding.py:163: CryptographyDeprecationWarning: OpenSSL version 1.0.1 is no longer supported by the OpenSSL project, please upgrade. A future version of cryptography will drop support for it.
  utils.CryptographyDeprecationWarning
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/snowboydetect.py", line 15, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.4/importlib/__init__.py", line 109, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 2254, in _gcd_import
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2226, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1191, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1161, in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 539, in _check_name_wrapper
  File "<frozen importlib._bootstrap>", line 1715, in load_module
  File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
ImportError: libpython3.5m.so.1.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 31, in <module>
    import snowboydecoder
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 5, in <module>
    import snowboydetect
  File "/home/pi/GassistPi/src/snowboydetect.py", line 18, in <module>
    _snowboydetect = swig_import_helper()
  File "/home/pi/GassistPi/src/snowboydetect.py", line 17, in swig_import_helper
    return importlib.import_module('_snowboydetect')
  File "/usr/lib/python3.4/importlib/__init__.py", line 109, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: libpython3.5m.so.1.0: cannot open shared object file: No such file or directory

Can someone help me? please

shivasiddharth, 2018-11-19:
Which board and which version of the os are you using ?

SilkierLyric146, 2018-11-19:
Raspberry pi 3 model B 

Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie

shivasiddharth, 2018-11-19:
You need to be on the Stretch version of Raspbian for this project to work with full functionality. I dont support/troubleshoot installations on Jessie.

closed 2018-11-19 by shivasiddharth

SilkierLyric146, 2018-11-19:
Okok thanks for the answer i will upgrade my pi os to stretch and then I'll try again.