shivasiddharth/GassistPi #639: Assistant service stops working when playback from Radio or Youtube is requested

Darmur, 2018-11-20:
### Describe the bug:
Yesterday I managed to have a fresh install of the package on top or latest Raspbian Lite, following the instructions reported in the README. After reading the old tickets, I solved the issue related to Youtube playback (replaced gtoken.py) and now I can trigger song playback with voice command (OK Google, play some Queen from youtube).

Playback is started properly in background (I can hear the music), but the service has a crash showing an exception, and it does not capture anymore the voice commands (push button to trigger and disable the microphones / stop VLC is still working fine).

I have no problem to ask questions or to trigger simple commands (OK Google, set volume at 50%), as soon as I trigger a playback the assistant in a faulty state and is not responsive anymore.

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
play some Queen from YouTube
```

### Paste the contents of the terminal below
```
pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id saved-model-id
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.3
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:failed to authenticate
AIY-HAT
Initializing GPIO 17 for assistant activity indication
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
[... 118 lines cut ...]
ERROR:__main__:Expecting value: line 1 column 1 (char 0)
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 673, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 446, in main
    self.process_event(event)
  File "/home/pi/GassistPi/src/main.py", line 321, in process_event
    oldvolume= json.load(vol)
  File "/usr/lib/python3.5/json/__init__.py", line 268, in load
    parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
  File "/usr/lib/python3.5/json/__init__.py", line 319, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.5/json/decoder.py", line 339, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.5/json/decoder.py", line 357, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
^Z
[3]+  Stopped                 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id saved-model-id
pi@raspberrypi:~ $

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.
I could not find my log in the /tmp/ directory

shivasiddharth, 2018-11-21:
This is not the correct way to update it. Please check https://github.com/shivasiddharth/GassistPi/issues/618#issuecomment-436975900

nclbill, 2018-11-21:
tks its working you are the best