shivasiddharth/GassistPi #643: [RPi0][pushbutton.py] Only way to trigger assistant on RPi0+pushbutton.py is with push button

Darmur, 2018-11-22:
### Describe the bug:
```
According to the readme:

_CUSTOM WAKEWORD ACTIVATION
7. Users using pushbutton.py or Pi Zero users have an option between using custom wakeword and GPIO trigerring. If custom wakeword is enabled, then GPIO trigger will not work. To enable GPIO triggering, set custom wakeword to 'Disabled'._

I do have a RPi0 running pushbutton.py, I tried to enable custom wake-word to disable GPIO triggering, but I can still trigger pressing the button. I also tried to trigger using the custom wake-words, playing the .wav file from the PC (to avoid issues with my pronunciation), but I never got the assistant triggered.

Ok Google does not work as well, as supposed to be.

I tried to have a look to the code, and the only check for custom_wakeword=True is to call the function start_detector(). I put a debug print just before the call of that function, and I never got it printed... this means the function start_detector() is never invoked by the script.

There is no other check for custom_wakeword=False to disable GPIO triggering.

Is it the readme outdated, or did I spotted an issue?

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Yes/No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-11-22:
Did you change the config.yaml after starting the pushbutton.py ?

jbrepogmailcom, 2019-07-19:
I have the same problem, but in config.yaml I have "Enabled" (correctly with "d"). Yet it is not working. Should the hotword wakeup work with the service that starts at boot, or do I need to run pusbutton.py? And for pushbutton.py, do I need to set up python environment? Thank you