shivasiddharth/GassistPi #644: Found a typo in main.py

exnerd, 2018-11-22:
### Describe the bug:
```
When setting the brightness of a lamp in Domoticz, the command does not work and the error is 'list index of out range'.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
Turning on Lamp one .
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
ERROR:__main__:list index out of range
Traceback (most recent call last):
  File "./GassistPi/src/main.py", line 675, in <module>
    Myassistant().main()
  File "./GassistPi/src/main.py", line 482, in main
    domoticz_control(str(usrcmd).lower(),configuration['Domoticz']['Devices']['Id'][i],configuration['Domoticz']['Devices']['Name'][i])
  File "/home/pi/GassistPi/src/actions.py", line 1395, in domoticz_control
    if domoticz_devices['result'][i]['HaveDimmer']:
IndexError: list index out of range

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

exnerd, 2018-11-22:
Found the typo on line 1395 and probably on line 1383. [i] should be replaced by [j].

shivasiddharth, 2018-11-22:
Can you try replacing the `i` with `devorder` `[i]` should be replaced with `[devorder]`. Let me know if it works, i shall update the master then.

exnerd, 2018-11-23:
@shivasiddharth  your solution worked. Keep up the work. I had some difficulties getting the domoticz integration working, but now it's working. I can now explore further the possibilities of this project.

Regards

closed 2018-11-23 by exnerd