shivasiddharth/GassistPi #648: gassistpi keeps restarting how to crash autostart- raspberrypi 3 B+

ykr412, 2018-11-23:
### Describe the bug:
it keeps restarting with the bootup music 
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No
yes
### 2. Is the issue related to assistant installation ?
Yes/No
no
## If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No
no
### 4. Is the assistant service restarting automatically?
Yes/No
yes
## If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

### Attach the log file. Log file can be found in /tmp directory. Remeber to copy your log file before shutting down or restarting the Pi.

shivasiddharth, 2018-11-23:
There is literally no information here.

ykr412, 2018-11-23:
I did not understant what information you needed i just have a problem with
the auto boot up the entrance music just repeating and i want to crash the
auto start to manually boot it up as i read . Please can you help me

shivasiddharth, 2018-11-23:
> 4. Is the assistant service restarting automatically?
> Yes/No
> yes
> 
> If answer to question 4 is Yes, then paste the contents of the terminal below.

ykr412, 2018-11-25:
pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'replace this with the model id'

INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.3
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:failed to authenticate
[0187d3e8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[018760d8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
CUSTOM-VOICE-HAT
Initializing GPIO 25 for assistant activity indication
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
device_model_id: replace this with the model id
device_id: EC61B385B85483A16D366E4259FB48EA

    This device is not registered. This means you will not be able to use
    Device Actions or see your device in Assistant Settings. In order to
    register this device follow instructions at:

    https://developers.google.com/assistant/sdk/guides/library/python/embed/register-device

ON_MUTED_CHANGED:
  {"is_muted": true}
ON_MUTED_CHANGED:
  {"is_muted": true}
ON_START_FINISHED
ON_START_FINISHED

ykr412, 2018-11-25:
pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'xxxxxxxxxxxxxxxxxxxxxxxxx'
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.3
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:failed to authenticate
[02569cd8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[025761f0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
CUSTOM-VOICE-HAT
Initializing GPIO 25 for assistant activity indication
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
device_model_id: enduring-honor-222821-gassitpi-0i5lhg
device_id: EC61B385B85483A16D366E4259FB48EA

ON_MUTED_CHANGED:
  {"is_muted": true}
ON_MUTED_CHANGED:
  {"is_muted": true}
ON_START_FINISHED
ON_START_FINISHED

alexkoppelman, 2018-11-27:
I have a very similar problem, Gassistpi crashes right after starting and restarts. The log output is;

Nov 27 22:30:14 skippy systemd[1]: Started Google Assistant-Voice.
Nov 27 22:30:21 skippy python[1259]: INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
Nov 27 22:30:21 skippy python[1259]: INFO:gmusicapi.Mobileclient1:version: 11.0.4
Nov 27 22:30:21 skippy python[1259]: INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
Nov 27 22:30:21 skippy python[1259]: INFO:gmusicapi.Mobileclient1:initialized
Nov 27 22:30:21 skippy python[1259]: INFO:gmusicapi.Mobileclient1:logged out
Nov 27 22:30:21 skippy python[1259]: Traceback (most recent call last):
Nov 27 22:30:21 skippy python[1259]:   File "/home/pi/GassistPi/src/main.py", line 44, in <module>
Nov 27 22:30:21 skippy python[1259]:     from actions import say
Nov 27 22:30:21 skippy python[1259]:   File "/home/pi/GassistPi/src/actions.py", line 15, in <module>
Nov 27 22:30:21 skippy python[1259]:     from mediaplayer import api
Nov 27 22:30:21 skippy python[1259]:   File "/home/pi/GassistPi/src/mediaplayer.py", line 21, in <module>
Nov 27 22:30:21 skippy python[1259]:     logged_in = api.login(configuration['Gmusicapi']['email'],configuration['Gmusicapi']['password'] , configuration['Gmusicapi']['deviceid'])
Nov 27 22:30:21 skippy python[1259]: KeyError: 'Gmusicapi'
Nov 27 22:30:21 skippy systemd[1]: gassistpi.service: Main process exited, code=exited, status=1/FAILURE
Nov 27 22:30:21 skippy systemd[1]: gassistpi.service: Unit entered failed state.
Nov 27 22:30:21 skippy systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Nov 27 22:30:22 skippy systemd[1]: gassistpi.service: Service hold-off time over, scheduling restart.
Nov 27 22:30:22 skippy systemd[1]: Stopped Google Assistant-Voice.

My ~/GassistPi/systemd/gassistpi.service

Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'XXX-XXX-XXX'
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

The Gmusicapi is installed and is version 11.0.4

I have not experienced any issues during the install process and this error occurs when running as a scheduled service as well as when I start it maunally. The googlesamples-assistant-hotword works without any poblems.

alexkoppelman, 2018-11-27:
Ok, apologies! After quite a bit of googling without luck I've removed the full git clone and reinstalled from scratch. This resolved the issue and it's working fine now. 
This makes me think I've screwed up the config.yaml or actions.py files

shivasiddharth, 2018-11-29:
@ykr412 Your assistant is starting normally, check the service and see if you have entered the model-id correctly.

ykr412, 2018-11-30:
INFO:snowboy:Keyword 2 detected at time: 2018-11-26 17:18:22
Turning on the microphone
INFO:snowboy:Keyword 2 detected at time: 2018-11-26 17:18:28
INFO:snowboy:Keyword 2 detected at time: 2018-11-26 17:18:38
Turning off the microphone
Turning on the microphone
Stopped
stopping vlc
INFO:snowboy:Keyword 2 detected at time: 2018-11-26 17:18:49
Stopped
stopping vlc

ykr412, 2018-11-30:
i fixed the issue that made it restart , but know the entrance music keeps repeating and when i say ok google it says sorry something went wrong try in a few seconds.

ykr412, 2018-11-30:
ON_CONVERSATION_TURN_STARTED
ON_RESPONDING_STARTED:
  {"is_error_response": true}
ON_RESPONDING_STARTED:
  {"is_error_response": true}
ON_RESPONDING_FINISHED
ON_RESPONDING_FINISHED
ON_ASSISTANT_ERROR:
  {"is_fatal": false}
ON_ASSISTANT_ERROR:
  {"is_fatal": false}

closed 2018-12-10 by ykr412

Prabindas001, 2020-08-24:
i am facing the same issue. my assistant keeps restarting just after playing the boot music. what should i do?

aydogmusserhat, 2021-01-08:
yes, me too, i install clean but always restarting while start music begins.