shivasiddharth/GassistPi #651: Missing File Error Redo

unsigneduk, 2018-11-25:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Describe the bug:
```
File credentials.json (file created during instalation) is not found by the assistant after startup. Even though Ive confirmed it is in the correct place and have tried trasnfering it to /root/.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No - unlikely, but could be.

## If answer to question 2 is Yes, then paste the contents of the terminal below.

```pi@raspberrypi:~ $ sudo /home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id ************** --device-model-id ******************************************
Initializing GPIOs 5 and 6 for assistant activity indication
ERROR:root:Error loading credentials: [Errno 2] No such file or directory: '/root/.config/google-oauthlib-tool/credentials.json'
ERROR:root:Run google-oauthlib-tool to initialize new OAuth 2.0 credentials.
pi@raspberrypi:~ $ sudo su
root@raspberrypi:/home/pi# cp -r /root/.config/* /home/pi/.config/
root@raspberrypi:/home/pi#

```

## If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes - it seems so but fails soon after, after attempting to start manually the above error apears. 

### 4. Is the assistant service restarting automatically?
No - not as far as im aware, the start up sound and "hi" is only heard once. 

## If answer to question 4 is Yes, then paste the contents of the terminal below.

```See above. 

```

## If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```as above.  PLEASE see Error #88 for further deatails on my error. 

Further more I think this is hardware related, by that I mean GassistPi wont work on my set up or some of those similar, my set up is - PiZeroW with Respeaker 2 mic Hat and zero4U usb hub. Raspian (9) 

Apologies @Shivasiddharth I hadnt another option other than to open a new issue regarding this as ther was no way of highlighting new comments on the already open (but seemingly) solved case. 

Also Thanks and once its working looks like its a fantasic piece of work.

shivasiddharth, 2018-11-25:
Your issue is not related to #88. In #88 you have posted the command to manually start the assistant as:

`sudo /home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id ************** --device-model-id ******************************************` 

You have used sudo to start the assistant that is why its looking for credentials in the root directory. Please see the command that i have given to [manually start] (https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant) , that has no sudo . Its a common error that people commit by adding sudo for all commands.

Pi Zero right now is under some turmoil, please check the issue #622 Frequent crashing is due to the way Google is capturing the audio streams. Its an upstream issue and unfortunately all that i can do is sit and wait for Google to fix.

unsigneduk, 2018-11-26:
Thanks for the quick reply Shivasiddharth. Schoolboy error on my part !! - Ill keep my eye out for any changes. At the moment it seems to be still running but unreponsive upon startup and manual start (using the correct commend) is having no effect either. Fingers crossed. Ill try a few things along the lines of what @colin1964 has tried in issue #622 and report back any findings. Thanks again.

closed 2018-11-26 by unsigneduk