shivasiddharth/GassistPi #657: Errors after upgrade/update

IngmarG, 2018-11-29:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```

### Describe the bug:
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No

### 2. Is the issue related to assistant installation ?
Yes/No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

shivasiddharth, 2018-11-29:
> ### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
> Yes/No

Duplicate of #534 . This is why sticking to issue template is important.

closed 2018-11-29 by shivasiddharth

IngmarG, 2018-11-29:
Do you mean the update/upgrade also changes the environment? Because I was one of the first contributors to that issue 534. I already made the changes AND already did an update/upgradeoir two.

shivasiddharth, 2018-11-29:
Sorry, i did not see that you had been a part of that discussion. Existing file was manually overwritten, so yes updates replace the overwritten files. If you are ok with testing a newer method of translation, that will not create this error, i shall reopen and relabel this issue. We can close #534 once and for all.

IngmarG, 2018-11-29:
What do you mean with " a newer method of translation" ?

Always open to test something new. 

On the issue:
I thought hat was exactly the point of an environment: to preserve it from changes.
I'll apply the patches and see what happens.

shivasiddharth, 2018-11-29:
"newer method" meaning a python package alternative to googletrans.

shivasiddharth, 2018-11-29:
I was of the thought that you ran the update within the environment.

IngmarG, 2018-11-29:
No, just in the main console. That's why I was surprised that I got this error again.

shivasiddharth, 2018-11-29:
Within the environment install goslate using:
```
source env/bin/activate
pip install goslate
```
Replace your existing actions.py file with the updated one from master. See if it works.

IngmarG, 2018-11-29:
My actions.py is a completely stripped down version of yours (because I just needed it to control a camera. no Kodi, no KS,l no Hue, no ...).
So i'll have to compare my lines to your new ones.
I'll work on it this evening and let you know (in this thread).

IngmarG, 2018-11-30:
Didn't work for me.
What did work was updating to the latest gTTS-Token (v1.1.3) (https://pypi.org/project/gTTS-token/) and a forked GoogleTrans which takes that gTTS-Token into account (https://github.com/BoseCorp/py-googletrans)

Sadly enough, I'm running into another issue now:
ERROR:__main__:'charmap' codec can't encode character '\u2019' in position 33: character maps to <undefined>
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 354, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 337, in main
    self.process_event(event)
  File "/home/pi/GassistPi/src/main.py", line 170, in process_event
    print(event)
  File "/usr/lib/python3.5/encodings/iso8859_15.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\u2019' in position 33: character maps to <undefined>

shivasiddharth, 2018-11-30:
usually, unicode errors are related to google's cloud. Try giving it sometime.