shivasiddharth/GassistPi #663: Deactivate spotify

DewGew, 2018-11-30:
How can I deactivate custom Spotify function?
I have Spotify already connected in my Google home app.

IngmarG, 2018-11-30:
From main.py, delete (or comment) the occurences of spotify (64, 673-676).

Haven't tested that, but it's kind of obvious

DewGew, 2018-12-01:
Thanks @IngmarG To deactivate functions I dont need and I dont want them to mess up things I have to change those in main.py? Or do I have to change any More files?

shivasiddharth, 2018-12-01:
Its just main.py. In the next update i will add switches in config.yaml for all the custom functions.

DewGew, 2018-12-01:
Great thanks..

closed 2018-12-01 by DewGew

shivasiddharth, 2018-12-01:
@DewGew , Replace your existing main.py and config.yaml with the files from this branch https://github.com/shivasiddharth/GassistPi/tree/custom-actions-switch  and check if the switch control works. You turn on or off the customs by setting the control to 'Enabled' or 'Disabled'.

IngmarG, 2018-12-01:
Haven't tested it, but this is great.

DewGew, 2018-12-03:
Seems to work..
But I think you have a 2 switches on kodi:
`Kodi:
  Kodi_Control: 'Enabled'
  control: 'Disabled'`

shivasiddharth, 2018-12-04:
I have fixed that now.

DewGew, 2018-12-05:
This works great.. 👍  merge this to master.