shivasiddharth/GassistPi #669: Force audio to HDMI in startup

shabdhu, 2018-12-02:
I have connected my speakers through hdmi audio output. The problem is I have to change the default audio from analog to hdmi eachtime during startup. And through SSH, changing audio to hdmi doesnt work for some reason(using raspi-config to force audio through hdmi or using "amixer cset numid=3 2"). The only way seems to connect a hdmi display and manually change the audio to hdmi using the taskbar audio icon. Not sure what's the issue here. Please help.

shabdhu, 2018-12-02:
but i had run the script for aux audio out previously. So though i run the script for hdmi audio out, the issue still persists. any idea how to roll back/uninstall the scripts i installed for aux audio out

shivasiddharth, 2018-12-02:
ou roll back or uninstall drivers only in I2S audio. Just run the scripts for HDMI that i have given.