shivasiddharth/GassistPi #675: High CPU when idle.  RPI 3b

williamhemmingsen, 2018-12-04:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.  
``` 
pi@raspberrypi:~/GassistPi/src $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a22082
Serial          : 000000000c387830
pi@raspberrypi:~/GassistPi/src $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~/GassistPi/src $

[... 100 lines cut ...]
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_direct.c:977:(snd1_pcm_direct_initialize_slave) requested or auto-format is not available
ALSA lib pcm_dmix.c:1064:(snd_pcm_dmix_open) unable to initialize slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
device_model_id: smartmirror-a061c-smartmirror-ns47oz
device_id: 4316D79BDB30D1A757D729D3C03BFFB0

ON_MUTED_CHANGED:
  {"is_muted": false}
ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED

```

williamhemmingsen, 2018-12-04:
Custom keyword is disabled, and the high cpu is even running manually.

Regards,

williamhemmingsen, 2018-12-04:
I ran some test.
Custom keyword off and OK Google on = ~30%
Custom keyword on and OK Google off = ~30%
Both off = ~30%

So maybe that will help.  As you can see, python is using almost 10x the CPU as Alexa.

x Top Processes Procs=141-mode=3-1=Base 3=Perf 4=Size 5=I/O[RootOnly] u=Argsqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqx
x  PID        %CPU      Size       Res      Res       Res       Res      Shared   Faults   Faults Command                                                                                                                                   x
x              Used        KB       Set      Text      Data       Lib        KB      Min      Maj                                                                                                                                           x
x     2244    113.3    383496    114020      3424    286148         0     33728        0        0 python                                                                                                                                    x
x     1061     13.8    530728     40240      2584    387596         0     23324        0        0 SampleApp                                                                                                                                 x
x     2164      1.0      6284      3940       156      3700         0      1668        0        0 nmon                                                                                                                                      x
x      477      0.5    115568     22000      1896     24292         0     13756        0        0 Xorg                                                                                                                                      x
x        1      0.0     28072      3092       828     17788         0      2612        0        0 systemd                                                                                                                                   x
x        2      0.0         0         0         0         0         0         0        0        0 kthreadd

cnoorman, 2018-12-04:
Same experience here: GassistPi uses about 30% CPU on RPi-3B (running manually). The nice thing is that it remains about 30% when running as a service. It is independent of use of custom wakeword.
CPU usage for Alexa increases from ~5% (running manually) to ~70% (running as a service). 
That can be solved with [this](https://github.com/alexa/avs-device-sdk/issues/539#issuecomment-429656178). Unfortunately this solution does not help for excess usage of GassistPi.

williamhemmingsen, 2018-12-04:
> 
> 
> Same experience here: GassistPi uses about 30% CPU on RPi-3B (running manually). The nice thing is that it remains about 30% when running as a service. It is independent of use of custom wakeword.
> CPU usage for Alexa increases from ~5% (running manually) to ~70% (running as a service).
> That can be solved with [this](https://github.com/alexa/avs-device-sdk/issues/539#issuecomment-429656178). Unfortunately this solution does not help for excess CPU usage of GassistPi.

The constant 30% really heats up the PI.

williamhemmingsen, 2018-12-04:
Found the source of the CPU issue, but no idea how to solve.  The source is somewhere in the GPIO code.
If you manually set GPIO=None in main.pi the CPU normalizes to about ~5-10% utilization.  About the same resource used by Alexa.

shivasiddharth, 2018-12-05:
I have multiple threads running in the Google Assistant, one for the GPIO (Stop and mute) and another for Custom Wakeword, both of them are critical to the project and cannot be got rid of. Actual plan was to run two threads separately for Muting and Stop functions, due the predicted rise in CPU usage, i have combined both inside a single thread.

williamhemmingsen, 2018-12-05:
For my project, I just disabled GPIO and wrote my own indicator script.  For reference this is the CPU average with Alexa, Google Assistant, And Magic Mirror running. ~20-25% (totall all CPU's)  A huge improvement.  As you can see with GPIO disabled, GassistPi is using the same amount of CPU as Alexa, a 5X improvement.

x Top Processes Procs=146-mode=3-1=Base 3=Perf 4=Size 5=I/O[RootOnly] u=Argsqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqx
x  PID        %CPU      Size       Res      Res       Res       Res      Shared   Faults   Faults Command                                                                                                                                   x
x              Used        KB       Set      Text      Data       Lib        KB      Min      Maj                                                                                                                                           x
x      402     21.3    372504     82216      3424    277016         0     14384        0        0 python                                                                                                                                    x
x      638     20.8    528976     32076      2584    384140         0     21228        0        0 SampleApp                                                                                                                                 x
x      993     17.3    604064    295096     53132    215128         0    250936      661        0 electron                                                                                                                                  x
x      988     15.8    287164     29712     53132     83332         0     13260      939        0 electron                                                                                                                                  x
x      913     10.4    728612     67780     53132    303096         0     22384        0        0 electron                                                                                                                                  x
x      477      3.5    100008     10572      1896     22256         0      6748        0        0 Xorg                                                                                                                                      x
x     1296      1.0      6284      4012       156      3700         0      1636        0        0 nmon                                                                                                                                      x
x        7      0.5         0         0         0         0         0         0        0        0 ksoftirqd/0                                                                                                                               x
x      709      0.5    140388      9744       116     70188         0      6640        0        0 lxpanel                                                                                                                                   x
x     1333      0.5         0         0         0         0         0         0        0        0 kworker/0:0                                                                                                                               x
x        1      0.0     28128      4540       828     17844         0      3836        0        0 systemd

Hoeby, 2018-12-06:
i saw this issue. Looked at the cpu usage of my pi. 
I uses 28~30% at idle.

To try i put a # in front of "import RPi.GPIO as GPIO" in the main.py
This drops the cpu to 4~5% at idle.

@Williamhemmings:
Could you post your script that you wrote?
I would like to use the mic mute and LED for speak/lissening, these need the GPIO function

williamhemmingsen, 2018-12-07:
> i saw this issue. Looked at the cpu usage of my pi.
> I uses 28~30% at idle.
>
> To try i put a # in front of "import RPi.GPIO as GPIO" in the main.py
> This drops the cpu to 4~5% at idle. [...]

Sorry, I'm not using GPIO at all.  I'm just using HTTP Post in my script to display a gif image when google is listening/speaking on a Magic Mirror.

Hoeby, 2018-12-07:
> Sorry, I'm not using GPIO at all. I'm just using HTTP Post in my script to display a gif image when google is listening/speaking on a Magic Mirror.

What a shame for us.
Hopefully shivasiddharth will take a look at it. 
When using google, most of the answers about CPU usage and GPIO is missing of the sleep command after a loop or if function

Hoeby, 2018-12-14:
On gitter there was an answer from user MCU93US.
He adviced to add **time.sleep(0.1)** in main.py, just underneat **while mutestopbutton:** 

I had import RPi.GPIO as GPIO disabled in the main.py, to get the CPU down.
First i enabled it again and restarted the services. After that the CPU was 28%, tested the mic ON/OFF function. It worked, but with a high CPU use.
Modified the main.py and restarted the services. After that i have a CPU of 5%, tested the mic ON/OFF function. Still works. Looks like the mod. is working.

My changed main.py part. Look at line 3 (it is around line 200)
```
def pushbutton(self):
        while mutestopbutton:
            time.sleep(0.1)
            if GPIO.event_detected(stoppushbutton):
                GPIO.remove_event_detect(stoppushbutton)
                now = time.time()
                count = 1
                GPIO.add_event_detect(stoppushbutton,GPIO.RISING)
                while time.time() < now + 1:
                     if GPIO.event_detected(stoppushbutton):
                         count +=1
                     time.sleep(.25)
                if count == 2:
                    self.buttonsinglepress()
                    GPIO.remove_event_detect(stoppushbutton)
                    GPIO.add_event_detect(stoppushbutton,GPIO.FALLING)
                elif count == 3:
                    self.buttondoublepress()
                    GPIO.remove_event_detect(stoppushbutton)
                    GPIO.add_event_detect(stoppushbutton,GPIO.FALLING)
                elif count == 4:
                    self.buttontriplepress()
                    GPIO.remove_event_detect(stoppushbutton)
                    GPIO.add_event_detect(stoppushbutton,GPIO.FALLING)

```

closed 2018-12-14 by shivasiddharth

gartensofa, 2019-03-13:
Hey guys,
i hope it is within protocol to open this back up here.
Gassist's Python process constantly takes up 60-80% CPU.
The GPIO-sleep is already implemented. Could anyone advise me on how to debug this?
System:
Raspberry Pi 2 Model B Rev 1.1
Linux rpi-kodi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
Raspbian GNU/Linux 9 (stretch)

commit 6d0fbfc1 (2018-12-14): Updated Say fucntion, main.py and pushbutton.py

Fixes #675  Fixes #690

1. Added a switch for major custom actions. Users can toggle the custom actions "ON" or "OFF".

2. Added a feature to allow users to change the say function voice to either male or female.

3. Fixed Pi CPU usage at idle.
PR #691 (2018-12-14): Custom actions switch
Fixes #675  Fixes #690

1. Added a switch for major custom actions. Users can toggle the custom actions "ON" or "OFF".

2. Added a feature to allow users to change the say function voice to either male or female.

3. Fixed Pi CPU usage at idle.
commit 61a8acb0 (2018-12-14): Custom actions switch (#691)

* Enable switches for custom actions

Users can turn on or off the custom actions by setting the control for respective custom actions in config.yaml to 'Enabled' or 'Disabled'

* Fix Kodi control repeat

* Updated Say fucntion, main.py and pushbutton.py

Fixes #675  Fixes #690

1. Added a switch for major custom actions. Users can toggle the custom actions "ON" or "OFF".

2. Added a feature to allow users to change the say function voice to either male or female.

3. Fixed Pi CPU usage at idle.

* Update README.md