shivasiddharth/GassistPi #677: Was there a change to the HEADLESS AUTOSTART on BOOT SERVICE SETUP?

cdmack, 2018-12-05:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   

processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 0000000026e0905b

### Describe the bug:
```
I have been using the headless start for about a year. Ive needed to turn it on and off while using facial recognition software due to the high cpu demand (cant run both), but recently I have not been able to disable the service with "sudo systemctl disable gassistpi.service " and "sudo systemctl stop gassistpi.service".

I am prompted with "Failed to disable unit: No such file or directory" and "Failed to stop gassistpi.service: Unit gassistpi.service not loaded."
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Yes

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```
The service works as it should, but I am trying to stop the service and the normal disable and stop no longer work for me.
```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

shivasiddharth, 2018-12-06:
Are you sure , you ran the systemd service installer?  After running the service installer, it should be enabled. Looks like it has not been enabled.

fragolinux, 2018-12-29:
problems here, too...
```
pi@raspberrypi:~ $ cd /home/${USER}
pi@raspberrypi:~ $ sudo chmod +x ./GassistPi/scripts/service-installer.sh
pi@raspberrypi:~ $ sudo ./GassistPi/scripts/service-installer.sh
pi@raspberrypi:~ $ sudo systemctl enable gassistpi.service
Created symlink /etc/systemd/system/multi-user.target.wants/gassistpi.service → /lib/systemd/system/gassistpi.service.
pi@raspberrypi:~ $ sudo systemctl start gassistpi.service
Failed to start gassistpi.service: Unit gassistpi.service is not loaded properly: Invalid argument.
See system logs and 'systemctl status gassistpi.service' for details.
pi@raspberrypi:~ $ sudo systemctl start gassistpi.service
Failed to start gassistpi.service: Unit gassistpi.service is not loaded properly: Invalid argument.
See system logs and 'systemctl status gassistpi.service' for details.
pi@raspberrypi:~ $ systemctl status gassistpi.service
● gassistpi.service - Google Assistant-Voice
   Loaded: error (Reason: Invalid argument)
   Active: inactive (dead)

Dec 29 21:08:22 raspberrypi systemd[1]: gassistpi.service: Service lacks both ExecStart= and ExecStop= setting. Refusing.
pi@raspberrypi:~ $ cat GassistPi/systemd/gassistpi.service
[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

```

shivasiddharth, 2018-12-29:
Did you run the installer multiple times? The line with the execute command is missing. Please check the service file from the master. You can replace the existing service file from master, and remove the line with main.py and just retain the line with pushbutton.py.
After that disable the service, re-run the service installer.

fragolinux, 2018-12-29:
unfortunately yes, it was stuck at grpcio and i hit ENTER to see if something was still working or stuck... but so the installer "saved" that "enter" for when it asked to visit the url to get the authcode and failed... so i rerun, it went waaaaaay faster than before, i waited for the url, got the authcode, put there, made the audio tests, all fine... so, i just need this, with proper mods in there, and i'm fine?
https://github.com/shivasiddharth/GassistPi/blob/master/systemd/gassistpi.service
and for changing language, i can change it now in src/config.yaml or need to recompile?
and as i'm on pi0w, no mqtt, right?
when run manually, it does not react to "ok googlel", and if i press the button on the aiy-hat kit, it says the following instead of being ready to answer:
```
Initializing GPIO 25 for assistant activity indication
Stopped
stopping vlc
```

shivasiddharth, 2018-12-29:
For the systemd files, yup those mods and you should be fine.  For changing the language, add a `--lang `argument along with the modelid and device id. The language change in the config.yaml is only for the custom say function. to change the language of the assistant, you need to use the --lang argument.

The pushbutton.py/pi zero assitsant comes with two different methods of assistant activation. 1. custom wakeword 2. Pushbutton They are mutually exclusive so you can have only one at a time. The default custom wakewords are alexa and snowboy. If you want a "Ok google" wakeword, you need to create your own model in snowboy. The pushbutton trigger is on GPIO 22 and the AIY's GPIO 23 is set as a stop button for music. You can interchange them in the config.yaml.

And yes Pizero does not support MQTT

closed 2018-12-29 by shivasiddharth

fragolinux, 2018-12-29:
i added " --lang 'it' " to the service, disabled previous, reinstalled new, enabled new, started new, in systemctl status i see that it started with the --lang parameter, but once it produces sound, it says HI instead of "ciao" (hi in italian)...
and i swapped 22 and 23 in config.yaml with no results, in systemctl status i continue to see vlc stopped and google don't start asking or answering... i even disabled both youtube and gmusicapi in config.yaml but i see warnings in systemctl status about missing json/api/keys
need to recompile after changing the config.yaml?
man, thanks for your work and your help, very appreciated, tell me how can do a donation :)

fragolinux, 2018-12-29:
if i say "alexa", both my echo dot and the rpi start flashing (rpi the aiykit led in the button), but if i ask something in italian, only "real" alexa answers, while the rpi continues blinking with the button... and does not react saying "snowboy", to prevent actual alexa to answer... 
and... how can i create a model for snowboy? Thanks

shivasiddharth, 2018-12-29:
Never mind the warnings, they are caused by the import of the modules/packages. The language code that you have given for the argument is wrong, for the language codes, refer to this page https://developers.google.com/assistant/sdk/reference/rpc/languages. The starting Hi is a pre-generated tone, I never found time to generate tones for other languages. Will do it sometime in the near future. The models that i have provided for snowboy are universal models, its always better to generate your own voice models and use. Visit this page to create your own voice model https://snowboy.kitt.ai/dashboard Remember to have a noise free environment while generating a voice model and try to use the existing pi setup itself that will give a better accuracy while triggering. 
For contributions to the project, you can use the donation button on the project page (https://github.com/shivasiddharth/GassistPi)

fragolinux, 2018-12-29:
oh, ok, i was thinking the code was the same as in the config.yaml
i'll do my snowboy model tomorrow, thanks
and for the other problems? Even if i say alexa, it does not answer, and keeps blinking the aiy led in the top button... and swap 22 and 23 in config.yaml didn't produce anything... need recompile?
donation done! :)

fragolinux, 2018-12-29:
how can i check logs... now that i set lang to "it-IT", when it starts it plays the sound various times, as it crashes while playing it or the like, and after 3 or 4 times (delayed about 30 seconds 1 with each next) it finally completes the startup sound and says HI... but finally it now answers if i ask something :)

shivasiddharth, 2018-12-29:
I have fixed the issue in the last commit https://github.com/shivasiddharth/GassistPi/commit/5cc2e91b928553c0730b15f1a0f8a68cd3987971 from now on, even if the installer is re-run several times, that should not affect the service file.

fragolinux, 2018-12-30:
running in console i get this, the vlc part goes on indefenitely, i had to kill the process from other ssh session (as you see i tried a ctrl-c without success):
```
pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'pi0w-aiy-XXXX'  --device-model-id 'pi0w-aiy-XXXX-model' --lang 'it-IT'
/home/pi/env/lib/python3.5/site-packages/gmusicapi/__init__.py:4: GmusicapiWarning: gmusicapi.clients.OAUTH_FILEPATH is deprecated and will be removed; use Musicmanager.OAUTH_FILEPATH
  from gmusicapi.clients import Webclient, Musicmanager, Mobileclient
/home/pi/env/lib/python3.5/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2018-12-30 08:25:24,094 - gmusicapi.Mobileclient1 (shared:183) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
[01a48ac8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01a4aed0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
AIY-HAT
Initializing GPIO 25 for assistant activity indication
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
^CStopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
WARNING:root:SoundDeviceStream read overflow (3200, 6400)
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
[... 251 lines cut ...]
    - 'Desk Light'
    - 'Table Light'
  ipaddresses:
    - '192.168.1.36'
    - '192.168.1.35'
  portID:
    - '1'
    - '1'

#Voice for the "SAY" function or text to speech. Set it to 'Female' or 'Male' depending on your preference.
#Note: This does not change Google's voice. this is only for the speech generated within custom actions.
Voice_Custom_Actions: 'Female'

#Custom wakeword On/Off.
#To enable custom wakeword, change status to Enabled and restart the assistant.
Wakewords:
  Ok_Google: 'Enabled'
  Custom_Wakeword: 'Enabled'
  Custom_wakeword_models:
    - '/home/pi/GassistPi/src/resources/alexa.umdl'
    - '/home/pi/GassistPi/src/resources/snowboy.umdl'

YouTube:
  YouTube_Control: 'Disabled'
```

shivasiddharth, 2018-12-30:
Seems like GPIO 22 on AIY HAT has been pulled down by default. Can you try changing it to some other pin ?

fragolinux, 2018-12-30:
> Seems like GPIO 22 on AIY HAT has been pulled down by default. Can you try changing it to some other pin ?

ehm, i'm using the stock original (1st version) hat with its own connector (second from the right), 2 wires for the led, 2 for the button... but button is fine, as i tested yesterday when they were configured as standard config.yaml, each time i pressed it, on console appeared the vlc error, so the button is not stuck...

[image]

fragolinux, 2018-12-30:
i'll try in the afternoon to shut down gassistpi and test with some basic gpio script, but as said, yesterday button was working just fine (at least, it triggered vlc every time i pressed it)...

shivasiddharth, 2018-12-30:
To clear things up, first AIY HAT's arcade button (23) was set to stop and Triggering button was (22). As you wanted the arcade button to trigger, i asked you to swap them both. Now 23 is for triggering and 22 is for stopping music. The AIY' 22 has been pulled low by default, so i asked you to change the 22 to 26.
The config file should look like this now:

```
#GPIOS used in the project. Feel free to change in case.
Gpios:
  AIY_indicator:
    - 25
  assistant_indicators:
    - 05
    - 06
  picontrol:
    - 12
    - 13
    - 24
  pushbutton_trigger:
    - 23
  servo:
    - 27
  stopbutton_music_AIY_pushbutton:
    - 26

```

fragolinux, 2018-12-30:
now as you said with stopbutton 26, no more vlc problems in console, but randomly as it's answering it just crashes in the middle of the sentence... if i can do some logs, tell me how...

shivasiddharth, 2018-12-30:
Its an ongoing issue with the grpc https://github.com/googlesamples/assistant-sdk-python/issues/303