shivasiddharth/GassistPi #680: audio_output_stream.cc

sugoei, 2018-12-09:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```pi@raspberrypi:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02082
Serial          : 00000000ffedb192
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
```After authentication, the Google Assistant Demo test started automatically. and recognized my voice but had an errors as below.

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```Registering...Done.

ON_MUTED_CHANGED:
  {"is_muted": false}
ON_START_FINISHED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "where are you"}
ON_RESPONDING_STARTED:
  {"is_error_response": false}[ERROR:audio_output_stream.cc(49)] Device could not be opened: default

ON_RESPONDING_FINISHED
ON_RENDER_RESPONSE:
{
  "text": "You can find me on all kinds of devices, like phones and Google Home. Pretty sweet digs, if you ask me",
  "type": 0
}
ON_RESPONDING_STARTED:
  {"is_error_response": true}
[ERROR:audio_output_stream.cc(49)] Device could not be opened: default
ON_RESPONDING_FINISHED
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

^Z
[2]+  Stopped                 sudo ./GassistPi/scripts/gassist-installer.sh

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

ykr412, 2018-12-09:
i got this error
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
{"text": "wireless turn device on"}
ON_RECOGNIZING_SPEECH_FINISHED:
{"text": "wireless turn device on"}
Turning On Device
ERROR:main:Language not supported: en
Traceback (most recent call last):
File "/home/pi/GassistPi/src/main.py", line 692, in 
Myassistant().main()
File "/home/pi/GassistPi/src/main.py", line 557, in main
ESP(str(usrcmd).lower())
File "/home/pi/GassistPi/src/actions.py", line 295, in ESP
say("Turning On " + name)
File "/home/pi/GassistPi/src/actions.py", line 230, in say
tts = gTTS(text=newword, lang=language)
File "/home/pi/env/lib/python3.5/site-packages/gtts/tts.py", line 119, in init
raise ValueError("Language not supported: %s" % lang)
ValueError: Language not supported: en

this is to control nodemcu

shivasiddharth, 2018-12-09:
Have you changed the language settings in the config.yaml ? The en option should be within quotes like 'en'.

ykr412, 2018-12-09:
#Change you language. Default is english
Language:
  Choice: 'en'
  Language options:
     - 'en' #'English'
     - 'fr' # 'French'
     - 'de' # 'German'
     - 'it' # 'Italian'
     - 'es' # 'Spanish'
     - 'nl' # 'Dutch'
     - 'sv' # 'Swedish'
Its the same

shivasiddharth, 2018-12-10:
What is the audio setup that you are using?

Hoeby, 2018-12-19:
yesterday i was running manual on my gassistpi.
I saw the same error. After a complet restart of the raspberry pi the error was gone. 

Mine .asoundrc 
```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```

sugoei, 2018-12-21:
My audio setup was hdmi and usb-mic. Now I am using respeaker and it works good.

sugoei, 2018-12-26:
I reinstalled rasbian and it works.

closed 2018-12-26 by sugoei