shivasiddharth/GassistPi #681: Nothing in the config.yamil works

ykr412, 2018-12-09:
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
{"text": "wireless turn device on"}
ON_RECOGNIZING_SPEECH_FINISHED:
{"text": "wireless turn device on"}
Turning On Device
ERROR:main:Language not supported: en
Traceback (most recent call last):
File "/home/pi/GassistPi/src/main.py", line 692, in 
Myassistant().main()
File "/home/pi/GassistPi/src/main.py", line 557, in main
ESP(str(usrcmd).lower())
File "/home/pi/GassistPi/src/actions.py", line 295, in ESP
say("Turning On " + name)
File "/home/pi/GassistPi/src/actions.py", line 230, in say
tts = gTTS(text=newword, lang=language)
File "/home/pi/env/lib/python3.5/site-packages/gtts/tts.py", line 119, in init
raise ValueError("Language not supported: %s" % lang)
ValueError: Language not supported: en

this is to control nodemcu

ykr412, 2018-12-09:
anything i ask in the config.yamil i get an error
ON_MUTED_CHANGED:
{"is_muted": false}
ON_MUTED_CHANGED:
{"is_muted": false}
ON_START_FINISHED
ON_START_FINISHED
ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
{"text": "who are you"}
ON_RECOGNIZING_SPEECH_FINISHED:
{"text": "who are you"}
Who do you think I am
ERROR:main:name 'Keyerror' is not defined
Traceback (most recent call last):
File "/home/pi/GassistPi/src/main.py", line 489, in main
say(selectedans[0])
File "/home/pi/GassistPi/src/actions.py", line 230, in say
tts = gTTS(text=newword, lang=language)
File "/home/pi/env/lib/python3.5/site-packages/gtts/tts.py", line 119, in init
raise ValueError("Language not supported: %s" % lang)
ValueError: Language not supported: en

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/home/pi/GassistPi/src/main.py", line 692, in 
Myassistant().main()
File "/home/pi/GassistPi/src/main.py", line 491, in main
except Keyerror:
NameError: name 'Keyerror' is not defined

shivasiddharth, 2018-12-09:
Listen, you have already posted the same error in #680 and i am looking into it. So whats the point in creating multiple posts for the same error ?

ykr412, 2018-12-09:
Im sorry for that my mistake

shivasiddharth, 2018-12-09:
I have reopened this issue. I have remved your comments from the other issue.

ykr412, 2018-12-09:
Thank you for your help i really appreciate it

shivasiddharth, 2018-12-09:
run the following command and share the output

```
source env/bin/activate
pip list
```

ykr412, 2018-12-09:
pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ pip list
Package                  Version   
------------------------ ----------
aftership                0.2       
appdirs                  1.4.3     
asn1crypto               0.24.0    
beautifulsoup4           4.6.3     
bs4                      0.0.1     
cachetools               3.0.0     
casttube                 0.2.0     
certifi                  2018.10.15
cffi                     1.11.5    
chardet                  3.0.4     
click                    6.7       
cryptography             2.4.1     
decorator                4.3.0     
enum34                   1.1.6     
feedparser               5.2.1     
future                   0.17.1    
futures                  3.1.1     
gmusicapi                11.0.3    
google-api-python-client 1.7.4     
google-assistant-grpc    0.2.0     
google-assistant-library 1.0.0     
google-assistant-sdk     0.5.0     
google-auth              1.6.1     
google-auth-httplib2     0.0.3     
google-auth-oauthlib     0.2.0     
googleapis-common-protos 1.5.5     
googletrans              2.3.0     
gpiozero                 1.4.1     
gpsoauth                 0.4.1     
grpcio                   1.16.1    
gTTS                     2.0.2     
gTTS-token               1.1.3     
httplib2                 0.12.0    
idna                     2.7       
ifaddr                   0.1.4     
ipaddress                1.0.22    
kodi-json                1.0.0     
lxml                     4.2.5     
MechanicalSoup           0.11.0    
mock                     2.0.0     
mutagen                  1.41.1    
numpy                    1.15.4    
oauth2client             4.1.3     
oauthlib                 2.1.0     
pafy                     0.5.4     
pathlib2                 2.3.2     
pbr                      5.1.1     
pip                      18.1      
pkg-resources            0.0.0     
proboscis                1.2.6.0   
protobuf                 3.6.1     
psutil                   5.4.8     
pushbullet.py            0.11.0    
pyasn1                   0.4.4     
pyasn1-modules           0.2.2     
PyAudio                  0.2.11    
PyChromecast             2.3.0     
pycparser                2.19      
pycryptodomex            3.7.0     
pyOpenSSL                18.0.0    
python-dateutil          2.7.5     
python-magic             0.4.15    
python-vlc               3.0.4106  
PyYAML                   3.12      
requests                 2.20.1    
requests-oauthlib        1.0.0     
RPi.GPIO                 0.6.3     
rsa                      4.0       
setuptools               40.6.2    
six                      1.11.0    
sounddevice              0.3.12    
spidev                   3.2       
spotipy                  2.4.4     
tenacity                 4.12.0    
uritemplate              3.0.0     
urllib3                  1.24.1    
validictory              1.1.2     
websocket-client         0.54.0    
wheel                    0.32.2    
youtube-dl               2018.11.18
zeroconf                 0.21.3

ykr412, 2018-12-09:
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "wireless turn device on"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "wireless turn device on"}
Turning On Device
WARNING:gtts.tts:Unable to get language list: 'NoneType' object is not subscriptable
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
ERROR:__main__:HTTPSConnectionPool(host='192.168.0.115', port=443): Max retries exceeded with url: /Device1=ON (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x6f91ee70>: Failed to establish a new connection: [Errno 111] Connection refused',))
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py", line 159, in _new_conn
    (self._dns_host, self.port), self.timeout, **extra_kw)
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/util/connection.py", line 80, in create_connection
    raise err
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/util/connection.py", line 70, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 343, in _make_request
    self._validate_conn(conn)
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 839, in _validate_conn
    conn.connect()
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py", line 301, in connect
    conn = self._new_conn()
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py", line 168, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
urllib3.exceptions.NewConnectionError: <urllib3.connection.VerifiedHTTPSConnection object at 0x6f91ee70>: Failed to establish a new connection: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 638, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/util/retry.py", line 398, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='192.168.0.115', port=443): Max retries exceeded with url: /Device1=ON (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x6f91ee70>: Failed to establish a new connection: [Errno 111] Connection refused',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 692, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 557, in main
    ESP(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 299, in ESP
    rq = requests.head("https://"+ip + dev + ctrl)
  File "/home/pi/env/lib/python3.5/site-packages/requests/api.py", line 101, in head
    return request('head', url, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='192.168.0.115', port=443): Max retries exceeded with url: /Device1=ON (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x6f91ee70>: Failed to establish a new connection: [Errno 111] Connection refused',))

shivasiddharth, 2018-12-09:
> ERROR:main:HTTPSConnectionPool(host='192.168.0.115', port=443): Max retries exceeded with url: /Device1=ON (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x6f91ee70>: Failed to establish a new connection: [Errno 111] Connection refused',))
> Traceback (most recent call last):
> File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py", line 159, in _new_conn
> (self._dns_host, self.port), self.timeout, **extra_kw)
> File "/home/pi/env/lib/python3.5/site-packages/urllib3/util/connection.py", line 80, in create_connection
> raise err
> File "/home/pi/env/lib/python3.5/site-packages/urllib3/util/connection.py", line 70, in create_connection
> sock.connect(sa)
> ConnectionRefusedError: [Errno 111] Connection refused
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
> File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 600, in urlopen
> chunked=chunked)
> File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 343, in _make_request
> self._validate_conn(conn)
> File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 839, in _validate_conn
> conn.connect()
> File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py", line 301, in connect
> conn = self._new_conn()
> File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py", line 168, in _new_conn
> self, "Failed to establish a new connection: %s" % e)
> urllib3.exceptions.NewConnectionError: <urllib3.connection.VerifiedHTTPSConnection object at 0x6f91ee70>: Failed to establish a new connection: [Errno 111] Connection refused
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
> File "/home/pi/env/lib/python3.5/site-packages/requests/adapters.py", line 449, in send
> timeout=timeout
> File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 638, in urlopen
> _stacktrace=sys.exc_info()[2])
> File "/home/pi/env/lib/python3.5/site-packages/urllib3/util/retry.py", line 398, in increment
> raise MaxRetryError(_pool, url, error or ResponseError(cause))
> urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='192.168.0.115', port=443): Max retries exceeded with url: /Device1=ON (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x6f91ee70>: Failed to establish a new connection: [Errno 111] Connection refused',))
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
> File "/home/pi/GassistPi/src/main.py", line 692, in 
> Myassistant().main()
> File "/home/pi/GassistPi/src/main.py", line 557, in main
> ESP(str(usrcmd).lower())
> File "/home/pi/GassistPi/src/actions.py", line 299, in ESP
> rq = requests.head("https://"+ip + dev + ctrl)
> File "/home/pi/env/lib/python3.5/site-packages/requests/api.py", line 101, in head
> return request('head', url, **kwargs)
> File "/home/pi/env/lib/python3.5/site-packages/requests/api.py", line 60, in request
> return session.request(method=method, url=url, **kwargs)
> File "/home/pi/env/lib/python3.5/site-packages/requests/sessions.py", line 533, in request
> resp = self.send(prep, **send_kwargs)
> File "/home/pi/env/lib/python3.5/site-packages/requests/sessions.py", line 646, in send
> r = adapter.send(request, **kwargs)
> File "/home/pi/env/lib/python3.5/site-packages/requests/adapters.py", line 516, in send
> raise ConnectionError(e, request=request)
> requests.exceptions.ConnectionError: HTTPSConnectionPool(host='192.168.0.115', port=443): Max retries exceeded with url: /Device1=ON (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x6f91ee70>: Failed to establish a new connection: [Errno 111] Connection refused',))

Your Gtts error has been fixed. The above mentioned one is because of a wrong port/ip address of your nodemcu device or your nodemcu device might not be powered on. As the primary issue has been fixed, closing this issue. Try controlling your nodemcu from browser to double check.

closed 2018-12-09 by shivasiddharth

shivasiddharth, 2018-12-09:
For the Nodemcu feature to work, you should use the firmware that i have given in the customization section. Some random firmwares will not work.

ykr412, 2018-12-09:
Im using the one in the first option and the same code that has been given , which fromware did you mean ? , it works in the web browser very well

shivasiddharth, 2018-12-09:
https://github.com/shivasiddharth/iotwemos/blob/master/Google-Home-NodeMCU.ino this is the firmware that you should be running.

ykr412, 2018-12-09:
Thats the one im using , do i have a problem?

shivasiddharth, 2018-12-09:
Last tie when you tried device might not have been online, try again once more if it crashes then post the output.

commit 6e85ee85 (2018-12-09): Update GassistPi-pip-requirements.txt

Fixes #681