shivasiddharth/GassistPi #682: Token issue again.

Mark-papi, 2018-12-09:
For some reason I got the same issue once again,  after I opened my pi one month after.
The issue was  #608 but I had fixed it,  and I got it again today.
I will post the log file tomorrow cause I can't right now.
Do we have any idea what's up with the new error after I was apply the fix?
It is not a permanent fix the #534 issue?

Mark-papi, 2018-12-09:
here's the log file 

2018-12-10 00:51:01,139 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-12-10 00:51:01,145 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com:443
2018-12-10 00:51:01,515 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-12-10 00:52:02,779 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com:443
2018-12-10 00:52:03,399 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2018-12-10 00:52:03,576 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET /translate_a/single?sl=auto&tk=174830.270632&q=Tuning+into+Radio+2&otf=1&oe=UTF-8&tl=en&ie=UTF-8&tsel=0&dt=at&dt=bd&dt=ex&dt=ld&dt=md&dt=qca&dt=rw&dt=rm&dt=ss&dt=t&client=t&hl=en&ssel=0 HTTP/1.1" 200 None
2018-12-10 00:52:03,580 DEBUG gtts.tts tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=\\!).|(?<=\\？).|(?<=\\！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z])\\, |\\،|\\、|\\;|\\]|\\[|\\¡|\\:|\\‥|\\\n|\\。|\\)|\\，|\\：|\\¿|\\…|\\(|\\—', re.IGNORECASE) from: [<function tone_marks at 0x74dbd198>, <function period_comma at 0x74dbd1e0>, <function other_punctuation at 0x74dbd228>]>
2018-12-10 00:52:03,581 DEBUG gtts.tts lang: en
2018-12-10 00:52:03,581 DEBUG gtts.tts pre_processor_funcs: [<function tone_marks at 0x74e0bc48>, <function end_of_line at 0x74dbd078>, <function abbreviations at 0x74dbd0c0>, <function word_sub at 0x74dbd108>]
2018-12-10 00:52:03,582 DEBUG gtts.tts slow: False
2018-12-10 00:52:03,582 DEBUG gtts.tts lang_check: True
2018-12-10 00:52:03,582 DEBUG gtts.tts text: Tuning into Radio 2
2018-12-10 00:52:03,589 DEBUG urllib3.connectionpool Starting new HTTP connection (1): translate.google.com:80
2018-12-10 00:52:03,717 DEBUG urllib3.connectionpool http://translate.google.com:80 "GET / HTTP/1.1" 302 226
2018-12-10 00:52:03,726 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com:443
2018-12-10 00:52:04,262 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2018-12-10 00:52:04,539 DEBUG gtts.tts Unable to get language list: 'NoneType' object is not subscriptable
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.4/site-packages/gtts/lang.py", line 36, in tts_langs
    langs.update(_fetch_langs())
  File "/home/pi/env/lib/python3.4/site-packages/gtts/lang.py", line 62, in _fetch_langs
    js_path = soup.find(src=re.compile(JS_FILE))['src']
TypeError: 'NoneType' object is not subscriptable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.4/site-packages/gtts/tts.py", line 117, in __init__
    langs = tts_langs()
  File "/home/pi/env/lib/python3.4/site-packages/gtts/lang.py", line 41, in tts_langs
    raise RuntimeError("Unable to get language list: %s" % str(e))
RuntimeError: Unable to get language list: 'NoneType' object is not subscriptable
2018-12-10 00:52:04,544 WARNING gtts.tts Unable to get language list: 'NoneType' object is not subscriptable
2018-12-10 00:52:04,545 DEBUG gtts.tts pre-processing: <function tone_marks at 0x74e0bc48>
2018-12-10 00:52:04,548 DEBUG gtts.tts pre-processing: <function end_of_line at 0x74dbd078>
2018-12-10 00:52:04,548 DEBUG gtts.tts pre-processing: <function abbreviations at 0x74dbd0c0>
2018-12-10 00:52:04,553 DEBUG gtts.tts pre-processing: <function word_sub at 0x74dbd108>
2018-12-10 00:52:04,553 DEBUG gtts.tts text_parts: 1
2018-12-10 00:52:04,560 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com:443
2018-12-10 00:52:04,961 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2018-12-10 00:52:05,254 ERROR __main__ substring not found
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 508, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 399, in main
    radio(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 222, in radio
    say("Tuning into " + name)
  File "/home/pi/GassistPi/src/actions.py", line 171, in say
    tts.save(ttsfilename)
  File "/home/pi/env/lib/python3.4/site-packages/gtts/tts.py", line 247, in save
    self.write_to_fp(f)
  File "/home/pi/env/lib/python3.4/site-packages/gtts/tts.py", line 187, in write_to_fp
    part_tk = self.token.calculate_token(part)
  File "/home/pi/env/lib/python3.4/site-packages/gtts_token/gtts_token.py", line 28, in calculate_token
    seed = self._get_token_key()
  File "/home/pi/env/lib/python3.4/site-packages/gtts_token/gtts_token.py", line 65, in _get_token_key
    lineStr = lineStr[lineStr.index("TKK='")+5:len(lineStr)]
ValueError: substring not found
2018-12-10 00:52:13,489 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-12-10 00:52:13,495 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com:443
2018-12-10 00:52:13,856 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-12-10 00:52:35,889 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): translate.google.com:443
2018-12-10 00:52:36,319 DEBUG urllib3.connectionpool https://translate.google.com:443 "GET / HTTP/1.1" 200 None
[... cut ...]
  File "/home/pi/env/lib/python3.4/site-packages/gtts_token/gtts_token.py", line 65, in _get_token_key
    lineStr = lineStr[lineStr.index("TKK='")+5:len(lineStr)]
ValueError: substring not found
2018-12-10 00:54:38,941 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-12-10 00:54:38,948 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com:443
2018-12-10 00:54:39,468 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None

shivasiddharth, 2018-12-09:
run the following command and share the output

```
source env/bin/activate
pip list
```

Mark-papi, 2018-12-09:
Ok thank you!! 
One moment to run it,  and I will post an update in 10-20 minutes max

Mark-papi, 2018-12-09:
aftership                0.2      
appdirs                  1.4.3    
beautifulsoup4           4.6.0    
bs4                      0.0.1    
cachetools               2.1.0    
certifi                  2018.4.16
cffi                     1.11.5   
chardet                  3.0.4    
click                    6.7      
decorator                4.3.0    
enum34                   1.1.6    
feedparser               5.2.1    
future                   0.16.0   
futures                  3.1.1    
gmusicapi                11.0.1   
google-api-python-client 1.7.3    
google-assistant-grpc    0.2.0    
google-assistant-library 1.0.0    
google-assistant-sdk     0.5.0    
google-auth              1.5.0    
google-auth-httplib2     0.0.3    
google-auth-oauthlib     0.2.0    
googleapis-common-protos 1.5.3    
googletrans              2.3.0    
gpsoauth                 0.4.1    
grpcio                   1.12.1   
gTTS                     2.0.0    
gTTS-token               1.1.1    
httplib2                 0.11.3   
idna                     2.7      
kodi-json                1.0.0    
lxml                     4.2.1    
MechanicalSoup           0.10.0   
mock                     2.0.0    
mutagen                  1.40.0   
netifaces                0.10.7   
oauth2client             4.1.2    
oauthlib                 2.1.0    
pafy                     0.5.4    
pathlib2                 2.3.2    
pbr                      4.0.4    
pip                      18.1     
proboscis                1.2.6.0  
protobuf                 3.6.0    
psutil                   5.4.6    
pushbullet.py            0.11.0   
pyasn1                   0.4.3    
pyasn1-modules           0.2.1    
PyAudio                  0.2.11   
PyChromecast             2.2.0    
pycparser                2.18     
pycryptodomex            3.6.1    
python-dateutil          2.7.3    
python-magic             0.4.15   
python-vlc               3.0.102  
requests                 2.19.1   
requests-oauthlib        1.0.0    
RPi.GPIO                 0.6.3    
rsa                      3.4.2    
scandir                  1.7      
setuptools               39.2.0   
six                      1.11.0   
sounddevice              0.3.11   
tenacity                 4.12.0   
uritemplate              3.0.0    
urllib3                  1.23     
validictory              1.1.2    
websocket-client         0.48.0   
wheel                    0.31.1   
youtube-dl               2018.6.14
zeroconf                 0.20.0

shivasiddharth, 2018-12-09:
Looks like you have not updated the project for quite sometime. We are talking about months here. Please re-run the installer.

closed 2018-12-09 by shivasiddharth