shivasiddharth/GassistPi #685: /home/root/modelid.txt not found

AlexHowser, 2018-12-10:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
pi@netmonitor:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 00000000f975c532
pi@netmonitor:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

### Describe the bug:
```
The user path switches automatically to /home/root and doesn´t find any modelid.txt file:

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```
Path: /home/{user}/client_secret_286205758875-q9ebhgj5dp127brlhi6fsi8b7ov3t00m.apps.googleusercontent.com.json
Project-ID: piassistradio
ModelId: piassistradio-piassistradio-cqfnc6

pi@netmonitor:~ $ sudo  ./GassistPi/scripts/gassist-installer.sh

Enter the your full credential file name including the path and .json extension: /home/pi/client_secret_286205758875-q9ebhgj5dp127brlhi6fsi8b7ov3t00m.apps.googleusercontent.com.json

Enter the your Google Cloud Console Project-Id: piassistradio

Enter the modelid that was generated in the actions console: piassistradio-piassistradio-cqfnc6

./GassistPi/scripts/gassist-installer.sh: Zeile 35: /home/root/modelid.txt: Datei oder Verzeichnis nicht gefunden
```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

shivasiddharth, 2018-12-10:
Did you use a sudo command to clone the git ?

AlexHowser, 2018-12-10:
Yeah, thats exactly what i did :/ THX!
but now Im running into another problem:
==========Changing particulars in service files==========

==========Changing particulars in service files for Pushbutton/Custom-wakeword==========
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt-get install python3-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/home/pi/env/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']

AlexHowser, 2018-12-10:
as far as i can tell its set to de-utf8

shivasiddharth, 2018-12-10:
Just as i predicted. Run the following and then try the installer.
```
export LC_ALL="de_AT.UTF-8"
export LC_CTYPE="de_AT.UTF-8"
sudo dpkg-reconfigure locales
```

AlexHowser, 2018-12-10:
I executed the commands, did a reboot, but sadly still the same error :/

shivasiddharth, 2018-12-10:
Run the following and paste the output.
`locale -a`

AlexHowser, 2018-12-10:
pi@netmonitor:~ $ locale -a
C
C.UTF-8
de_AT.utf8
POSIX

shivasiddharth, 2018-12-10:
Thanks, can you now just run the following and post the output?
`locale`

AlexHowser, 2018-12-10:
pi@netmonitor:~ $ locale
LANG=de_AT.UTF-8
LANGUAGE=de_AT.UTF-8
LC_CTYPE="de_AT.UTF-8"
LC_NUMERIC="de_AT.UTF-8"
LC_TIME="de_AT.UTF-8"
LC_COLLATE="de_AT.UTF-8"
LC_MONETARY="de_AT.UTF-8"
LC_MESSAGES="de_AT.UTF-8"
LC_PAPER="de_AT.UTF-8"
LC_NAME="de_AT.UTF-8"
LC_ADDRESS="de_AT.UTF-8"
LC_TELEPHONE="de_AT.UTF-8"
LC_MEASUREMENT="de_AT.UTF-8"
LC_IDENTIFICATION="de_AT.UTF-8"
LC_ALL=de_AT.UTF-8

shivasiddharth, 2018-12-10:
Run `sudo dpkg-reconfigure locales` and do not restart. Without restarting, try the installer.

shivasiddharth, 2018-12-10:
If that did not work then try running `export LC_ALL=C` and then the installer.

AlexHowser, 2018-12-10:
same outcome. executed the command, config, pops up, then I selected "all locales",
reran installer

shivasiddharth, 2018-12-10:
Did you try this https://github.com/shivasiddharth/GassistPi/issues/685#issuecomment-445935867 ?

AlexHowser, 2018-12-10:
yes, it delivers no output, installer throws the same error 
is it possible that AT locales do not work at all?

shivasiddharth, 2018-12-10:
Which month's Raspbian release are you using ?

AlexHowser, 2018-12-10:
pi@netmonitor:~ $ uname -a
Linux netmonitor 4.14.86+ #1175 Thu Dec 6 13:47:13 GMT 2018 armv6l GNU/Linux

shivasiddharth, 2018-12-10:
Did you at any point run a dist-upgrade command ? The latest Stable release is on 4.14.79 (November release).

AlexHowser, 2018-12-10:
yeah, that´s possible..

shivasiddharth, 2018-12-10:
Download the latest stable release and try again. Its not possible to roll-back a dist upgrade.

AlexHowser, 2018-12-11:
@shivasiddharth thanks, I ll try it later. Thanks for that quick support so far!

shivasiddharth, 2018-12-28:
Do we have a status on this error ?

lephongtech, 2019-01-10:
I use command (without sudo): git clone https://github.com/shivasiddharth/GassistPi
error:
line 35: /home/root/modelid.txt: No such file or directory
When run : ./GassistPi/scripts/gassist-installer.sh

adas4190, 2019-01-14:
Same here. Any fix?

shivasiddharth, 2019-01-16:
Unless the clone command command was executed as a root user, this will not happen.

shivasiddharth, 2019-02-11:
Closing this for inactivity.

closed 2019-02-11 by shivasiddharth