shivasiddharth/GassistPi #686: NodeMcu via webbrowser error

ykr412, 2018-12-10:
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "wireless turn device on"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "wireless turn device on"}
Turning On Device
WARNING:gtts.tts:Unable to get language list: 'NoneType' object is not subscriptable
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
ERROR:__main__:HTTPSConnectionPool(host='192.168.0.115', port=443): Max retries exceeded with url: /Device1=ON (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x6f798e50>: Failed to establish a new connection: [Errno 111] Connection refused',))
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py", line 159, in _new_conn
    (self._dns_host, self.port), self.timeout, **extra_kw)
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/util/connection.py", line 80, in create_connection
    raise err
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/util/connection.py", line 70, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 343, in _make_request
    self._validate_conn(conn)
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 839, in _validate_conn
    conn.connect()
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py", line 301, in connect
    conn = self._new_conn()
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py", line 168, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
urllib3.exceptions.NewConnectionError: <urllib3.connection.VerifiedHTTPSConnection object at 0x6f798e50>: Failed to establish a new connection: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 638, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/util/retry.py", line 398, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='192.168.0.115', port=443): Max retries exceeded with url: /Device1=ON (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x6f798e50>: Failed to establish a new connection: [Errno 111] Connection refused',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 692, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 557, in main
    ESP(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 299, in ESP
    rq = requests.head("https://"+ip + dev + ctrl)
  File "/home/pi/env/lib/python3.5/site-packages/requests/api.py", line 101, in head
    return request('head', url, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='192.168.0.115', port=443): Max retries exceeded with url: /Device1=ON (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x6f798e50>: Failed to establish a new connection: [Errno 111] Connection refused',))

shivasiddharth, 2018-12-10:
Looks like the IP address of your ESP device is blank in the config.yaml. Thats why the pi is not able to send commands to the ESP. Please double check that.

ykr412, 2018-12-10:
I checked it and its there

ykr412, 2018-12-10:
ConnectionError: HTTPSConnectionPool(host='192.168.0.115', port=443): Max retries exceeded with url: /Device1=ON (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x6f798e50>: Failed to establish a new connection: [Errno 111] Connection refused',))

 
Even in the error code the ip is writen

shivasiddharth, 2018-12-10:
Replace your existing actions.py with the updated script from master.

closed 2018-12-10 by shivasiddharth

ykr412, 2018-12-10:
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "wireless turn lamp on"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "wireless turn lamp on"}
Turning On Lamp
WARNING:gtts.tts:Unable to get language list: 'NoneType' object is not subscriptable
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
ERROR:__main__:HTTPSConnectionPool(host='192.168.0.115', port=443): Max retries exceeded with url: /Device1=ON (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x6f5976d0>: Failed to establish a new connection: [Errno 111] Connection refused',))
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py", line 159, in _new_conn
    (self._dns_host, self.port), self.timeout, **extra_kw)
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/util/connection.py", line 80, in create_connection
    raise err
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/util/connection.py", line 70, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 343, in _make_request
    self._validate_conn(conn)
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 839, in _validate_conn
    conn.connect()
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py", line 301, in connect
    conn = self._new_conn()
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py", line 168, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
urllib3.exceptions.NewConnectionError: <urllib3.connection.VerifiedHTTPSConnection object at 0x6f5976d0>: Failed to establish a new connection: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 638, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/util/retry.py", line 398, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='192.168.0.115', port=443): Max retries exceeded with url: /Device1=ON (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x6f5976d0>: Failed to establish a new connection: [Errno 111] Connection refused',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 692, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 557, in main
    ESP(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 299, in ESP
    rq = requests.head("https://"+ip + dev + ctrl,verify=False)
  File "/home/pi/env/lib/python3.5/site-packages/requests/api.py", line 101, in head
    return request('head', url, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='192.168.0.115', port=443): Max retries exceeded with url: /Device1=ON (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x6f5976d0>: Failed to establish a new connection: [Errno 111] Connection refused',))

ykr412, 2018-12-10:
same error after i replaced the action.py

shivasiddharth, 2018-12-11:
Codes are clean there is nothing wrong with the scripts. Try sending the request from browser on pi, not the browser from a pc or laptop. Check if your pi and esp are under same gateway. If you are running a firewall disable that. If there is a bug or errata in the scripts, I can fix that but for other things I can only give pointers for users to work on.

shivasiddharth, 2018-12-11:
I need some time, need to implement everything by myself to check where things have gone wrong.

ykr412, 2018-12-11:
does python 3.5 have effect on this error ?

shivasiddharth, 2018-12-12:
Try replacing your actions.py with the file from master.

commit 3c0c74c3 (2018-12-10): Update actions.py

Fixes #686