shivasiddharth/GassistPi #687: Youtube playback problem

V1JCM, 2018-12-11:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02082
Serial          : 000000001e7d2e18

```
Paste your Board and OS details here.   
``
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
`

### Describe the bug:
```
GassistPI works otherwise fine, but could not get Youtube playback to work. I have inserted the Google_cloud_api_key in the config.yaml. When trying to ge the youtube playback to happen, google says initiating the playback, then says the song can't be found. When run from the prompt, I get errors like 
ImportError: No module named 'google.appengine'

I have no /pi/mediavolume.json file, other hidden files I do have
I did: 
 source env/bin/activate
 pip install -U youtube-dl
 sudo pip install -U youtube-dl
Did not help...

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play I got you from YouTube"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play I got you from YouTube"}
stopping vlc
Getting youtube link
WARNING:gtts.tts:Unable to get language list: 'NoneType' object is not subscriptable
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
i got you
WARNING:googleapiclient.discovery_cache:file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 36, in autodetect
    from google.appengine.api import memcache
ImportError: No module named 'google.appengine'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ImportError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ImportError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
It is no longer available in youtube
WARNING:gtts.tts:Unable to get language list: 'NoneType' object is not subscriptable
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:03] Decoding of say.mp3 finished.
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

```

shivasiddharth, 2018-12-11:
If a video is blocked in your country, you will get such a prompt. Can you open youtube and see if this video is playing or has been region locked? https://www.youtube.com/watch?v=uEJuoEs1UxY

V1JCM, 2018-12-12:
Hi Sid, thank you for your prompt reply.

The video seems to work (what a lovely girl). I also double checked the API key in Google Cloud and config.yaml. I even tried it with ' ' in config.yaml, no help.

Installation of all went fine and everything else seems to work. Have had meaningful discussions...

I have had a different installation previously (Voice KIT V1) and was able to get Youtube working with the info I found online.

Cheers,
Janne

shivasiddharth, 2018-12-12:
After this line https://github.com/shivasiddharth/GassistPi/blob/f97dec1ad53307d0241005a5550d1dbdbc896c1f/src/actions.py#L1037
 add `print(urlid)` after that manually start the assistant, give the same command and share the terminal output.

V1JCM, 2018-12-12:
I went a bit farther adding this:
-------
1037 urlid=youtube_search(track)
        print ("Urlid below")
        print(urlid) #Sid's advice 12.12.2018
        print ("Urlid above")
-------
**Nothing shows below, the output is the same as the last time. So my uneducated (=no python skills) guess is youtube_search fails and 'try' exits.**

ON_CONVERSATION_TURN_STARTED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play I got you from YouTube"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play I got you from YouTube"}
stopping vlc
Getting youtube link
WARNING:gtts.tts:Unable to get language list: 'NoneType' object is not subscriptable
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
i got you
WARNING:googleapiclient.discovery_cache:file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 36, in autodetect
    from google.appengine.api import memcache
ImportError: No module named 'google.appengine'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ImportError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ImportError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
It is no longer available in youtube
WARNING:gtts.tts:Unable to get language list: 'NoneType' object is not subscriptable
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:02] Decoding of say.mp3 finished.
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

shivasiddharth, 2018-12-12:
It could be an issue with your API. The API key should be within ' ' . But you have said that you already tried that.  Comment out these two lines.

https://github.com/shivasiddharth/GassistPi/blob/f97dec1ad53307d0241005a5550d1dbdbc896c1f/src/actions.py#L1047-L1048

Then whatever is the error, that will appear.

V1JCM, 2018-12-12:
I re-inserted the '' in config.yaml, no help. I also commented the TRY:-bit in actions.py. And now we got results. Please see below, what have I done wrong/has gone wrong? (I <***>d the API key)

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play I got you from YouTube"}
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play I got you from YouTube"}
stopping vlc
Getting youtube link
WARNING:gtts.tts:Unable to get language list: 'NoneType' object is not subscriptable
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
i got you
WARNING:googleapiclient.discovery_cache:file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 36, in autodetect
    from google.appengine.api import memcache
ImportError: No module named 'google.appengine'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ImportError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ImportError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0 or google-auth
ERROR:__main__:<HttpError 400 when requesting https://www.googleapis.com/discovery/v1/apis/youtube/v3/rest?key=AIzaS<**********>Mc returned "Bad Request">
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 692, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 549, in main
    YouTube_No_Autoplay(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1037, in YouTube_No_Autoplay
    urlid=youtube_search(track)
  File "/home/pi/GassistPi/src/youtube_search_engine.py", line 27, in youtube_search
    developerKey=DEVELOPER_KEY)
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/_helpers.py", line 130, in positional_wrapper
    return wrapped(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery.py", line 230, in build
    raise e
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery.py", line 222, in build
    requested_url, discovery_http, cache_discovery, cache, developerKey)
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery.py", line 275, in _retrieve_discovery_doc
    raise HttpError(resp, content, uri=actual_url)
googleapiclient.errors.HttpError: <HttpError 400 when requesting https://www.googleapis.com/discovery/v1/apis/youtube/v3/rest?key=AIzaSyD<***********>Mc returned "Bad Request">
^CException ignored in: <module 'threading' from '/usr/lib/python3.5/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 1288, in _shutdown
    t.join()
  File "/usr/lib/python3.5/threading.py", line 1054, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.5/threading.py", line 1070, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt

shivasiddharth, 2018-12-12:
This means that the youtube data api is not working with the key. First check if you have enabled youtube data v3 api. then click on the key and make sure that it does not have any restrictions. Key without restrictions should look like this.

[image]

[image]

V1JCM, 2018-12-13:
Restrictions: none; none; none. Check.

V1JCM, 2018-12-13:
And in the API list there are a lot of API's, all enabled.

shivasiddharth, 2018-12-13:
Lot of APIs does not matter, please check if Youtube Data API has been enabled.

V1JCM, 2018-12-13:
Yes sir, Youtube Data API has been enabled all the time. No metrics data.

shivasiddharth, 2018-12-13:
> {
> "error": {
> "errors": [
> {
> "domain": "usageLimits",
> "reason": "keyInvalid",
> "message": "Bad Request"
> }
> ],
> "code": 400,
> "message": "Bad Request"
> }
> }

Delete the old Key and generate a new one. This is clearly a key issue.

shivasiddharth, 2018-12-13:
And after changing the post the output without hiding or removing the key. After i fix the issue, you can delete the comment.

V1JCM, 2018-12-14:
Will do tonight! Thank you!

V1JCM, 2018-12-14:
Whoa! And now it works! All we did differently was that after creating the API key, we edited it and changed the name to "API Key 3 Sid". Everything else was the same, except that the key in config.yaml is between ''. And it works. You have any idea as to what had gone wrong (besides the obvious "error 90" possibility)?

We will go on to implement Kodi! We owe you not a cup of coffee but a lunch, thank you!

shivasiddharth, 2018-12-16:
Previously while creating the API keys, we had a drop down to choose the APIs that we require the key to work with. After that, google made some changes and by default the keys started to work with all APIs enabled. I believe that it would have something to do. I frequently change keys so, i cannot say for sure as to whether that is the actual reason. Glad that you got the project working.