shivasiddharth/GassistPi #690: Change Voice does not change the hard-coded comments

V1JCM, 2018-12-14:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```

### Describe the bug:
```
Chose not to fill in the form, I guess the question is more generic. Can't find the answer from git. Goes like this:
1) "Hey google, change voice"
- Google changes voice, say to a male voice
2) Use GassistPi command like: "Hey Google, play I got you from Youtube"
- Google says "Getting autoplay links" in a default/female voice

Is there something to be done?

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).
Yes/No

### 2. Is the issue related to assistant installation ?
Yes/No

### If answer to question 2 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

shivasiddharth, 2018-12-14:
The voice that says "Playing... etc etc" for custom commands is the voice of text to speech converter. That voice is not changeable with the current API that i am using. But an option to change the voice for custom commands is coming in the next update (if time permits mostly this weekend). You will not get variations like the google's voice but you can choose between male and female.

V1JCM, 2018-12-14:
OK, great! No rush for our part, just wondering if anything can be done.

shivasiddharth, 2018-12-14:
Replace your existing actions.py and config.yaml files with the new files from master. Open a terminal and run:
```
sudo apt-get install espeak
source env/bin/activate
pip install pyttsx3
```
Now you should be able to change the voice to male or female in the config.yaml.

closed 2018-12-14 by shivasiddharth

V1JCM, 2018-12-15:
Testing it, the voice is somewhat mechanical. And the startup "Hi!" comes with a different voice. Do you have any plans to update the API?

shivasiddharth, 2018-12-15:
'Hi' is google's voice. I generated an audio file by recording my Google Home say "Hi". This new voice is the updated API. Old API did not have voice modulation capabilities. So you are saying that when the male voice is opted, even the startup tone should be male ? Sounds reasonable.

shivasiddharth, 2018-12-15:
You should create a .wav file not .mp3 and it should be mono. I would suggest you to import the provided .wav files to some audio editor tool to check for the formats.

I am taking tiny steps to iron out the custom audio thingy. Previously gtts was used which had different tone, modulation and voice for each language (some languages will have a male voice and some a female one). The broad idea is to first create a voice with the same modulation for all the languages and then use a plugin to change the pitch to match a female or male voice. 

By changing the API last night, i have managed to get a uniform voice for the different languages. Next step would be to alter its picth. Thats work in progress.

V1JCM, 2018-12-15:
ok, cool. Happy coding!

shivasiddharth, 2018-12-16:
I am receiving 👎s for the new voice from the users. On a scale of 1-10 how bad is it when compared to the old non-male voice. I really need inputs, if the new voice is that bad, then i will have to change direction for further work.

V1JCM, 2018-12-16:
Don't want to be rude mate, please do not get offended. You are doing a great job for us all.
But frankly, the voice is 2/10. Especially when you compare it to Google male (or female) voices, they sound quite human.

shivasiddharth, 2018-12-16:
No issue mate. Feedback always helps. The entire project has been built and modified only based on users' feedbacks and perspectives.

shivasiddharth, 2018-12-16:
Sorry to bother you again.  Please find attached some male and female voices does this sound better ? The ones marked suffixed with "p" are the male versions of their counterparts.
[Audio Male and Female.zip](https://github.com/shivasiddharth/GassistPi/files/2683663/Audio.Male.and.Female.zip)

V1JCM, 2018-12-16:
Happy to help! These are all better than the currently playing voice.

#1 did not play, 5p.wav is my favourite of male voices and 6.wav of female. Both #2 are also good. #3 both sound like late professor Hawking. 4.wav female is ok, but #4 male sounds like he has too much vibration.

shivasiddharth, 2018-12-16:
Robotic voice have been fixed. The voices now should be much better and if a male voice is chosen, now startup tone will also be male.

open a terminal and run:
```
sudo apt-get install sox
sudo apt-get install libsox-fmt-mp3
```
After that, rename your existing directory and re-clone the git.

V1JCM, 2018-12-17:
OK, will do tonight after work. I'll report back to you then.

V1JCM, 2018-12-17:
Did:
 sudo apt-get install sox
 sudo apt-get install libsox-fmt-mp3
Rename old GassistPI folder and:
 git clone https://github.com/shivasiddharth/GassistPi
then copied API key from config.yaml to the new GassistPI
Something is wrong as the service will not run. Manual start will cause all this:

/home/pi/env/lib/python3.5/site-packages/gmusicapi/__init__.py:4: GmusicapiWarning: gmusicapi.clients.OAUTH_FILEPATH is deprecated and will be removed; use Musicmanager.OAUTH_FILEPATH
  from gmusicapi.clients import Webclient, Musicmanager, Mobileclient
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.1.1
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
<string>:2: GmusicapiWarning: login is deprecated and may break unexpectedly; prefer Mobileclient.oauth_login
INFO:gmusicapi.Mobileclient1:failed to authenticate
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 44, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 19, in <module>
    from langdetect import detect
ImportError: No module named 'langdetect'

V1JCM, 2018-12-17:
Seems to work now, thanks. Only it takes like 4-5 seconds to answer the "do you have a family" -question. And the speech volume needs to be set too.

Just a development idea: what about giving the user several voice options instad of just male and female?

V1JCM, 2018-12-17:
This project has a different approach(?): 
https://dronebotworkshop.com/hacking-google-aiy-voice-kit-2/
I had that for a while, and I believe it used google speech as such to say custom frases. Like here:
"
def power_off_pi():
    aiy.audio.say('Good bye!')
    subprocess.call('sudo shutdown now', shell=True)
"

Would it be possible to use that approach in your project?

shivasiddharth, 2018-12-17:
AIY uses pico2wave not cloud speech. Pico2wave has limited language support. It does not support Swedish, Dutch etc. With Google assistant available in a number of languages and more to come, I felt that gtts is more viable option for the long run of the project. Of course I am always on the lookout for alternatives, if some package better than gtts comes up, we will migrate to that.

commit 6d0fbfc1 (2018-12-14): Updated Say fucntion, main.py and pushbutton.py

Fixes #675  Fixes #690

1. Added a switch for major custom actions. Users can toggle the custom actions "ON" or "OFF".

2. Added a feature to allow users to change the say function voice to either male or female.

3. Fixed Pi CPU usage at idle.
PR #691 (2018-12-14): Custom actions switch
Fixes #675  Fixes #690

1. Added a switch for major custom actions. Users can toggle the custom actions "ON" or "OFF".

2. Added a feature to allow users to change the say function voice to either male or female.

3. Fixed Pi CPU usage at idle.
commit 61a8acb0 (2018-12-14): Custom actions switch (#691)

* Enable switches for custom actions

Users can turn on or off the custom actions by setting the control for respective custom actions in config.yaml to 'Enabled' or 'Disabled'

* Fix Kodi control repeat

* Updated Say fucntion, main.py and pushbutton.py

Fixes #675  Fixes #690

1. Added a switch for major custom actions. Users can toggle the custom actions "ON" or "OFF".

2. Added a feature to allow users to change the say function voice to either male or female.

3. Fixed Pi CPU usage at idle.

* Update README.md
commit 0a1a9770 (2018-12-16): Update voices

1. Fixed robotic voice.
2. Added a male startup tone.
3. Added male tones for mic mute and unmute. Fixes #690